srsran/srsRAN_Project #944: OSC RIC: appmgr can’t connect to dbaas – config seems to be ignored

lmendyk, 2024-11-21:
When starting appmgr, in the logs I get:
```
dial tcp [::1]:6379: connect: connection refused
{"ts":1732200055823,"crit":"ERROR","id":"appmgr","mdc":{"CONTAINER_NAME":"","HOST_NAME":"","PID":"8","POD_NAME":"","SERVICE_NAME":"","SYSTEM_NAME":"","time":"2024-11-21T14:40:55Z","xm":"0.4.3"},"msg":"Database connection not ready, waiting ..."}
{"ts":1732200060849,"crit":"ERROR","id":"appmgr","mdc":{"CONTAINER_NAME":"","HOST_NAME":"","PID":"8","POD_NAME":"","SERVICE_NAME":"","SYSTEM_NAME":"","time":"2024-11-21T14:41:00Z","xm":"0.4.3"},"msg":"Database connection not ready, waiting ..."}
{"ts":1732200065875,"crit":"ERROR","id":"appmgr","mdc":{"CONTAINER_NAME":"","HOST_NAME":"","PID":"8","POD_NAME":"","SERVICE_NAME":"","SYSTEM_NAME":"","time":"2024-11-21T14:41:05Z","xm":"0.4.3"},"msg":"Database connection not ready, waiting ..."}
…
```

The config file in the container seems to be OK, but the IP address and port seem to be ignored as no matter what I change it to I still get the same errors in the log .

```
root@appmgr:/opt/ric/config# cat appmgr.yaml 
loglevel: 1
local:
  host: ":8080"
xapp:
  namespace: "ricxapp"
  tarDir: "/tmp"
  schema: "descriptors/schema.json"
  config: "config/config-file.json"
  tmpConfig: "/tmp/config-file.json"
db:
  sessionNamespace: "XMSession"
  host: "ric-dbaas-service-tcp:6379"
  prot: "tcp"
  maxIdle: 80
  maxActive: 12000

```

pgawlowicz, 2024-11-21:
could you remove the container and the image, and try to build again?

pgawlowicz, 2024-11-22:
Closing as it is not issue of srsRAN Project.

closed 2024-11-22 by pgawlowicz