srsran/srsRAN_Project #958: UE cannot connect to srsRAN

mireianogue, 2024-11-27:
## Issue Description ##
I have problems connecting my UE to srsRAN. 

## Setup Details ##
2 PC Lenovo IntelCore i5-10600 CPU
SIM sysmoISIM-SJA5 
CrosCall core Z5
USRP B210
2 antennas

Here are the files: (I haven't attached the gnb.log file because it doesn't contain anything)
gnb config file: 
[gnb_rf_b200_tdd_n78_10.txt](https://github.com/user-attachments/files/17938714/gnb_rf_b200_tdd_n78_10.txt)
core files: 
[nrf.txt](https://github.com/user-attachments/files/17938719/nrf.txt)
[amf.txt](https://github.com/user-attachments/files/17938723/amf.txt)
[upf.txt](https://github.com/user-attachments/files/17938732/upf.txt)

I have already programmed the SIM and the core:
[image]
[image]

## Expected Behavior ##
I expect to see trafic and the mobile phone connected to the network:
[image]

## Actual Behaviour ##
Actually, I have achived to see the network in app called Qualipoc:
[image]
[image]
[image]

But in the console of gnb i see this:
[image]

Does anyone has the same problem?

pgawlowicz, 2024-11-27:
could you share you gnb logs?

mireianogue, 2024-11-28:
Here it is the gnb log file:
[gnb.log.txt.log](https://github.com/user-attachments/files/17943884/gnb.log.txt.log)

mireianogue, 2024-11-28:
Also, i'm going to share the singalling that I'm obtaining in Qualipoc app:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

mireianogue, 2024-12-02:
Where can I find this file?

mireianogue, 2024-12-02:
[ngap_pcap_test.txt](https://github.com/user-attachments/files/17981354/ngap_pcap_test.txt)
[ngap_ue.txt](https://github.com/user-attachments/files/17981388/ngap_ue.txt)
[ngap.txt](https://github.com/user-attachments/files/17981385/ngap.txt)

mireianogue, 2024-12-02:
I'm going to share also, the Amf logs:
[image]

mireianogue, 2024-12-03:
[image]
[image]

mireianogue, 2024-12-03:
[gnb_mac.zip](https://github.com/user-attachments/files/17990607/gnb_mac.zip)
[gnb_ngap.zip](https://github.com/user-attachments/files/17990613/gnb_ngap.zip)

pgawlowicz, 2024-12-03:
https://docs.srsran.com/projects/project/en/latest/user_manuals/source/outputs.html#pcaps

mireianogue, 2024-12-03:
i have modified these wireshark files as shown in this link:
[gnb_mac.zip](https://github.com/user-attachments/files/17990788/gnb_mac.zip)
[gnb_ngap.zip](https://github.com/user-attachments/files/17990791/gnb_ngap.zip)

mireianogue, 2024-12-12:
I have started again the costs UE tutorial and the mobile phone can see the network, but cannot attach. What do you mean for "NSG"?
[image]

pgawlowicz, 2024-12-12:
i guess Network Signal Guru app.

mireianogue, 2024-12-16:
Hello, 
I'm troubling with the same problem. But now i'm obtaining a diferent gnb_mac_pcap message:
 [gnb_mac_pcap.zip](https://github.com/user-attachments/files/18154435/gnb_mac_pcap.zip)

Do you able to tell me what's the error? Thank you in advance.

pgawlowicz, 2024-12-16:
please share ngap.pcap

mireianogue, 2024-12-16:
[image]
[image]

pgawlowicz, 2024-12-16:
looks like sth is wrong with the core network config.

mireianogue, 2024-12-16:
Maybe in the subscription part?

mireianogue, 2024-12-16:
[image]
[image]

mireianogue, 2024-12-16:
I can share the core files configurations but i think that they are right:
[amf.zip](https://github.com/user-attachments/files/18154640/amf.zip)
[nrf.zip](https://github.com/user-attachments/files/18154641/nrf.zip)
[upf.zip](https://github.com/user-attachments/files/18154642/upf.zip)

pgawlowicz, 2024-12-16:
do you use the newest version of the srsRAN project?

mireianogue, 2024-12-16:
I installed srsRAN Project on July 2024 using the source code located https://github.com/srsran/srsran_project. 
But i have in the same both PC the srsRAN for LTE. Maybe is that a problem?

pgawlowicz, 2024-12-16:
please update with the newest version. Do you run srsRAN LTE at the same time?

mireianogue, 2024-12-16:
Okeis i'm going to update with the newest version. No, i haven't! I only run srsRAN_Project (5G). 
Thank you. I will notify you if i achieved to attach the UE.

mireianogue, 2024-12-17:
Hello, 
I've achieved to attach the UE. The problem was that the OPc value was wrong by mistake. Thank you for your help! You can close this issue.

closed 2024-12-17 by pgawlowicz