srsran/srsRAN_Project #961: Add SoapySDR support

ohault, 2024-12-01:
## Issue Description ##
Direct dependency of applications with specific hardware devices/drivers introduces serious limitations. 
SoapySDR exists to solve this architectural problem at scale for SDR devices and applications.

## Expected Behavior ##
Support of SoapySDR in this srsRAN project.

## Actual Behaviour ##
SoapySDR is currenty supported only in the srsRAN 4G project.

pgawlowicz, 2024-12-02:
Adding support for SoapySDR is not planned currently. 

However, some users successfully used srsRAN-5G with SoapySDR, please check [here](https://github.com/srsran/srsRAN_Project/discussions/222#discussioncomment-7120314).

closed 2024-12-02 by pgawlowicz

ohault, 2024-12-02:
Hi @pgawlowicz, thank you for your answer, but may I ask you to stop closing tickets that are obviously not done, especially when there are important features requested by the community about the future of srsRAN. Currently this only degrade the brand image of srsRAN altogether. Thank you for your understanding.

pgawlowicz, 2024-12-02:
Hi @ohault, thank you for your feedback. 
Please note that we keep the Issues section focused on real code issues (i.e., bugs), we do not track feature requests here. We kindly ask that you move your post to the Discussions section, which is better suited for suggestions and community input. Please note that from our side, we do not plan to support this feature.
Thank you for your understanding.

pgawlowicz, 2024-12-02:
Please refer to this post: https://github.com/srsran/srsRAN_Project/discussions/24

ohault, 2024-12-02:
> Please refer to this post: #24

Then I would suggest to also update this text in the template used to create a new issue "Before filing an issue, search for solutions here:
- srsRAN Project discussions page (https://github.com/srsran/srsRAN_Project/discussions)"

Mind you to convert this ticket into a Discussion, meanwhile, I don't really understand what there is to discuss about SoapySDR support ?

Using Github Projects feature already activated in this repository will be, as far as I know more appropriate.