srsran/srsRAN_Project #971: Late: 35; Underflow: 1; Overflow: 0; message when I am connecting galaxy M33 5G with enb through USRP x310

sruthi81, 2024-12-05:
Here I am sharing the problem when I am connecting my ue with gnb. I am running Open5GS and srsRAN in same machine and usrp x310 is used as RU. 
[image]
 "Late: 35; Underflow: 1; Overflow: 0;" message.

Please help me to solve this issue.

pgawlowicz, 2024-12-05:
could you share your configs and log?

sruthi81, 2024-12-05:
Yes Sir.

# This configuration file example shows how to configure the srsRAN Project gNB to allow srsUE to connect to it. 
# This specific example uses ZMQ in place of a USRP for the RF-frontend, and creates an FDD cell with 10 MHz bandwidth. 
# To run the srsRAN Project gNB with this config, use the following command: 
#   sudo ./gnb -c gnb_zmq.yaml

amf:
  addr: 127.0.0.5                 # The address or hostname of the AMF.
  bind_addr: 127.0.0.5             # A local IP that the gNB binds to for traffic from the AMF.

ru_sdr:
  device_driver: uhd                # The RF driver name.
  device_args: addr=192.168.10.2 #tx_port=tcp://127.0.0.1:2000,rx_port=tcp://127.0.0.1:2001,base_srate=23.04e6 # Optionally pass arguments to the selected RF driver.
  #sync:  gpsdo                   # Specify the sync source used by the RF.
#device_args: tx_port=tcp://192.168.0.105:2000,rx_port=tcp://192.168.0.114:2001,base_srate=23.04e6 # Optionally pass arguments to the selected RF driver.
  clock: internal  
  srate: 23.04                      # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 30                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 30                       # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 368500                  # ARFCN of the downlink carrier (center frequency).
  band: 3                           # The NR band.
  channel_bandwidth_MHz: 20         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 15                    # Subcarrier spacing in kHz used for data.
  plmn: "00101"                     # PLMN broadcasted by the gNB.
  tac: 7                            # Tracking area code (needs to match the core configuration).
  pdcch:
    common:
      ss0_index: 0                  # Set search space zero index to match srsUE capabilities
      coreset0_index: 12            # Set search CORESET Zero index to match srsUE capabilities
    dedicated:
      ss2_type: common              # Search Space type, has to be set to common
      dci_format_0_1_and_1_1: false # Set correct DCI format (fallback)
  prach:
    prach_config_index: 1           # Sets PRACH config to match what is expected by srsUE

cu_cp:
  inactivity_timer: 7200            # Sets the UE/PDU Session/DRB inactivity timer to 7200 seconds. Supported: [1 - 7200].

log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: info                   # Logging level applied to all layers.
  hex_max_size: 0

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.

I cant open the amf log file, it shows permission denied.

pgawlowicz, 2024-12-05:
you can find logs at /tmp/gnb.log

sruthi81, 2024-12-10:
ok

sruthi81, 2025-01-10:
Sir, 
Please help me to resolve this issue
 Thanks

pgawlowicz, 2025-01-10:
please try to run this script:
https://github.com/srsran/srsRAN_Project/blob/main/scripts/srsran_performance

and try again.

sruthi81, 2025-01-13:
[image]
Sir, is it correct?

sruthi81, 2025-01-13:
[image]

sruthi81, 2025-01-13:
same result I got after running the script.

sruthi81, 2025-01-15:
please help me to resolve the issue.

sruthi81, 2025-01-15:
[image]
The gnb started but frame size issue is there how to resolve.

pgawlowicz, 2025-01-15:
could you share your config? Which OS version do you use?

sruthi81, 2025-01-20:
amf:
  addr: 127.0.0.5                 # The address or hostname of the AMF.
  bind_addr: 127.0.0.5             # A local IP that the gNB binds to for traffic from the AMF.

ru_sdr:
  device_driver: uhd                # The RF driver name.
  device_args: addr=192.168.10.2 #tx_port=tcp://127.0.0.1:2000,rx_port=tcp://127.0.0.1:2001,base_srate=23.04e6 # Optionally pass arguments to the selected RF driver.
  #sync:  gpsdo                   # Specify the sync source used by the RF.
#device_args: tx_port=tcp://192.168.0.105:2000,rx_port=tcp://192.168.0.114:2001,base_srate=23.04e6 # Optionally pass arguments to the selected RF driver.
  clock: internal  
  srate: 23.04                      # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 30                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 30                       # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 368500                  # ARFCN of the downlink carrier (center frequency).
  band: 3                           # The NR band.
  channel_bandwidth_MHz: 20         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 15                    # Subcarrier spacing in kHz used for data.
  plmn: "00101"                     # PLMN broadcasted by the gNB.
  tac: 7                            # Tracking area code (needs to match the core configuration).
  pdcch:
    common:
      ss0_index: 0                  # Set search space zero index to match srsUE capabilities
      coreset0_index: 12            # Set search CORESET Zero index to match srsUE capabilities
    dedicated:
      ss2_type: common              # Search Space type, has to be set to common
      dci_format_0_1_and_1_1: false # Set correct DCI format (fallback)
  prach:
    prach_config_index: 1           # Sets PRACH config to match what is expected by srsUE

cu_cp:
  inactivity_timer: 7200            # Sets the UE/PDU Session/DRB inactivity timer to 7200 seconds. Supported: [1 - 7200].

log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: info                   # Logging level applied to all layers.
  hex_max_size: 0

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.

sruthi81, 2025-01-20:
and I am using Ubuntu 22.04.5 LTS

sruthi81, 2025-01-20:
please help me to resolve this issue

sruthi81, 2025-02-06:
please help me to resolve the issue

newt2105, 2025-04-03:
hi @sruthi81, did you solve this problem?

sruthi81, 2025-04-04:
Yes the internet issue solved but getting the underflow overflow issue and not able to trace the gnb. please help me to resolve the issue.

newt2105, 2025-04-05:
hi @sruthi81, I have same problem but haven't fixed yet