srsran/srsRAN_Project #972: srsUE not attaching to gNB (ZMQ)

antonioo420, 2024-12-05:
## Issue Description ##
I am running srsUE, srsRAN and the dockerized version of Open5GS, just like this [tutorial](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zeromq-based-setup). The issue is that srsUE is not attaching to the gNB, and in the Open5GS console i can see no messages from the UE.

On top of that srsUE is not creating the tun_srsue interface so i cannot add the route to the interface with this command

`sudo ip netns exec ue1 ip route add default via 10.45.1.1 dev tun_srsue`

## Setup Details ##
UE: srsUE
RAN: srsRAN
5GC: Open5GS dockerized version
OS: Ubuntu 22.04
RF: ZMQ

## Expected Behavior ##
I expect srsUE to attach to gNB.

## Actual Behaviour ##
srsUE does not attach and tun_srsue is not created

## Steps to reproduce the problem ##
The ones portrayed on this [tutorial](https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#zeromq-based-setup)

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]
Attached are the conf files

[ue_rf.zip](https://github.com/user-attachments/files/18024008/ue_rf.zip)

[gnb_zmq.zip](https://github.com/user-attachments/files/18024023/gnb_zmq.zip)

pgawlowicz, 2024-12-05:
srsUE uses config for 20MHz cell, while gnb uses config for 10MHz cell.

antonioo420, 2024-12-10:
After changing gnb bandwidth to 20Mhz now it works. I have made the changes you said in #350 .

Thank you!

closed 2024-12-10 by antonioo420