srsran/srsRAN_Project #973: N2: Connection to AMF on 127.0.1.100:38412 completed can't continue

mcmtwjyy, 2024-12-06:
## Issue Description ##
[Describe the issue in detail]
Follow the official example, I got the result as below:

N2: Connection to AMF on 127.0.1.100:38412 completed. The ubuntu stopped, without any further prompts.
I'm sure that firewall is down.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[What you expect to happen]

Cell pci=1, bw=20 MHz, 1T1R, dl_arfcn=627340 (n78), dl_freq=3410.1 MHz, dl_ssb_arfcn=626976, ul_freq=3410.1 MHz

N2: Connection to AMF on 127.0.1.100:38412 completed
==== gNB started ===
Type <t> to view help

I can't see the "==== gNB started ===" on the panel
## Actual Behaviour ##
[What happens instead e.g. error message]
[image]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

mcmtwjyy, 2024-12-06:
I'm not sure whether this is normal

mcmtwjyy, 2024-12-06:
tail -f /var/log/open5gs/amf.log

[image]

the problem maybe cannot find served TAI. How to deal with?

mcmtwjyy, 2024-12-06:
sorry, I have found the open5gs - amf.yaml to change the parameter of plmn_support: - plnm_id: mcc: 001 mnc: 01. Now the amf.log:
[image]

mcmtwjyy, 2024-12-06:
[image]
[image]

pgawlowicz, 2024-12-06:
you need to set the TAI correctly. 
Please use this tutorial as reference; https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html#gnb

mcmtwjyy, 2024-12-06:
> you need to set the TAI correctly. Please use this tutorial as reference; https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html#gnb

Thanks. I download the amf.yaml accorrding to the link, but when loading, it doesn't run.
[image]
[image]

pgawlowicz, 2024-12-06:
just update your own config according to the tutorial. it shows the parameters that need to be set correctly.

mcmtwjyy, 2024-12-06:
> just update your own config according to the tutorial. it shows the parameters that need to be set correctly.

Thanks. Maybe I have the same problem as the issue 'UE not registered to srsran gnb #896', which still dom't have a conclusion.