srsran/srsRAN_Project #977: Low End-to-End Bitrate with iperf3 Despite High Downlink Throughput CU/DU split option 2 deployment

Moussa-Guemdani, 2024-12-10:
## Issue Description
When measuring the end-to-end bitrate using **iperf3**, the reported value is significantly lower than expected. The downlink throughput reaches **555 Mbps**, but the end-to-end bitrate fluctuates around **130 Mbps**.

## Setup Details
- **SDR**: USRP X310  
- **Channel Bandwidth**: 80 MHz  
- **Sampling Rate (srate)**: 92.16 Msps  
- **Modulation**: QAM256  
- **MIMO Layers**: 2  

### Core Network Server (5GC)
- **Architecture**: 64-bit  
- **Memory**: 16 GiB  
- **CPU**: Intel® Core™ i7-8700 @ 3.20 GHz (6 Cores)  
- **NIC**: 10-Gigabit SFI/SFP+ (between core and CU)  

### CU/DU Server
- **Architecture**: 64-bit  
- **Memory**: 256 GiB  
- **CPU**: AMD Ryzen Threadripper PRO 7955WX (16 Cores)  
- **NIC**: 10-Gigabit SFI/SFP+ (between DU and USRP)  

## Expected Behavior
The end-to-end bitrate should align closely with the downlink throughput of **555 Mbps**.

## Steps to Reproduce
1. Measure downlink throughput using iperf3.  
2. Observe the end-to-end bitrate fluctuating around **130 Mbps**.

## Additional Information
- Interfaces and CPU utilization have been checked, and no bottlenecks were identified.  
- TCP window sizes have been increased:  
  ```bash
  sudo sysctl -w net.ipv4.tcp_rmem="4096 12582912 33554432"
  sudo sysctl -w net.ipv4.tcp_wmem="4096 12582912 33554432"
  ```

If any further details are needed, please let me know!

robertfalkenberg, 2025-01-03:
To start troubleshoot this issue you could start with UDP iperf3 `-u` test and gradually increase the bitrate `-b 100M`, `-b 200M`,... until you notice an increased loss rate at the iperf3 receiver.

Are you able to achieve higher UDP E2E throughput on the air interface with little "nok" rate above 200...230M ?

Moussa-Guemdani, 2025-01-05:
Thank you for your assistance!
Here are some of the results I’ve observed so far:

[image]

However, I occasionally experience a drop in air interface throughput, which seems to correlate with a drop in CQI:
[image]

Do you have any insights on what might be causing this behavior?