srsran/srsRAN_Project #979: Documentation update request for feature description about Transmission Modes / Number of Antennas supported

ohault, 2024-12-11:
As we can read on https://docs.srsran.com/projects/4g/en/latest/feature_list.html, there is a clear description of Transmission Modes with number of antennas supported.

Could the same be specified also for https://docs.srsran.com/projects/project/en/latest/general/source/2_features_and_roadmap.html#current-features

pgawlowicz, 2024-12-11:
By default, it operated with a single antenna (see [configuration reference](https://docs.srsran.com/projects/project/en/latest/user_manuals/source/config_ref.html))
```
cell_cfg:
  pci: 1                                # Required UINT (1). Sets the Physical Cell ID. Supported: [0-1007].
  sector_id: 0                          # Optional UINT (0). Sets the Sector ID (4-14 bits). This value is concatenated with the gNB Id to form the NR Cell Identity (NCI). If not specified, a unique value for the DU is automatically derived. Supported: [0 - 16383].
  dl_arfcn: 536020                      # Required UINT (536020). Sets the Downlink ARFCN.
  band: auto                            # Optional TEXT (auto). Sets the NR band being used for the cell. If not specified, will be set automatically based on ARFCN. Supported: all release 17 bands.
  common_scs: 15                        # Required UINT (15). Sets the subcarrier spacing in KHz to be used by the cell. Supported: [15, 30].
  channel_bandwidth_MHz: 20             # Required UINT (20). Sets the channel Bandwidth in MHz, the number of PRBs will be derived from this. Supported: [5, 10, 15, 20, 25, 30, 40, 50, 60, 70, 80, 90, 100].
  nof_antennas_dl: 1                    # Optional UINT (1). Sets the number of antennas for downlink transmission. Supported: [1, 2, 4].
  nof_antennas_ul: 1                    # Optional UINT (1). Sets the number of antennas for uplink transmission. Supported: [1, 2, 4].
  plmn: 00101                           # Required TEXT (00101). Sets the Public Land Mobile Network code. Format: 7-digit PLMN code containing MCC & MNC.
  tac: 7                                # Required UINT (7). Sets the Tracking Area Code.
```

If you want MIMO, you enable multiple antennas as in the example [mimo_usrp.yml](https://github.com/srsran/srsRAN_Project/blob/main/configs/mimo_usrp.yml)

ohault, 2024-12-12:
Hello @pgawlowicz , I have provided the URL for updating the documentation because on the srsRAN 4G project, it is clear and sharp, but it is not the case on srsRAN 5G project documentation.

e.g. nof_antennas_dl: 1 + nof_antennas_ul: 1 => ? total number of antennas = 2 ? But then how to configure a single antenna for both UL+DL ?

Can you please update the 5G documentation in regards to Transmission Modes / Number of Antennas supported ?

pgawlowicz, 2024-12-13:
The concept of Transmission Mode (TM), which was used in LTE, is not utilized in 5G NR (i.e., no TMs are specified in 5G).

When you set `nof_antennas_dl` and `nof_antennas_ul`, it informs the gNB about the number of antennas available for downlink (DL) and uplink (UL) transmissions, respectively. By default, a single physical antenna is shared for both DL and UL, so the values of nof_antennas_dl and nof_antennas_ul are summing up.

If multiple TX antennas are configured, the gNB will dynamically select between TX/RX diversity or MIMO based on the prevailing channel conditions and the capabilities of the UE.

ohault, 2024-12-13:
> When you configure `nof_antennas_dl` and `nof_antennas_ul`, it informs the gNB about the number of antennas available for downlink (DL) and uplink (UL) transmissions, respectively. By default, a single physical antenna is used for both DL and UL, meaning the values of `nof_antennas_dl` and `nof_antennas_ul`are not additive.
> 
> If multiple transmission antennas are configured, the gNB will automatically adapt, selecting either TX/RX diversity or MIMO based on the current channel conditions and the capabilities of the UE.

@pgawlowicz , thank you for your answer. 

I figure out that indeed, I should have been writing my first question as "Could the features documentation about the supported way of using 5G antennas be improved (Beamforming, MIMO, SISO, ...) and avoid any confusion using the word " the same" for shortcoming. My apology.

Can you plz update the documentation accordingly with your explanations above? It should help many newcomers.

pgawlowicz, 2024-12-13:
OK, thank you for your feedback. We will discuss it internally.

closed 2025-01-28 by brendan-mcauliffe