srsran/srsRAN_Project #986: X310 MIMO2x2 DL throughput does not reach expectation. Also similar problems for SISO UL throughput with larger bandwidth

Timez-zx, 2024-12-14:
## Issue Description ##
I use srsRAN with x310 USRP devices to test the MIMO 2x2 performance. I tried both 80 Mhz and 90 Mhz. However, the SISO DL throughput is around 90Mbps, the MIMO 2x2 performance is almost the same. The MCS for both setup is almost the same (27 for DL, 24 for UL). But when I try the 20 Mhz, the MIMO 2x2 's throughput doubles the SISO. I am not sure why?

Similarly, for the SISO, when I increase the bandwidth, the UL MCS just gradually decreases. From 40 Mhz to 90 Mhz, the UL MCS decreases from 28 to 20, leading to almost the same throughput performance. (The tx_gain at sender and rx_gain at receiver has been tuned to largest, but it does not work)

## Setup Details ##
UE: Amarisoft UE Box E
gNB: Ubuntu 22.04 with low-latency kernel and usrp-x310 (10G network to transfer data). Using UHD to communicate with Amarisoft UE (gNB and UE are connected by copper fiber)

## Expected Behavior ##
1. The DL throughput of MIMO 2x2 doubles the SISO
2. As the bandwidth increases, the UL MCS does not reduce a lot, bringing higher UL throughput

## Actual Behaviour ##
1. For 90 Mhz and 80 Mhz, even for 40 Mhz, MiMO's 2x2 performance is almost the same as SISO (range from 100Mbps to 60Mbps). Only 20 Mhz behaves expected results
2. As bandwidth inceases, the UL throughput also increases

## Steps to reproduce the problem ##
I have attached the configuration file I used to this issues
[mimo_usrp.txt](https://github.com/user-attachments/files/18134051/mimo_usrp.txt)
[gnb_rf_x310_tdd_n78_90mhz.txt](https://github.com/user-attachments/files/18134052/gnb_rf_x310_tdd_n78_90mhz.txt)
[gnb_rf_x310_tdd_n78_80mhz.txt](https://github.com/user-attachments/files/18134053/gnb_rf_x310_tdd_n78_80mhz.txt)
[gnb_rf_x310_tdd_n78_40mhz.txt](https://github.com/user-attachments/files/18134054/gnb_rf_x310_tdd_n78_40mhz.txt)
[gnb_rf_x310_tdd_n78_20mhz.txt](https://github.com/user-attachments/files/18134055/gnb_rf_x310_tdd_n78_20mhz.txt)

pgawlowicz, 2024-12-16:
Please check our new tutorial on [gNB performance turning](https://docs.srsran.com/projects/project/en/latest/tutorials/source/tuning/source/index.html).

Timez-zx, 2024-12-16:
> Please check our new tutorial on [gNB performance turning](https://docs.srsran.com/projects/project/en/latest/tutorials/source/tuning/source/index.html).

Thanks for your advise. I have tried that. But it does not work. For the DL MIMO throughput, I have found the bottleneck is related to the performance of UPF.

For now, I found some new issues related to UL MCS. Even if I tune the usrp rx_gain to maximum, the MCS can still not reach the maximum. I use a fiber to connect both amarisoft and srsRAN, which should be idea channel quality. 

I use usrpx310 with 80Mhz now, the UL MCS ranges from 17 to 24. Do you know what happens on this new issue?

pgawlowicz, 2024-12-16:
did you tune nof PHY threads? For example:
```
expert_execution:
  threads:
    upper_phy:
      pdsch_processor_type: concurrent
      nof_pusch_decoder_threads: 4
      nof_ul_threads: 2
      nof_dl_threads: 8
```

Timez-zx, 2024-12-16:
I have never tried to increase the thread before, so I just tried the configuration you have mentioned:
[image]
This is the log I got from srsRAN. The UL part mcs and signal strength are still lower than expected. It seems to be not related to cpu processing part? Because even I use default parameter, I can still get similar power strength.

pgawlowicz, 2024-12-16:
did you try tuning  `rx_gain`?

Timez-zx, 2024-12-16:
Yes. I use the maximum rx_gain: 37.5 dB.  For the sender side, I use tx_gain: 90 dB. They are both maximum gain I can adjust to.

Timez-zx, 2024-12-16:
I have change a little to adapt to my pc [srs_gnb.yml.txt](https://github.com/user-attachments/files/18157499/srs_gnb.yml.txt).

This is the result from the configuration you provided:
[image]

pgawlowicz, 2024-12-16:
could you share amariUE config?

Timez-zx, 2024-12-16:
Yes. 
This is the main configuration:
[ue-nr-sa-rb273-srs-x310.txt](https://github.com/user-attachments/files/18157552/ue-nr-sa-rb273-srs-x310.txt)

This is for controlling the gain:
[1chan-rb273-srs-x310.txt](https://github.com/user-attachments/files/18157557/1chan-rb273-srs-x310.txt)

pgawlowicz, 2024-12-16:
does it have any impact if you reduce tx gain in amariUE?

Timez-zx, 2024-12-16:
Yes. The MCS will reduce a lot, from 24 to 17 if I reduce tx_gain from 90 to 80.

Timez-zx, 2024-12-16:
RF cable. We try to avoid any attenuation at first, so only use RF cable. I also try to change some cables, but cannot solve this problem.

pgawlowicz, 2024-12-16:
could you try to add an attenuator? like 10-20db?

Timez-zx, 2024-12-16:
We do not have an attenuator to be installed. What's the attenuator for?

pgawlowicz, 2024-12-16:
I would be afraid of killing the x310 if connecting it without attenuator.

The UE seems to have a lot of power headroom, you might try to push it to transmit with a bit higher power by tuning those parameters:
```
pusch:
    target_sinr: 10                         # Optional FLOAT (10). Sets the target SINR in dB. Supported: [-5 - 30].
    ref_path_loss: 70                       # Optional FLOAT (70). Sets the reference path-loss for target PUSCH SINR in dB. Supported: [50 - 120].
    pl_compensation_factor: 1               # Optional FLOAT (1). Sets the fractional path-loss compensation factor in PUSCH power control. Supported: [0.4, 0.5, 0.6, 0.7, 0.8, 0.9, 1].
```

Timez-zx, 2024-12-16:
I have tuned this parameters, but seems that it does work well, the mcs just stuck at 24:

  pusch:
      target_sinr: 25 
      ref_path_loss: 80                    
      pl_compensation_factor: 1 
[image]
This result is based on my 80Mhz configuration.

pgawlowicz, 2024-12-16:
if you do pl_compensation_factor: 0.5?

Timez-zx, 2024-12-16:
pusch:
      target_sinr: 25 
      ref_path_loss: 80                    
      pl_compensation_factor: 0.5
[image]
Almost the same result

pgawlowicz, 2024-12-16:
what is the RSRP on the UE side? maybe you can decrease gnb tx_gain, as the DL channel is really good.

Timez-zx, 2024-12-16:
[image]
This is the report from UE side.

Timez-zx, 2024-12-16:
I reduce the gnb tx_gain from 31.5 to 20:

This is the new log from gNB:
[image]

This is the new log from UE:
[image]

pgawlowicz, 2024-12-16:
could you reduce to gnb tx_gain to 5 and see the impact? 
Also there is a huge CFO, do you have access to an external clock?

Timez-zx, 2024-12-17:
I have changed the tx_gain at gNB to 5. At the same time, I do not have external clock, but I have set that manually from UE side:

This is gNB's result:
[image]

This is amarisoft side result:
[image]

Timez-zx, 2024-12-18:
Our RF cable seems to be the issue. After changing the RF cable to antenna, even if our amarisoft UE box still needs to use the maximum tx_gain, the srsRAN does not need to use the maximum rx_gain.

closed 2024-12-18 by Timez-zx