srsran/srsRAN_Project #988: 4x4 MIMO 100 MHz CPU Optimization for L/U/O Package

osmanmerty, 2024-12-16:
## Setup Details ##

- **gNodeB SDR:** USRP X410 with QSFP-100G
- **gNodeB Antenna:** [Cross-Polarized MIMO Antenna](https://www.taoglas.com/product/guardian-8in1-multi-mount-1multi-band-gnss-45g-4g-mimo-and-3wi-fi-mimo/)
- **UE:** Samsung S23
- **Core Network:** Open5GS
- **gNodeB Host _lscpu_:** 

```
Architecture:                    x86_64
CPU op-mode(s):                  32-bit, 64-bit
Address sizes:                   46 bits physical, 57 bits virtual
Byte Order:                      Little Endian
CPU(s):                          48
On-line CPU(s) list:             0-47
Vendor ID:                       GenuineIntel
Model name:                      Intel(R) Xeon(R) Gold 6442Y
CPU family:                      6
Model:                           143
Thread(s) per core:              2
Core(s) per socket:              24
Socket(s):                       1
Stepping:                        8
Frequency boost:                 enabled
CPU max MHz:                     2601.0000
CPU min MHz:                     800.0000
BogoMIPS:                        5200.00
```
**Note:** CPU turbo frequency is 4 GHz. 

## Issue Description ##
Hello srsRAN Team,

We are trying to realize MIMO with srsRAN Project. So far we have been able to successfully use the default configuration with 100 MHz bandwidth for 2x2 MIMO. However, for 4x4 MIMO, we have only been able to use up to 50 MHz bandwidth without underflow or late packages. When we try to use 100 MHz with 4x4 MIMO, despite configuring our system so that its performance is adequate, we still encounter some underflows during the iperf test (occasionally we receive late packets as well, but these are much rarer).

We did the recommended configurations in the srsRAN performance script and those [recommended by Ettus](https://kb.ettus.com/Getting_Started_with_4G_LTE_using_Eurecom_OpenAirInterface_(OAI)_on_the_USRP_2974). Also, we isolated CPUs 1-40 (hyper-threading is disabled) and allocated them in the gNB configuration as follows (we also tried allocating threads in the conf but that didn't make a significant change):

```
expert_execution:
  cell_affinities:                      
    -
      l1_dl_cpus:       1,2,3,4,5,6,7,8,9,10,11,12,13
      l1_ul_cpus:       31,32,33,34,35                                    
      l2_cell_cpus:     14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30                       
  affinities:
    low_priority_cpus:  36,37,38,39,40                                           
  threads:                               
    lower_phy:
     execution_profile: quad  
```

Isolating and allocating the CPUs like this helped significantly reduce latepackets/underflows, but we're still getting some during iperf test (`iperf3 -c 10.45.1.2 -i 1 -t 60 -u -b 1000M`), as seen below. Also, from the system monitor, we noticed that not all but only a few of the CPUs we allocated were being used. Our run command is:

`sudo taskset -c 1,2,3,4,5,6,7,8,9,10,11,12,13,14,15,16,17,18,19,20,21,22,23,24,25,26,27,28,29,30,31,32,33,34,35,36,37,38,39,40 ./gnb -c ../../../configs/gnb_x410.yml `

[image]

If possible, how can we configure the system and allocate the CPUs so that they are all used and the underflows we encounter are prevented? More generally, how can we determine the optimal way of pinning CPUs in the gNB conf file and what other things can we do to improve the system's performance?

Additionally, sometimes the RI fed back by the user doesn't go up to 4. Is this related to a physical issue with the setup?

Thanks

pgawlowicz, 2024-12-16:
Our new tutorial on the performance tuning might help: https://docs.srsran.com/projects/project/en/latest/tutorials/source/tuning/source/index.html

pgawlowicz, 2024-12-19:
@ninjab3s could you comment here?

dominikheinz, 2025-04-01:
@osmanmerty Did you manage to solve this? I have the same issue, and with the tuned profile the performance is even worse.
I have a bunch of underflow's and late packets, eventually resulting in `[rx_flow_ctrl_state] Flow control state mismatch`.
This essentially renders the system unusable after a couple of minutes.

I have a 13900H (Miniforum ms-01 PC). So shouldnt be the hardware. And if I understand correctly, underflow means the system is in a waiting state/the queue is empty?