srsran/srsRAN_Project #992: UE is not attached to gnb

shirinebadi, 2024-12-17:
## Issue Description ##
UE stuks in attaching to gnb.

## Setup Details ##
My setup includes srsUE, srsRAN, and dockerized 5gc interfaced via ZMQ. I run srsUE on QEMU(on my laptop) and srsRAN and 5g core are running on my laptop.

## Expected Behavior ##
srsUE attaches to gnb.

## Actual Behaviour ##
srsUE freezes in attaching to gnb. I believe ZMQ connection is setup, but there might be a timing problem in decoding the packets. I attached  configurations  and logs.
[ue.zip](https://github.com/user-attachments/files/18171050/ue.zip)

[gnb.zip](https://github.com/user-attachments/files/18171053/gnb.zip)

[ue-log.zip](https://github.com/user-attachments/files/18171070/ue-log.zip)
[gnb-log.zip](https://github.com/user-attachments/files/18171080/gnb-log.zip)

pgawlowicz, 2024-12-17:
Please check the IP addresses used in zmq

shirinebadi, 2024-12-17:
Thanks for your reply.
10.0.0.160 is my host private ip that Qemu can ping it. I also port forwarded 2002 from Qemu to host, so if I start a server in Qemu in port 2000, host can connect to it by `nc 0.0.0.0 2002` command. I did these tests to verify ips are accessible.

pgawlowicz, 2024-12-17:
with `srate=23.04e6`, you to not need this in your ue config:
```
[expert]
lte_sample_rates = true
```

shirinebadi, 2024-12-17:
I removed this line. I think UE is attached successfully:
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=334
Random Access Complete.     c-rnti=0x4601, ta=0
RRC Connected
Failed to setup/configure GW interface
RRC NR reconfiguration successful.

pgawlowicz, 2024-12-17:
change ue config:
```
[gw]
#netns = ue1
ip_devname = tun_srsue
ip_netmask = 255.255.255.0
```

and do you run it with sudo? It will create a tun interface.

shirinebadi, 2024-12-17:
I'm not running it with sudo, since I'm using buildroot.

pgawlowicz, 2024-12-17:
yes, the UE is attached.

shirinebadi, 2024-12-17:
Thanks, but why it is not assigning an ip to it?

pgawlowicz, 2024-12-17:
the core assigned an IP:
```
open5gs_5gc  | 12/17 22:43:11.405: [upf] INFO: UE F-SEID[UP:0x86 CP:0x54b] APN[srsapn] PDN-Type[1] IPv4[10.45.1.2] IPv6[] (../src/upf/context.c:485)
open5gs_5gc  | 12/17 22:43:11.405: [upf] INFO: UE F-SEID[UP:0x86 CP:0x54b] APN[srsapn] PDN-Type[1] IPv4[10.45.1.2] IPv6[] (../src/upf/context.c:485)
```

the UE cannot set it, as it cannot create the tun interface.

closed 2024-12-17 by shirinebadi