srsran/srsRAN_Project #1018: ue dissconnected after less than 1 minutes

faratell, 2025-01-18:
Hello.
I use a PC with an Intel(R) Core(TM) i7-7700K CPU running at 4.20GHz and a USRP B210 to create an SRSRAN gNB with the IP address 192.168.75.58. For the 5G core, I utilize Open5GS in another machine  with the IP address 192.168.75.158, and run components NRF, SCP, AMF, SMF, UPF, AUSF, UDM, PCF, NSSF, UDR, and BSF, all configured. My gNB is successfully connected to Open5GS. I am using an Honor X9B phone that supports 5G connectivity, along with an Oyeitimes 4G SIM card. The SIM card connects well and provides internet access, with a 5G signal displayed on my phone. However, even though the User Equipment (UE) remains stationary and does not move, it disconnects after approximately one minute. I have attached the pcap, NGAP, and log files for further analysis. I also discussed this issue on GitHub at the following link: https://github.com/srsran/srsRAN_Project/issues/865#issuecomment-2548645444 but have not yet received final solutions to resolve it.
[gnb_Conf.txt](https://github.com/user-attachments/files/18463628/gnb_Conf.txt)

[amf.txt](https://github.com/user-attachments/files/18463527/amf.txt)

[gnb.zip](https://github.com/user-attachments/files/18463547/gnb.zip)

[image]

[image]

pgawlowicz, 2025-01-20:
could you share your open5gs config?

faratell, 2025-01-21:
[upf.txt](https://github.com/user-attachments/files/18485767/upf.txt)

[amf.txt](https://github.com/user-attachments/files/18485780/amf.txt)

faratell, 2025-02-01:
> could you share your open5gs config?

have you any idea about my problem?

faratell, 2025-02-03:
> could you increase log level in gnb and send the new logs?

[gnb_DEBUG.txt](https://github.com/user-attachments/files/18640381/gnb_DEBUG.txt)

pgawlowicz, 2025-02-12:
this is only console trace. Could you send gnb.log file?