srsran/srsRAN_Project #1019: Assertion Failure in rlc_tx_am_entity when Connecting OAI CU to srsDU with COTS UE (Open5GS)

Moussa-Guemdani, 2025-01-20:
Hello srs team,

I am trying to connect OAI CU with srsDU. The connection between the CU and DU is successfully established, but when I attempt to connect a COTS UE, I encounter the following assertion failure:

~/srsRAN_Project/lib/rlc/rlc_tx_am_entity.cpp:75:
srsran::rlc_tx_am_entity::rlc_tx_am_entity(srsran::gnb_du_id_t, srsran::du_ue_index_t,
srsran::rb_id_t, const srsran::rlc_tx_am_config&, srsran::rlc_tx_upper_layer_data_notifier&,
srsran::rlc_tx_upper_layer_control_notifier&, srsran::rlc_tx_lower_layer_notifier&,
srsran::rlc_metrics_aggregator&, srsran::rlc_pcap&, srsran::task_executor&,
srsran::task_executor&, srsran::timer_manager&): Assertion
`config.pdcp_sn_len == pdcp_sn_size::size12bits || config.pdcp_sn_len == pdcp_sn_size::size18bits' failed
- Cannot create RLC TX AM, unsupported pdcp_sn_len=invalid. du=0 ue=1 DRB1.

Steps to Reproduce

    Setup OAI CU and srsDU.
    Ensure the connection between CU and DU is established.
    Attempt to connect a COTS UE.
    Observe the error in rlc_tx_am_entity.cpp.

Expected Behavior

The COTS UE should attach successfully without triggering an assertion failure.
Observed Behavior

The assertion failure occurs due to an unsupported PDCP SN length (pdcp_sn_len=invalid), which causes the srsDU process to crash.

OAI CU Configuration File:
```
gnb-cu.sa.f1.test.with.srsDU.conf

Active_gNBs = ( "cu-rfsim");
Asn1_verbosity = "none";
Num_Threads_PUSCH = 8;

gNBs =
(
 {
    gNB_ID = 0xe00;
    gNB_name  =  "cu-rfsim";
    tracking_area_code  =  1;
    plmn_list = ({ mcc = 001; mnc = 01; mnc_length = 2; snssaiList = ({ sst = 1; }) });
    nr_cellid = 12345678L;

    tr_s_preference = "f1";

    local_s_if_name     = "lo";
    local_s_address     = "127.0.10.1";
    remote_s_address    = "127.0.10.2";
    #local_s_portc       = 38472;
    #local_s_portd       = 38472;
    #remote_s_portc      = 38472;
    #remote_s_portd      = 38472;

    amf_ip_address = ({ ipv4 = "192.168.1.41"; });

    NETWORK_INTERFACES :
    {
        GNB_IPV4_ADDRESS_FOR_NG_AMF              = "192.168.1.40/24";
        GNB_IPV4_ADDRESS_FOR_NGU                 = "192.168.1.40/24";
        GNB_PORT_FOR_S1U                         = 2152;
    };
  }
);

security = {
  ciphering_algorithms = ( "nea0" );
  integrity_algorithms = ( "nia2", "nia0" );
  drb_ciphering = "yes";
  drb_integrity = "no";
};

```
srsDU Configuration:
                                                                        
```
f1ap:                               
  cu_cp_addr: 192.168.1.41 # 127.0.10.1  (F1_C)              # Required TEXT (127.0.10.1).>
  bind_addr: 192.168.1.40 #127.0.10.2                 # Required TEXT (127.0.10.2). >

nru:                                
  #queue_size: 40960                      # Optional INT (2048). Sets the F1-U PDU que>
  bind_addr: 192.168.1.40 #127.0.10.2       

cell_cfg:
  dl_arfcn: 663334
  band: 77
  channel_bandwidth_MHz: 60
  common_scs: 30
  plmn: "00101"
  tac: 1
  pci: 1

  tdd_ul_dl_cfg:  
    dl_ul_tx_period: 10                     # Optional INT (10). Sets the TDD pattern periodicity in slots. The combination of this value and the chosen numerology must lead to a >
    nof_dl_slots: 6                         # Optional INT (6). Number of consecutive full Downlink slots. Supported: [0-80].
    nof_dl_symbols: 10                       # Optional INT (8). Number of Downlink symbols at the beginning of the slot following full Downlink slots. Supported: [0-13].
    nof_ul_slots: 3                         # Optional INT (3). Number of consecutive full Uplink slots. Supported: [0 - 80].
    nof_ul_symbols: 2                       # Optional INT (0). Number of Uplink symbols at the end of the slot preceding the first full Uplink slot. Supported: [0-13].
    

ru_sdr:
  device_driver: uhd
  device_args: type=x300, addr=192.168.30.2, num_send_frames=32, num_recv_frames=32, send_frame_size=8000, recv_frame_size=8000, stream_args=replay_buffered #, send_buff_size=8160>
  srate: 92.16 #122.88 #184.32  #184.32 #184.32 #122.88  #122.88 #92.16  #92.16 #61.4 , 184.32 MHz
  otw_format: default #sc16 ##default
  clock: default
  sync: default
  tx_gain: 31.5
  rx_gain: 31.5
  time_alignment_calibration: -170
 

metrics:
  enable_json_metrics: true
  autostart_stdout_metrics: true
```

alvasMan, 2025-01-20:
Hi -- 

Could you attach the following:

1. An F1AP pcap between the CU and DU
2. The DU log. F1AP at info and RLC at info too.
3. The commit hash of the version of srsRAN_Project that you are using.

Seems like for whatever reason the RLC does not have PDCP sequence number length information.

Moussa-Guemdani, 2025-01-21:
Hello,

1. f1ap pcap
[gnb_f1ap.txt](https://github.com/user-attachments/files/18483628/gnb_f1ap.txt)

2. f1ap and rlc ate info:
[du.log](https://github.com/user-attachments/files/18483615/du.log)

3. commit hach:
commit 9d5dd742a70e82c0813c34f57982f9507f1b6d5d (HEAD -> main, tag: release_24_10, origin/main, origin/HEAD)

alvasMan, 2025-01-21:
As you can see in in frame number 20 of the attached pcap (UE Context Setup), there is no information about PDCP SN length in the DRB to Setup item. This is required by the spec for the F1AP UE context setup, see table 9.2.2.1 of TS 38.473 (the F1AP spec).

[image]

The PDCP length is used by the DU's RLC layer to provide the services and functions expected by the PDCP, such as RLC SDU discard and indication of successful delivery of upper layers PDUs (see section 4.3 and 4.4 of TS 38.322). So, it's not really an optional requirement but a mandatory one, as far as I can see.

That said, the DU should have rejected the request rather then asserting. I will provide a fix that rejects that request.
For getting an attach going, I believe you (or OAI) will need to fix the CU to provide the correct PDCP SN information.

Keilwerth11270, 2025-01-23:
To summarize: 

1. OAI CU needs to include PDCP SN length in F1AP UE Context Setup per TS 38.473
2. srsDU needs to properly reject (rather than assert) when this information is missing

Based on page 64 of the [spec](https://www.etsi.org/deliver/etsi_ts/138400_138499/138473/15.08.00_60/ts_138473v150800p.pdf) this particular parameter is set in the DRB to Be Setup Item IEs. The corresponding function in the OAI code seems to be found [here](https://github.com/OPENAIRINTERFACE/openairinterface5g/blob/develop/openair2/F1AP/f1ap_cu_ue_context_management.c#L413).

As @alvasMan correctly pointed out... that DL PDCP SN Length setter is missing. Based on the table, that looks to be the last item in the DRB setup. You could try making changes locally to implement that. The overall setup function is following the order according to the table, so it would be added after [this section](https://github.com/OPENAIRINTERFACE/openairinterface5g/blob/develop/openair2/F1AP/f1ap_cu_ue_context_management.c#L502) in the code if you wanted to follow that convention.

[Here](https://www.etsi.org/deliver/etsi_ts/138300_138399/138323/16.02.00_60/ts_138323v160200p.pdf) I am checking out the specification for PDCP SN in 5G and it looks like based on pages 28 and 29, SRBs will always have 12 bit length for SN while DRBs can have 12 or 18. [This website](https://www.sharetechnote.com/html/5G/5G_PDCP.html) is useful if you want to read more about it. 

If you do successfully resolve this on your local machine, please let us know by updating this thread!!

alvasMan, 2025-01-28:
We reject now F1AP UE context setups with invalid DRBs. It will be present in the next release. Please re-open if you still have issues/the assert.

closed 2025-01-28 by alvasMan