srsran/srsRAN_Project #1035: srsRAN with LiteON RU - Missed incoming User-Plane messages

bharani1990, 2025-02-04:
We are trying to setup a test bed with the following components,
	. LITEON FlexFi (firmware version- 02.00.10)
	. srsRAN Project
	. FibroLAN Falcon switch
	. Intel NIC - E810XXVDA2
	. OS Version - Ubuntu 22.04
	. Core - Open5gs
In our setup, server running CU/DU and RU are connected via the FibroLAN switch. The devices are synchronized via PTP. Note that we are using the same interface for both PTP Sync and C/U Plane traffic.

Please find the screenshots of PTP processes in the CU/DU server below,
[image]
[image]

Upon running the gNB software, we could see the following message in the log file (please find it attached), 
	2025-02-04T12:48:17.730545 [OFH     ] [W] Sector#0: missed incoming User-Plane PRACH messages for slot '500.19' and sector#0

We have crosschecked the configurations (gNB config attached) in the RU. It would be really helpful if you could help us in resolving this issue. Also it would be great if you could comment on our infrastructure setup, especially the fact that we are using the same interface for PTP sync and C/U Plane traffic.

[gnb_log.zip](https://github.com/user-attachments/files/18659245/gnb_log.zip)

[gnb_ru_liteon_tdd_n78_100mhz.yml.txt](https://github.com/user-attachments/files/18659258/gnb_ru_liteon_tdd_n78_100mhz.yml.txt)

additionally, attaching the screenshot which describes the oru_status before and after the gnb start. (Note: duConnected: NotReady --> in both cases)

[image]

bharani1990, 2025-02-20:
hallo, all - can you please help me?

satiskumar0597, 2025-02-26:
We are experiencing the same issue as well. Hopefully, someone can help.

mooncake-22, 2025-03-12:
From my experience, the `DuConnected` status will turn to `Ready` when a UE successfully attached.