srsran/srsRAN_Project #1045: srsRAN ERROR: CU-CP failed to connect to AMF

sruthi81, 2025-02-12:
--== srsRAN gNB (commit 4cf7513e9) ==--

The PRACH detector will not meet the performance requirements with the configuration {Format 0, ZCZ 0, SCS 1.25kHz, Rx ports 1}.
Lower PHY in quad executor mode.
N2: Failed to connect to AMF on 127.0.0.5:38412. error="Connection refused" timeout=0ms
srsRAN ERROR: CU-CP failed to connect to AMF

I had installed open5gs by using docker. /etc/open5gs file contains all .yaml file. Please help me to resolve this issue. 

[AMF.txt](https://github.com/user-attachments/files/18762986/AMF.txt)
[gnb.txt](https://github.com/user-attachments/files/18762985/gnb.txt)

[image]

pgawlowicz, 2025-02-12:
Hi, did you follow one of our tutorials? for example:
https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html

sruthi81, 2025-02-12:
Yes Sir. Last two weeks before it connected very well. I have updated my amf.yaml.

pgawlowicz, 2025-02-12:
Could you use the old amf.yaml config? could you put here ngap.pcap?

sruthi81, 2025-02-13:
ngap.pcap file is empty
What would be the meaning?

sruthi81, 2025-02-13:
Old amf configuration missed and this is the amf status

Feb 13 09:27:42 iisst systemd[1]: open5gs-amfd.service: Scheduled restart job, >
Feb 13 09:27:42 iisst systemd[1]: Stopped Open5GS AMF Daemon.
Feb 13 09:27:42 iisst systemd[1]: open5gs-amfd.service: Start request repeated >
Feb 13 09:27:42 iisst systemd[1]: open5gs-amfd.service: Failed with result 'cor>
Feb 13 09:27:42 iisst systemd[1]: Failed to start Open5GS AMF Daemon.
lines 1-12/12 (END)...skipping...
× open5gs-amfd.service - Open5GS AMF Daemon
     Loaded: loaded (/lib/systemd/system/open5gs-amfd.service; enabled; vendor preset: enabled)
     Active: failed (Result: core-dump) since Thu 2025-02-13 09:27:42 IST; 11min ago
    Process: 1262155 ExecStart=/usr/bin/open5gs-amfd -c /etc/open5gs/amf.yaml (code=dumped, signal=ABRT)
   Main PID: 1262155 (code=dumped, signal=ABRT)
        CPU: 13ms

Feb 13 09:27:42 iisst systemd[1]: open5gs-amfd.service: Scheduled restart job, restart counter is at 27026.
Feb 13 09:27:42 iisst systemd[1]: Stopped Open5GS AMF Daemon.
Feb 13 09:27:42 iisst systemd[1]: open5gs-amfd.service: Start request repeated too quickly.
Feb 13 09:27:42 iisst systemd[1]: open5gs-amfd.service: Failed with result 'core-dump'.
Feb 13 09:27:42 iisst systemd[1]: Failed to start Open5GS AMF Daemon.

Can I get the old amf details?