srsran/srsRAN_Project #1047: CU cant connect to the NearRT-RIC

Gmclaro, 2025-02-12:
Hello
I have this problems that when enable both of this fields
```
e2:
  enable_cu_cp_e2: true # Optional BOOLEAN (false). Enables the CU-CP E2 agent. Supported: [false, true].
  enable_cu_up_e2: true # Optional BOOLEAN (false). Enables the CU-UP E2 agent. Supported: [false, true].

```

in the CU configuration. When i run the config this error appears

```
--== srsRAN CU (commit cc2869f96) ==--

terminate called after throwing an instance of 'std::out_of_range'
  what():  _Map_base::at
Aborted
```

If the fields are both disabled the cu does not connect to the SC-RIC
with this output

´´´
--== srsRAN CU (commit cc2869f96) ==--

N2: Connection to AMF on 10.0.23.62:38412 completed
F1-C: Listening for new connections on 127.0.10.1:38472...
==== CU started ===
Type <h> to view help

´´´

This was run with a CU-DU split connected to a USRP, the DU connects successfully to the RIC and CU.

pgawlowicz, 2025-02-13:
Please update to the newest project version.

Gmclaro, 2025-02-13:
Hi 
I noticed yesterday that you updated somethings in the repository, and i pulled the updates
But a new errors have appeared
I didn't change the configs
Now when i enable cu_cp and cu_up this pops up

```
--== srsRAN CU (commit a041e3162) ==--
N2: Connection to AMF on 10.0.23.62:38412 completed
E2AP: Failed to connect to RIC on 10.0.2.10:36421. error="Connection refused" timeout=0ms
F1-C: Listening for new connections on 127.0.10.1:38472...
Segmentation fault
```

When i disable them and try to connect the du to it this shows up

```
--== srsRAN DU (commit a041e3162) ==--

INI was not able to parse nru.++
Run with --help for more information.
```

the cu and du configuration file are as follows

```
cu_cp:
  amf:
    addr: 10.0.23.62 
    bind_addr: 10.0.22.176 
    supported_tracking_areas: 
      - tac: 1
        plmn_list:
          - plmn: "00101"
            tai_slice_support_list:
              - sst: 1
  f1ap:
    bind_addr: 127.0.10.1

e2:
  enable_cu_cp_e2: true
  enable_cu_up_e2: false 
  addr: 10.0.2.10 
  port: 36421 
  bind_addr: 10.0.22.176 
  sctp_init_max_attempts: 3
  e2sm_kpm_enabled: true
  e2sm_rc_enabled: true 

```
f1ap:
  cu_cp_addr: 127.0.10.1
  bind_addr: 127.0.10.2
nru:
  bind_addr: 127.0.10.2

ru_sdr:
  srate: 23.04                                                   
  device_driver: uhd                                             
  device_args: type=b200,num_recv_frames=64,num_send_frames=64  
  otw_format: sc12
  tx_gain: 80
  rx_gain: 40

e2: 
  enable_du_e2: true         
  addr: 10.0.2.10                 
  port: 36421                  # Optional UINT (36421). Sets the RIC Port. Supported: [20000 - 40000].
  bind_addr: 10.0.22.176                  # Optional TEXT (127.0.0.1). Sets the local IP address to bind for RIC connection. 
  sctp_rto_initial: 120                 # Optional INT (120). Sets the initial retransmission timeout when creating the SCTP 
  sctp_rto_min: 120                     # Optional INT (120). Sets the minimum retransmission timeout for the SCTP connection.
  sctp_rto_max: 500                     # Optional INT (500). Sets the maximum retransmission timeout for the SCTP connection.
  sctp_init_max_attempts: 3          # Optional INT (3). Sets the maximum retransmission attempts for the initial SCTP 
  sctp_max_init_timeo: 500              # Optional INT (500). Sets the maximum retransmission timeout for the initial SCTP 
  e2sm_kpm_enabled: true           # Optional BOOLEAN (false). Enabled the E2SM KPM service module. 
  e2sm_rc_enabled: true               # Optional BOOLEAN (false). Enabled the E2SM RC service module. Supported: [false, true].

cell_cfg:
  dl_arfcn: 650000
  band: 78
  channel_bandwidth_MHz: 20
  common_scs: 30
  plmn: "00101"
  tac: 1
  pci: 1

```

Gmclaro, 2025-02-24:
Hi
It's working know 
I updated the project and changed the configurations to latest versions
I'm going to close the issue now

closed 2025-02-24 by Gmclaro

Gmclaro, 2025-03-05:
`

> > Hi It's working know I updated the project and changed the configurations to latest versions I'm going to close the issue now
> 
> Hi [Gmclaro](https://github.com/Gmclaro), can you share the configuration of CU for me, i have the same issue like you but i have not fixed yet

Hi @newt2105 
My CU  is like this at the moment, 

```yaml

gnb_id: 411 # Optional UINT (411). Sets the numerical ID associated with the gNB.
gnb_id_bit_length: 22 # Optional UNIT. Sets the bit length of the gnb_id above. Format: integer between [22 - 32]

cu_cp:
  f1ap:
    bind_addr:
      10.0.23.73  # Configure the F1AP bind address, this will enable the CU-cp to connect to the DU
  amf:
    addr: 10.0.23.62 # The address or hostname of the AMF.
    bind_addr: 10.0.23.73 # A local IP that the gNB binds to for traffic from the AMF.
    supported_tracking_areas: # Configure the TA associated with the CU-CP
      - tac: 1
        plmn_list:
          - plmn: "00101"
            tai_slice_support_list:
              - sst: 1

e2:
  enable_cu_cp_e2: true                # Optional BOOLEAN (false). Enables the CU-CP E2 agent. Supported: [false, true]. 
  enable_cu_up_e2: true                # Optional BOOLEAN (false). Enables the CU-UP E2 agent. Supported: [false, true]. 
  addr: 10.0.23.52                     # Optional TEXT (127.0.0.1). Sets the RIC IP address.
  port: 36421                           # Optional UINT (36421). Sets the RIC Port. Supported: [20000 - 40000].
  bind_addr: 10.0.23.73                # Optional TEXT (127.0.0.1). Sets the local IP address to bind for RIC connection. Supported: IPv4 address.
  e2sm_kpm_enabled: true               # Optional BOOLEAN (false). Enabled the E2SM KPM service module. Supported: [false, true].
  e2sm_rc_enabled: true                # Optional BOOLEAN (false). Enabled the E2SM RC service module. Supported: [false, true].

cu_up:
  f1u:
    socket: # Define UDP/IP socket(s) for F1-U interface.
      - # Socket 1
        bind_addr: 10.0.23.73 # Required TEXT. Sets the address that the F1-U socket will bind to.
        udp:
          max_rx_msgs: 256    # maximum packets read from the socket in a single syscall.
          pool_threshold: 0.9 # Pool occupancy threshold, after which packets are dropped.

  ngu:
    socket:
      - bind_addr: 10.0.22.217

metrics: 
  cu_cp_statistics_report_period: 1     # Optional UINT (1). Sets the CU-CP statistics report period in seconds. Set this value to 0 to disable this feature. Supported: [0 - inf]. 
  cu_up_statistics_report_period: 1     # Optional UINT (1). Sets the CU-UP statistics report period in seconds. Set this value to 0 to disable this feature. Supported: [0 - inf]. 
  pdcp_report_period: 1000                 # Optional UINT (0). Sets the PDCP metrics report period. 

```

newt2105, 2025-03-05:
Hi [Gmclaro](https://github.com/Gmclaro),
Thanks for CU configuration, but i'm still not connected CU with RIC yet. 

Here is log of CU:
```
ubuntu@cu:~/srsRAN_Project/build/apps/cu$ sudo ./srscu -c cu.yml

--== srsRAN CU (commit 8120ffbce) ==--

N2: Connection to AMF on 10.0.10.96:38412 completed
E2AP: Failed to connect to RIC on 10.0.10.73:36421. error="Connection timed out" timeout=340896ms
F1-C: Listening for new connections on 10.0.10.36:38472...
Segmentation fault
```  
here is my CU config:
```
cu_cp:
  f1ap:
    bind_addr:
      10.0.10.36  # Configure the F1AP bind address, this will enable the CU-cp to connect to the DU
  amf:
    addr: 10.0.10.96 # The address or hostname of the AMF.
    bind_addr: 10.0.10.36 # A local IP that the gNB binds to for traffic from the AMF.
    supported_tracking_areas: # Configure the TA associated with the CU-CP
      - tac: 1
        plmn_list:
          - plmn: "00101"
            tai_slice_support_list:
              - sst: 1

e2:
  enable_cu_cp_e2: true                # Optional BOOLEAN (false). Enables the CU-CP E2 agent. Supported: [false, true].
  enable_cu_up_e2: true                # Optional BOOLEAN (false). Enables the CU-UP E2 agent. Supported: [false, true].
  addr: 10.0.10.73                     # Optional TEXT (127.0.0.1). Sets the RIC IP address.
  port: 36421                           # Optional UINT (36421). Sets the RIC Port. Supported: [20000 - 40000].
  bind_addr: 10.0.10.36                # Optional TEXT (127.0.0.1). Sets the local IP address to bind for RIC connection. Supported: IPv4 address.
  e2sm_kpm_enabled: true               # Optional BOOLEAN (false). Enabled the E2SM KPM service module. Supported: [false, true].
  e2sm_rc_enabled: true                # Optional BOOLEAN (false). Enabled the E2SM RC service module. Supported: [false, true].
``` 

And here is my RIC:
```
ubuntu@ric:~/flexric/build/examples/ric$ ./nearRT-RIC -c ~/flexric/conf_files/ric.conf
Setting the config -c file to /home/ubuntu/flexric/conf_files/ric.conf
[LibConf]: loading service models from SM_DIR: /usr/local/lib/flexric/
[LibConf]: reading configuration for NearRT_RIC
[LibConf]: NearRT_RIC IP: 10.0.10.73
[LibConf]: E2_Port Port: 36421
[LibConf]: E42_Port Port: 36422
[NEAR-RIC]: nearRT-RIC IP Address = 10.0.10.73, PORT = 36421
[NEAR-RIC]: Initializing
[NEAR-RIC]: Loading SM ID = 2 with def = ORAN-E2SM-KPM
[NEAR-RIC]: Loading SM ID = 143 with def = RLC_STATS_V0
[NEAR-RIC]: Loading SM ID = 3 with def = ORAN-E2SM-RC
[NEAR-RIC]: Loading SM ID = 145 with def = SLICE_STATS_V0
[NEAR-RIC]: Loading SM ID = 148 with def = GTP_STATS_V0
[NEAR-RIC]: Loading SM ID = 142 with def = MAC_STATS_V0
[NEAR-RIC]: Loading SM ID = 146 with def = TC_STATS_V0
[NEAR-RIC]: Loading SM ID = 144 with def = PDCP_STATS_V0
[iApp]: Initializing ...
[iApp]: nearRT-RIC IP Address = 10.0.10.73, PORT = 36422
[NEAR-RIC]: Initializing Task Manager with 4 threads

``` 

For detail, my VM IP are:
DU: 10.0.10.174,
CU: 10.0.10.36,
RIC: 10.0.10.73,
Core: 10.0.10.96

Can you give me some advice?
Do I have to connect DU at the same time?
Thanks a lot!

newt2105, 2025-03-08:
> Hi [@newt2105](https://github.com/newt2105) , sorry for just responding now I'm using SC-RIC and I noticed that you're using flexRIC. My logs will not be helpful to you. I'll do my best to help, but you might want to think about using SC-RIC.
> 
> > Hi [Gmclaro](https://github.com/Gmclaro), Can you show me log of RIC when it connected successfully with CU and DU? Thanks.

thanks [Gmclaro](https://github.com/Gmclaro), I will try it.

newt2105, 2025-03-24:
Hi [Gmclaro](https://github.com/Gmclaro),
Can you show me the log when you connect successful DU with USRP? And do you use phone to connect with USRP?

I just connect DU with USRP N321 but it have some problem like this:
```
[INFO] [0/FIFO_1] Initializing block control (NOC ID: 0xF1F0000000000000)
Cell pci=1, bw=10 MHz, 1T1R, dl_arfcn=368640 (n3), dl_freq=1843.2 MHz, dl_ssb_arfcn=368190, ul_freq=1748.2 MHz

F1-C: Connection to CU-CP on 10.0.10.92:38472 completed
E2AP: Connection to Near-RT-RIC on 10.0.10.221:36421 completed
==== DU started ===
Type <h> to view help
Late: 43776; Underflow: 7411; Overflow: 127;
Late: 37792; Underflow: 7238; Overflow: 103;
Late: 47877; Underflow: 8017; Overflow: 135;
Late: 47131; Underflow: 6809; Overflow: 109;
Late: 42037; Underflow: 7483; Overflow: 92;
``` 
Thank you!