srsran/srsRAN_Project #1050: Configuring srsRAN 7.2 Split with USRP B210 as O-RU and GNU Radio Integration

MaryamJavedEE, 2025-02-12:
Hey everyone, I'm completely new to this, so please excuse any lack of technicality. I'm working on an srsRAN 5G project and need to deploy it using the 7.2 split while using a USRP B210 as my O-RU. I also need to integrate GNU Radio in such a way that the gNB sends data to a UDP block (or a similar block) in GNU Radio, which then receives and processes the data.

I've realized that I need to use the ru_ofh configuration for this, but in the ru_sdr configuration, I couldn't find any reference to a specific port that the gNB would send its data to. Since GNU Radio requires a defined port number to listen for incoming data, I need to figure out how to properly configure this in ru_ofh. I also dont know what network interface to mention in the configuration?

My main question is: How do I set up the ru_ofh configuration to accommodate this scenario, considering that GNU Radio sits between the gNB and the USRP B210? (The GNU Radio flow diagram consists of blocks that, when programmed into the USRP, enable it to function as an O-RU.)

Note: Both the srsran 5G project as well as the gnu radio will be running on the same laptop, and the USRP will be connected to the same laptop as well</div>

jjose001, 2025-03-03:
Hi,
I recommed you to open a discussion aswell

pgawlowicz, 2025-03-04:
You would need to implement RU emulator in GnuRadio, that would communicate with gnb using OFH interface and send data to USRP using UHD block. It is not an easy task. In general, we use Split 8 for USRP.

closed 2025-11-12 by ninjab3s