srsran/srsRAN_Project #1060: Cannot setup DU F1U in CU-DU split

o2valente, 2025-02-20:
Hello,

Im deploying a CU-DU split in different machines, but when i try to setup the DU's F1U i'm having some assertion errors when binding an IP address. This happens when i configure  both a localhost IP or a network IP. 

Error on the DU:
```
--== srsRAN DU (commit a041e3162) ==--

Lower PHY in quad executor mode.
/home/nap/srsRAN_Project/lib/f1u/du/split_connector/f1u_split_connector.cpp:50: srsran::srs_du::f1u_split_connector::f1u_split_connector(const srsran::gtpu_gateway_maps&, srsran::gtpu_demux*, srsran::dlt_pcap&, uint16_t, std::string): 
Assertion `not udp_gw_maps.default_gws.empty()' failed - Cannot create CU F1-U split connector, no default GW present
Aborted
```

Currently the CU is connecting to the core, if no configuration for the F1U is provided the DU can connect to the CU, and the UE receives an IP address (although it does not have connection).

CU config:
```                                                         
gnb_id: 411 # Optional UINT (411). Sets the numerical ID associated with the gNB.
gnb_id_bit_length: 22 # Optional UNIT. Sets the bit length of the gnb_id above. Format: integer between [22 - 32]

cu_cp:
  f1ap:
    bind_addr:
      10.0.23.73 # Configure the F1AP bind address, this will enable the CU-cp to connect to the DU
      # tem que ser igual cu-cup address da config da DU
  amf:
    addr: 10.0.23.62 # The address or hostname of the AMF.
    bind_addr: 10.0.23.73 # A local IP that the gNB binds to for traffic from the AMF.
    supported_tracking_areas: # Configure the TA associated with the CU-CP
      - tac: 1
        plmn_list:
          - plmn: "00101"
            tai_slice_support_list:
              - sst: 1

  mobility:
    trigger_handover_from_measurements: true # Optional BOOLEAN (false). Sets whether or not to start HO if neighbor cells become stronger. Supported: [false, true]
    cells: # Optional TEXT. Sets the list of cells known to the CU-CP, their configs (if not provided over F1) and their respective neighbor cells.
      - # Define Cell 1
        nr_cell_id: 0x19B0 # Required UINT. The ID of this serving cell.
        periodic_report_cfg_id: 1 # Optional UINT. The periodical report configuration to use for this serving cell.
        ncells: # Required TEXT. List of neighbor cells.
          - nr_cell_id: 0x19B1 # Required UINT. The ID of this neighbor cell.
            report_configs: [2] # Required TEXT. List of report configurations to use for measurements of this neighbor cell.

        gnb_id_bit_length: 22 # Optional UINT. The ID of this gNB. Supported: [22 - 32]
        pci: 1 # Optional UINT. The PCI of this serving cell. Supported: [0-1007]
        ssb_arfcn: 632628
        band: 78 # Optional UINT. The NR band of this serving cell cell. Must be present if not provided over F1.
        ssb_scs: 30 # Optional UINT. The SSB subcarrier spacing of this serving cell in KHz. Must be present if not provided over F1.
        ssb_period: 20 # Optional UINT. The SSB period of this serving cell in ms.  Must be present if not provided over F1.
        ssb_offset: 0 # Optional UINT. The SSB offset of this serving cell. Must be present if not provided over F1.
        ssb_duration: 5 # Optional UINT. The SSB duration of this serving cell in subframes. Must be present if not provided over F1.
      - nr_cell_id: 0x19B1
        # periodic_report_cfg_id: 2
        ncells:
          - nr_cell_id: 0x19B0
            report_configs: [2]
        gnb_id_bit_length: 22 # Optional UINT. The ID of this gNB. Supported: [22 - 32]
        pci: 2 # Optional UINT. The PCI of this serving cell. Supported: [0-1007]
        ssb_arfcn: 632628
        band: 78 # Optional UINT. The NR band of this serving cell cell. Must be present if not provided over F1.
        ssb_scs: 30 # Optional UINT. The SSB subcarrier spacing of this serving cell in KHz. Must be present if not provided over F1.
        ssb_period: 20 # Optional UINT. The SSB period of this serving cell in ms.  Must be present if not provided over F1.
        ssb_offset: 0 # Optional UINT. The SSB offset of this serving cell. Must be present if not provided over F1.
        ssb_duration: 5 # Optional UINT. The SSB duration of this serving cell in subframes. Must be present if not provided over F1.

    report_configs: # Sets the report configuration for triggering handover
      - report_cfg_id: 1 # Report config ID 1
        report_type: periodical # Sets the report type as periodical
        report_interval_ms: 480 # Sets to report every 480ms
      - report_cfg_id: 2
        report_type: event_triggered
        event_triggered_report_type: a3
        meas_trigger_quantity: rsrp
        meas_trigger_quantity_offset_db: 6
        hysteresis_db: 0
        time_to_trigger_ms: 100
        report_interval_ms: 1024

cu_up:
  f1u:
    socket: # Define UDP/IP socket(s) for F1-U interface.
      - # Socket 1
        bind_addr: 127.0.10.1 # Required TEXT. Sets the address that the F1-U socket will bind to.
        ext_addr: 10.0.23.73
        udp:
          max_rx_msgs: 256    # maximum packets read from the socket in a single syscall.
          pool_threshold: 0.9 # Pool occupancy threshold, after which packets are dropped.

  ngu:
    socket:
      - bind_addr: 10.0.23.73
```

DU config:

```
gnb_id: 411 # Optional UINT (411). Sets the numerical ID associated with the gNB.
gnb_id_bit_length: 22 # Optional UNIT. Sets the bit length of the gnb_id above. Format: integer between [22 - 32]

gnb_du_id: 0

cells:
  - pci: 1
    dl_arfcn: 632628 # Optional TEXT. Set the ARFCN for the above PCI. Note, this must match the value used in the cell_cfg section.

f1ap:
  cu_cp_addr: 10.0.23.73
  bind_addr: 10.0.22.176

f1u:
  socket:
    - bind_addr: 127.0.0.1
      ext_addr: 10.0.22.176
      five_qi: 9
      udp:
        max_rx_msgs: 256 # maximum packets read from the socket in a single syscall.
        pool_threshold: 0.9 # Pool occupancy threshold, after which packets are dropped.

ru_sdr:
  srate: 23.04 # Required FLOAT (61.44). Sets the sampling rate of the RF-frontend in MHz.
  device_driver: uhd # Required TEXT (uhd). RF device driver name. Supported: [uhd, zmq].
  device_args: type=b200,num_recv_frames=64,num_send_frames=64 # Optional TEXT. An argument that gets passed to the selected RF driver.
  otw_format: sc12
  tx_gain: 80
  rx_gain: 40

cell_cfg:
  dl_arfcn: 632628
  sector_id: 1
  band: 78
  channel_bandwidth_MHz: 20
  common_scs: 30
  plmn: "00101"
  tac: 1
  pci: 1
```

I noticed that the NRU field was removed and that this F1U field is quite recent, i've update the code yesterday (19/02). Are you aware of this problem? I dont know if there is any configuration error, but doesnt seem like it as i've tried various combinations.

I have the CU in a Virtual Machine and the DU in an host computer with an USRP B210.

Best regards,
And thanks for the project!

alvasMan, 2025-02-26:
@o2valente, can you drop the 5QI part of the socket config?

```
f1u:
  socket:
    - bind_addr: 127.0.0.1
      ext_addr: 10.0.22.176
      # five_qi: 9 # Remove this part!!!
      udp:
        max_rx_msgs: 256 # maximum packets read from the socket in a single syscall.
        pool_threshold: 0.9 # Pool occupancy threshold, after which packets are dropped.
```

Basically, you are telling the DU to use that socket for 5QI 9, but no socket for all other 5QIs is configured. Configuring a "default" socket for all other 5QIs is still required, which you can do by removing the `five_qi` field. And if you want specific sockets for specific 5QIs (useful for example for setting DSCP codes), you can, but I feel that is not what you are trying to accomplish here. 

This should be caught at the validation layer and print a more user friendly message, but should be easily resolved by removing that field.

alvasMan, 2025-02-26:
@fllay , you have this in your CU config:
```
          ext_addr: 8.8.8.8     
```
which makes no sense to me. Remove it, and if the issue persists open another discussion, as this does not seem to be related to the issue seen by @o2valente.

fllay, 2025-02-26:
@alvasMan   I updated my question here. Please help

https://github.com/srsran/srsRAN_Project/discussions/1048

o2valente, 2025-02-26:
Hi @alvasMan, thanks for your response and explanation. Indeed in this example i was not trying to setup sockets for specific 5QIs, but its definitely an interesting concept (hence why i had it in the config file) and i will try it in the future!

After removing the `five_qi: 9` i was able to run the DU without errors!

Another thing is that i noticed that using `ext_addr` for the F1-U did not establish the user-plane tunnels correctly, so i had to bind the host address directly with `bind_addr`... im not sure if this is the way it was intended to be but its the only way i managed to have it working.

Then in the F1-U of the CU, it was conflicting with the `ngu` in the 2152 port, so i had to add a new host interface. This is for anyone trying to have DU-CU-CORE all in different machines.

But its all working now!
Thanks for the feedback!

closed 2025-02-27 by o2valente