srsran/srsRAN_Project #1067: DU connectivity with Foxconn RU (Missing UP PRACH messages)

sushila-s, 2025-02-25:
We are trying to connect srsRAN 7.2 setup with a Foxconn RU. We are seeing the same issue as reported in: https://github.com/srsran/srsRAN_Project/issues/639. Can you help with the right resolution for this?  

Errors in gnb.log:  
2025-02-25T00:18:57.817016 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '380.19' and sector#0
2025-02-25T00:18:57.827020 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '381.19' and sector#0
2025-02-25T00:18:57.837024 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '382.19' and sector#0
2025-02-25T00:18:57.847026 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '383.19' and sector#0
2025-02-25T00:18:57.857021 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '384.19' and sector#0
2025-02-25T00:18:57.867026 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '385.19' and sector#0
2025-02-25T00:18:57.877022 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '386.19' and sector#0
2025-02-25T00:18:57.887023 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '387.19' and sector#0
2025-02-25T00:18:57.897027 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '388.19' and sector#0
2025-02-25T00:18:57.907012 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '389.19' and sector#0
2025-02-25T00:18:57.917024 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '390.19' and sector#0
2025-02-25T00:18:57.927017 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '391.19' and sector#0
2025-02-25T00:18:57.937011 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '392.19' and sector#0
2025-02-25T00:18:57.947020 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '393.19' and sector#0
2025-02-25T00:18:57.957015 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '394.19' and sector#0
2025-02-25T00:18:57.967020 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '395.19' and sector#0
2025-02-25T00:18:57.977026 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '396.19' and sector#0
2025-02-25T00:18:57.987023 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '397.19' and sector#0
2025-02-25T00:18:57.997033 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '398.19' and sector#0
2025-02-25T00:18:58.000547 [OFH     ] [I] Real-time timing worker woke up late, skipped '4' symbols
2025-02-25T00:18:58.007029 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '399.19' and sector#0
2025-02-25T00:18:58.017023 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '400.19' and sector#0
2025-02-25T00:18:58.027020 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '401.19' and sector#0
2025-02-25T00:18:58.037022 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '402.19' and sector#0
2025-02-25T00:18:58.047020 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '403.19' and sector#0
2025-02-25T00:18:58.057018 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '404.19' and sector#0
2025-02-25T00:18:58.067017 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '405.19' and sector#0
2025-02-25T00:18:58.077027 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '406.19' and sector#0
2025-02-25T00:18:58.081759 [OFH     ] [I] Real-time timing worker woke up late, skipped '2' symbols
2025-02-25T00:18:58.084081 [OFH     ] [I] Real-time timing worker woke up late, skipped '2' symbols
2025-02-25T00:18:58.087021 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '407.19' and sector#0
2025-02-25T00:18:58.097018 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '408.19' and sector#0
2025-02-25T00:18:58.107025 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '409.19' and sector#0
2025-02-25T00:18:58.117031 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '410.19' and sector#0
2025-02-25T00:18:58.127027 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '411.19' and sector#0
2025-02-25T00:18:58.137018 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '412.19' and sector#0
2025-02-25T00:18:58.147029 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '413.19' and sector#0
2025-02-25T00:18:58.157027 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '414.19' and sector#0
2025-02-25T00:18:58.167028 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '415.19' and sector#0
2025-02-25T00:18:58.177030 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '416.19' and sector#0
2025-02-25T00:18:58.187027 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '417.19' and sector#0
2025-02-25T00:18:58.197018 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '418.19' and sector#0
2025-02-25T00:18:58.207025 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '419.19' and sector#0
2025-02-25T00:18:58.217022 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '420.19' and sector#0
2025-02-25T00:18:58.227024 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '421.19' and sector#0
2025-02-25T00:18:58.237022 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '422.19' and sector#0
2025-02-25T00:18:58.247030 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '423.19' and sector#0
2025-02-25T00:18:58.257020 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '424.19' and sector#0
2025-02-25T00:18:58.267026 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '425.19' and sector#0
2025-02-25T00:18:58.277023 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '426.19' and sector#0
2025-02-25T00:18:58.287022 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '427.19' and sector#0
2025-02-25T00:18:58.297022 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '428.19' and sector#0
2025-02-25T00:18:58.307013 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '429.19' and sector#0
2025-02-25T00:18:58.317027 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '430.19' and sector#0
2025-02-25T00:18:58.327021 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '431.19' and sector#0
2025-02-25T00:18:58.337028 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '432.19' and sector#0
2025-02-25T00:18:58.347016 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '433.19' and sector#0
2025-02-25T00:18:58.357029 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '434.19' and sector#0
2025-02-25T00:18:58.367035 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '435.19' and sector#0
2025-02-25T00:18:58.377021 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '436.19' and sector#0
2025-02-25T00:18:58.387024 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '437.19' and sector#0
2025-02-25T00:18:58.397024 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '438.19' and sector#0
2025-02-25T00:18:58.407025 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '439.19' and sector#0
2025-02-25T00:18:58.417016 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '440.19' and sector#0
2025-02-25T00:18:58.427025 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '441.19' and sector#0
2025-02-25T00:18:58.437022 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '442.19' and sector#0
2025-02-25T00:18:58.447018 [OFH     ] [W] Missed incoming User-Plane PRACH messages for slot '443.19' and sector#0
[... 58 lines cut ...]
What should these values be? And what in the RU config file should be matched with these? Is this the only one? Including a few more that may be relevant:   
```  
RRH_EN_EAXC_ID = 0
<!-- RRH_EAXC_ID_TYPE1: Specify the eAxC ID for type1 message -->
RRH_EAXC_ID_TYPE1 = 0x0, 0x1, 0x2, 0x3
<!-- RRH_EAXC_ID_TYPE3: Specify the eAxC ID for type3 message -->
RRH_EAXC_ID_TYPE3 = 0x8, 0x9, 0xA, 0xB
 
<!-- RRH_TRX_EN_BIT_MASK: Bit-mask of 4 TRx, bit 0: TRx0, bit1: TRx1, bit2: TRx2, bit3: TRx3 -->
RRH_TRX_EN_BIT_MASK = 0x03
<!-- RRH_RF_EN_BIT_MASK: Bit-mask of 4 PA/LNA, bit0: PA0/LNA0, bit1: PA1/LNA1, bit2: PA2/LNA2, bit3: PA3/LNA3 -->
RRH_RF_EN_BIT_MASK = 0x03

<!-- RRH_MAX_PRB: Maximum PRBs -->
RRH_MAX_PRB = 51 

3. Other info: 
4. VLAN is set to 2 on both 
4. PTP is sync is good
5. Checked with spectrum analyzer, RU seems to be transmitting a signal. On RU, RRH_state=DATA_EXCHANGING 
6. UE does not camp, have tried with OnePlus 8T and Pixel 7a 

Can you help debug? Some guidance on the specific config file params to match with RU config will help.

Thanks!

sushila-s, 2025-02-25:
Attaching complete config files.   

[gnb_ru_rpqn4800e_tdd_n78_20mhz_2x2.txt](https://github.com/user-attachments/files/18952074/gnb_ru_rpqn4800e_tdd_n78_20mhz_2x2.txt)  

[RRHconfig_xran.txt](https://github.com/user-attachments/files/18952067/RRHconfig_xran.txt)

sushila-s, 2025-03-11:
Hi @andrepuschmann  can you or someone on the team help with this please. We are still stuck with this issue. 

To add to the above information, tcpdump on DU shows a lot of outgoing eCPRI packets from DU, but I dont see any of these reaching the RU. I also don't see anything coming from RU to DU. Apart from the ethernet name and MAC address, do we need to configure any other RU info? 

We are really stuck! Any help is much appreciated! 
Need some help with the right eXaC <-> ports config if that is the problem. Or is there something else we are missing? 

tcpdump on DU: 
wcsng@wcsng-50:~$ sudo tcpdump -e -q -i  enp4s0f0np0 -Q inout vlan
16:30:34.239978 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240009 6c:ad:ad:00:0d:66 (oui Unknown) > 6c:b3:11:21:f8:7e (oui Unknown), 802.1Q, length 64: vlan 1, p 1, 802.3LLC, dsap Unknown (0x10) Individual, ssap 802.1B I (0x02) Command, ctrl 0x1000: Information, send seq 0, rcv seq 8, Flags [Command], length 42
16:30:34.240016 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240017 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240043 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240044 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240082 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240087 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240113 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 64: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240115 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 64: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240116 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240118 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240155 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240160 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240198 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240208 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240233 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240234 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240259 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240261 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240298 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240304 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240329 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240330 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240368 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240369 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240406 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240413 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240450 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240452 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240477 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
16:30:34.240479 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), 802.1Q, length 1464: vlan 2, p 0, Unknown Ethertype (0xaefe), 
[... cut ...]
23:29:51.706497 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), IPv4, length 66: 192.167.1.31.57466 > 192.167.1.51.830: tcp 0
23:29:52.746311 6c:ad:ad:00:0d:66 (oui Unknown) > 6c:b3:11:21:f8:7e (oui Unknown), IPv4, length 398: 192.167.1.51.830 > 192.167.1.31.57466: tcp 332
23:29:52.746506 6c:b3:11:21:f8:7e (oui Unknown) > 6c:ad:ad:00:0d:66 (oui Unknown), IPv4, length 66: 192.167.1.31.57466 > 192.167.1.51.830: tcp 0
^C
30 packets captured
33 packets received by filter
0 packets dropped by kernel

mooncake-22, 2025-03-12:
Your DU send the O-FH packets with `VLAN tag 2` and `length 1464`. However, your RU only get `tcp` packets from `qse-eth` port. It seems that the qse-eth port on RU doesn't get any O-FH packets from your DU. You may check your network setup again. Hope this will help you.

andrepuschmann, 2025-03-20:
Hey @sushila-s ,

I would really like to ask you to follow the guide and config here https://docs.srsran.com/projects/project/en/latest/tutorials/source/oranRU/source/rus/foxconn.html

If your RU is not receiving any packets there is something wrong, either in the config or maybe also the cabling? Have you tried to run `tcpdump` on the RU?

Btw, can you please post the output of the console when you execute the `init_rrh_config_enable_cuplane` script. Note that the `RRH_SRC_MAC_ADDR` is actually not read from the config file but is replaced by the actual MAC of the interface. If that is wrong you can put whatever you want and it will not work.

Thanks
Andre

sushila-s, 2025-03-25:
@andrepuschmann Thanks. I am following the config from here https://docs.srsran.com/projects/project/en/latest/tutorials/source/oranRU/source/rus/foxconn.html

I did check the RRH_SRC_MAC_ADDR in the output of init_rrh_config_enable_cuplane script and it is picked up correctly. 

I have some specific questions: In the sample DU config here: https://github.com/srsran/srsRAN_Project/blob/main/configs/gnb_ru_rpqn4800e_tdd_n78_20mhz_2x2.yml
You have:
```
      prach_port_id: [4, 5, 6, 7]
      dl_port_id: [0, 1]
      ul_port_id: [0, 1, 2, 3]
```
Is this correct for a 2x2 config? It doesn't work for us, I get errors running this. 
For a 2x2 config, I assume  prach_port_id and ul_port_id should have two fields? What should these be. 

I tried the following: 
```
      prach_port_id: [4,5]
      dl_port_id: [0, 1]
      ul_port_id: [0, 1]
```
With this, I see the following in RU log (rrh_log_print.log):   
```
[2025-03-11 22:30:27.073] xRN: total=14280986 c_early=0 c_on=1142079 c_late=397 err_tci=0 err_ecpri=0 err_port=142808 err_sct=0
```  
This is the response we got from Foxconn team:   
``` "err_port=142808" means your "RRH_EAXC_ID_TYPE1" settings are wrong.```

Foxconn RU config file has these comments: 
```
 <!-- RRH_EN_EAXC_ID: Enable using eAxC ID field defined in O-RAN spec.                            -->
<!--                 When 0 is set, RU port ID=0,1,2,3 are used for PDSCH/PUSCH if RRH_TRX_EN_BIT_MASK = 0x0F -->
<!--                 When 0 is set, RU port ID=4,5,6,7 are used for PRACH       if RRH_TRX_EN_BIT_MASK = 0x0F -->
<!--                 When 0 is set, RU port ID=0,1     are used for PDSCH/PUSCH if RRH_TRX_EN_BIT_MASK = 0x03 -->
<!--                 When 0 is set, RU port ID=2,3     are used for PRACH       if RRH_TRX_EN_BIT_MASK = 0x03 -->
```
So I  tried matching with the RU config file comments and used 
```
      prach_port_id: [2,3]
      dl_port_id: [0, 1]
      ul_port_id: [0, 1]
```
With this, I don't see the error in the RU log file, but we still see no PRACH on the DU 

In both cases, I see the following error messages in gnb.log 
```
2025-02-25T00:18:57.817016 [OFH ] [W] Missed incoming User-Plane PRACH messages for slot '380.19' and sector#0
2025-02-25T00:18:57.827020 [OFH ] [W] Missed incoming User-Plane PRACH messages for slot '381.19' and sector#0
2025-02-25T00:18:57.837024 [OFH ] [W] Missed incoming User-Plane PRACH messages for slot '382.19' and sector#0
2025-02-25T00:18:57.847026 [OFH ] [W] Missed incoming User-Plane PRACH messages for slot '383.19' and sector#0
```

Can you help with this. What should the values for this be for 2x2 MIMO. 

Note: The complete DU and RU config files are uploaded in the issue. We have the Ethernet interface name and MAC addresses configured correctly (wireshark capture shows some OFH packets going to RU MAC), VLAN 2 is setup on both sides, and other config params for 2x2 are setup (num of ul/dl antennas = 2, Cell and RU bandwidth config = 20Mhz,  PRBs = 51). Are there some other parmeters we should be looking at

ismagom, 2025-04-04:
HI @sushila-s, 

I think you have lates in OFH packets, see ` c_late=397` in the RU log. Check the PTP sync and follow this guide here https://docs.srsran.com/projects/project/en/latest/tutorials/source/tuning/source/index.html

You should not see warnings in the gnb log and `c_late=0` in the RU log