srsran/srsRAN_Project #1075: Connection issue in 5G srsRAN using USRP X310

Anwesha1D, 2025-02-28:
Hello everybody,

[image]
[image]
[image]

We have been trying to set up 5G srsRAN using two USRP X310s and an Octoclock as the external clock source. We use VERT2450 antennas which function at 2.4 GHz. 
Our gNB to core connection is successful, but there is an issue in attaching the UE. We only see one line of overflow, underflow report on the UE side and the rest of the procedures do not happen. 
We have attached the UE and gNB config files, the core parameters (sample.yaml) and the screenshots of the terminal updates for the core, gNB and UE. We would be really grateful if someone could give us some guidance regarding how to fix this issue.

[gnb.txt](https://github.com/user-attachments/files/19032200/gnb.txt)
[sample.txt](https://github.com/user-attachments/files/19032201/sample.txt)
[ue_rf.txt](https://github.com/user-attachments/files/19032199/ue_rf.txt)

pgawlowicz, 2025-03-03:
Try removing ` send_frame_size = 1472, recv_frame_size = 1472`, or passing higher values like 8000

jjose001, 2025-03-03:
Hello, I am having the same problem. I have the setup installed on a VMware workstation environment a VM running ubuntu 22. For setup using a usrp b210, is it totally necessary to apply the "srsran_performance" script to avoid late/underflow/overflow messages?

gnb.yml: 
[gnb_b210_conf.txt](https://github.com/user-attachments/files/19052184/gnb_b210_conf.txt)

pgawlowicz, 2025-03-04:
Not sure, if VM provides a sufficient performance. Could you check with barebone OS?

jjose001, 2025-03-06:
Hi @pgawlowicz,

Running srsRAN on a virtual machine with: 8 CPUs and 16 GBs of RAM, I got the following output (see attached screenshot1

[image]

[image]

) with different processors working at maximum load. Now, after installing the native Ubuntu OS, and running the srsran_performance script the error has disappeared, and the CPU load I get now is: 

[image].

pgawlowicz, 2025-03-06:
Great! Thanks for reporting back.

closed 2025-03-06 by pgawlowicz

Anwesha1D, 2025-03-17:
Hello @pgawlowicz. Thank you for your suggestion. I tried changing the frame size, but it gives a warning saying that our NIC's max frame size is 1472, and the frame size gets set to 1472.

[image]

pgawlowicz, 2025-03-24:
@Anwesha1D change the mtu size of the interface with ifconfig or ip command.