srsran/srsRAN_Project #1076: RRC released for ZMQ setup

PsArzzz, 2025-03-03:
### **Issue Description**

I am using Open5GS as the core network and srsRAN to set up the gNodeB and UE, with ZeroMQ for communication between the gNodeB and UE. However, after the UE establishes an RRC connection, it disconnects approximately ten seconds later. Upon reviewing the gNB logs, I discovered that the PDU session establishment is timing out. I seek guidance on resolving this issue to **enable the UE to successfully establish and maintain a continuous PDU session with the AMF**.

### **Setup Details**
The setup consists of four virtual machines, each running Ubuntu 22.04. The details are as follows:

1、Open5GS Control Plane (5GC C-Plane): IP Address: 192.168.83.131
2、Open5GS User Plane (5GC U-Plane): IP Address: 192.168.83.130
3、srsRAN Project ZMQ RAN (gNodeB): IP Address: 192.168.83.136
4、srsRAN 4G ZMQ UE (NR-UE): IP Address: 192.168.83.137

### **Actual Behaviour**
1、The following output appears on the srsUE terminal:
[image]
2、Logs from UE:
[ue.log](https://github.com/user-attachments/files/19051500/ue.log)
3、Logs from gNB:
[gnb.log](https://github.com/user-attachments/files/19051508/gnb.log)
[image]
4、Logs from AMF:
The logs generated after approximately 17:30 correspond to my most recent experiments.
[amf.log](https://github.com/user-attachments/files/19051534/amf.log)

**Configuration Files:**
5、amf.yaml(On Open5GS 5GC C-Plane)
[amf_yaml.txt](https://github.com/user-attachments/files/19051573/amf_yaml.txt)
6、nrf.yaml(On Open5GS 5GC C-Plane)
[nrf_yaml.txt](https://github.com/user-attachments/files/19051588/nrf_yaml.txt)
7、smf.yaml(On Open5GS 5GC C-Plane)
[smf_yaml.txt](https://github.com/user-attachments/files/19051590/smf_yaml.txt)
8、upf.yaml(On Open5GS 5GC U-Plane)
[upf_yaml.txt](https://github.com/user-attachments/files/19051598/upf_yaml.txt)
9、gnb_zmq.yaml
[gnb_zmq_yaml.txt](https://github.com/user-attachments/files/19051601/gnb_zmq_yaml.txt)
10、ue_zmq.conf
[ue_zmq_conf.txt](https://github.com/user-attachments/files/19051606/ue_zmq_conf.txt)

**Additional Information**
My issue is identical to the one described in [srsRAN_Project Issue #589](https://github.com/srsran/srsRAN_Project/issues/589). I attempted the suggested solution by setting the APN to 'srsapn' in both the AMF's subscriber configuration and the ue_zmq.conf file, but it did not resolve the problem. Additionally, I observed that the 'DNN Not Supported OR Not Subscribed in the Slice' error appears in the AMF logs only when there is a mismatch between the APN settings in the subscriber configuration and ue_zmq.conf. Currently, I am not encountering this error, yet the connection still drops.

pgawlowicz, 2025-03-04:
Did you start any traffic between the UE and the core (e.g., ping)?

PsArzzz, 2025-03-04:
> Did you start any traffic between the UE and the core (e.g., ping)?
No, after I started the UE, the connection dropped after about ten seconds, so I didn't start any traffic.
So what should I do to test?

Also, I found that my SMF is not working. I checked its status and here is what it shows:
```
open5gs-smfd.service - Open5GS SMF Daemon
     Loaded: loaded (/lib/systemd/system/open5gs-smfd.service; enabled; vendor preset: enabled)
     Active: activating (auto-restart) (Result: exit-code) since Tue 2025-03-04 16:46:05 CST; 1s ago
    Process: 7259 ExecStart=/usr/bin/open5gs-smfd -c /etc/open5gs/smf.yaml (code=exited, status=255/EXCEPTION)
   Main PID: 7259 (code=exited, status=255/EXCEPTION)
        CPU: 23ms

Mar 04 16:42:34 a systemd[1]: open5gs-smfd.service: Main process exited, code=exited, status=255/EXCEPTION
Mar 04 16:42:34 a systemd[1]: open5gs-smfd.service: Failed with result 'exit-code'.
```

pgawlowicz, 2025-03-04:
If the UE is inactive, it will be released. You can start pinging any IP (e.g. 8.8.8.8) so the connection will be kept.

PsArzzz, 2025-03-04:
After running sudo ./src/srsue ue_zmq.conf, I opened another window to ping 8.8.8.8, but it still got released.

PsArzzz, 2025-03-04:
> After running sudo ./src/srsue ue_zmq.conf, I opened another window to ping 8.8.8.8, but it still got released.

and the reason is the same.

pgawlowicz, 2025-03-04:
Is the UE assigned an IP address? It should print sth like this:
```
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=2094
Random Access Complete.     c-rnti=0x4602, ta=0
RRC Connected
PDU Session Establishment successful. IP: 10.45.1.2
RRC NR reconfiguration successful.
```

PsArzzz, 2025-03-04:
My PDU setup failed, and the UE did not get an IP address.
It print:
```
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=174
Random Access Complete.     c-rnti=0x4601, ta=0
RRC Connected
Received RRC Release
```

In the amf.log print:
```
03/04 10:48:46.042: [amf] DEBUG: amf_state_operational(): OGS_EVENT_NAME_SBI_CLIENT (../src/amf/amf-sm.c:81)
{"type":"/nnssf-nsselection/v2","title":"Cannot find NSI by S-NSSAI[SST:1 SD:0x1]","status":403,"instance":"/network-slice-information"}03/04 10:48:46.042: [amf] ERROR: [imsi-001010000000001] HTTP response error [403] (../src/amf/nnssf-handler.c:63)
```
How can I solve this problem?

pgawlowicz, 2025-03-04:
It seems there is some mismatch between the UE database in open5gs, and what UE sends. Did you set the slice for the UE in the Open5gs?