srsran/srsRAN_Project #1078: CU-DU split issue with UE authentification

BurmazDaniel, 2025-03-03:
Hello!

I am not able to authentificate UE to the core using the new f1u socket.
CU ip adresses are: 192.168.250.161 and 192.168.250.80
DU ip: 10.255.100.11 (wireguard tunnel able to ping CU)

Here is my CU.yml:
```
gnb_id: 411
gnb_id_bit_length: 22

cu_cp:
  f1ap:
    bind_addr: 192.168.250.161
  amf:
    addr: 192.168.250.67
    port: 38412
    bind_addr: 192.168.250.161
    supported_tracking_areas:
      - tac: 1
        plmn_list:
          - plmn: "00101"
            tai_slice_support_list:
              - 
                sst: 1
                sd: 16777215

  mobility:
    trigger_handover_from_measurements: true
    cells:
      - 
        nr_cell_id: 0x19B0
        periodic_report_cfg_id: 1
        ncells:
          - nr_cell_id: 0x19B1
            report_configs: [2]
        gnb_id_bit_length: 22
        pci: 1
        ssb_arfcn: 632628
        band: 78
        ssb_scs: 30
        ssb_period: 20
        ssb_offset: 0
        ssb_duration: 5
      - nr_cell_id: 0x19B1
        ncells:
          - nr_cell_id: 0x19B0
            report_configs: [2]
        gnb_id_bit_length: 22
        pci: 2
        ssb_arfcn: 632628
        band: 78
        ssb_scs: 30
        ssb_period: 20
        ssb_offset: 0
        ssb_duration: 5

    report_configs:
      - report_cfg_id: 1
        report_type: periodical
        report_interval_ms: 480
      - report_cfg_id: 2
        report_type: event_triggered
        event_triggered_report_type: a3
        meas_trigger_quantity: rsrp
        meas_trigger_quantity_offset_db: 6
        hysteresis_db: 0
        time_to_trigger_ms: 100
        report_interval_ms: 1024

cu_up:
  f1u:
    socket:
      - 
        bind_addr: 192.168.250.80
        ext_addr:
        udp:
          max_rx_msgs: 256
          pool_threshold: 0.9

  ngu:
    socket:
      - bind_addr: 192.168.250.161
```

Here is my DU.yml:
```
gnb_id: 411
gnb_id_bit_length: 22
gnb_du_id: 0

f1ap:
  cu_cp_addr: 192.168.250.161
  bind_addr: 10.255.100.11

ru_sdr:
  device_driver: uhd
  device_args: type=b200,num_recv_frames=64,num_send_frames=64
  srate: 23.04
  otw_format: sc12
  tx_gain: 80
  rx_gain: 60

cells:
  - pci: 1
    dl_arfcn: 632628

f1u:
  socket:
    - bind_addr: 10.255.100.11
      ext_addr: 192.168.250.80
      udp:
        max_rx_msgs: 256
        pool_threshold: 0.9

cell_cfg:
  dl_arfcn: 632628
  band: 78
  channel_bandwidth_MHz: 20
  common_scs: 30
  plmn: "00101"
  tac: 1
  pci: 1
  nof_antennas_dl: 1
  nof_antennas_ul: 1
  pdsch:
    mcs_table: qam256
```

Core seem to be angry. I had 0 problems with gnb and with the older nru solution:

[image]

Thank you!

BurmazDaniel, 2025-03-16:
Up? :)

h57jafari, 2025-03-21:
Hi, could you solve this? 
please share the amf.yaml and upf.yaml and ue.conf files. I am having similar problem, but after running ue command. CU side crashes with error of "segmentation fault".

BurmazDaniel, 2025-03-22:
> Hi, could you solve this? please share the amf.yaml and upf.yaml and ue.conf files. I am having similar problem, but after running ue command. CU side crashes with error of "segmentation fault".

What RU are you using? Segmentation fault is more related to tx/rx side of RU.

h57jafari, 2025-03-22:
Hi, thanks for reply, I am using ZMQ.

pgawlowicz, 2025-06-12:
@BurmazDaniel  any update on this issue?