srsran/srsRAN_Project #1081: srsUE on FPGA no progress

shirinebadi, 2025-03-05:
I'm trying to run srsUE on FPGA (RISC-v) connected to srsgnb through ZMQ. However it stuck and didn't make any progress:

[image]

I attached the ue.conf and here is the logs:
1970-01-01T00:00:57.079043 [UE     ] [I] Using binary ./srsue with arguments: ue.conf 
1970-01-01T00:00:58.498179 [UE     ] [I] Built in Release mode using commit 80ffb2f on branch master.
1970-01-01T00:01:01.249896 [RF     ] [I] Configuring physical DL channel 0 with band-pass filter (0.0, 0.0)
1970-01-01T00:01:01.250306 [RF     ] [I] Configuring physical UL channel 0 with band-pass filter (0.0, 0.0)

[[ue.zip](https://ue.zip/)](https://github.com/user-attachments/files/19096775/ue.zip)

Thank you in advance.

shirinebadi, 2025-03-06:
"Update"

I think there were some problems with the board, but it is fixed now. 

Currently the UE hangs on attaching to gNB.  I'm pretty much sure about the IP addresses. Both entities can ping each other.

[gnb.zip](https://github.com/user-attachments/files/19116543/gnb.zip)

pgawlowicz, 2025-03-07:
could you try with a 5MHz cell first?

shirinebadi, 2025-03-07:
Is that possible with ZMQ setup?
I changed the BW in gnb config file to 5MHz but it didn't start:
```
--== srsRAN gNB (commit e5d5b44b9) ==--

Unable to derive a valid SSB pointA and k_SSB for CORESET#0 index=12, SearchSpace#0 index=0 and cell bandwidth=5Mhz
srsRAN ERROR: Invalid configuration detected.
```

pgawlowicz, 2025-03-07:
yes, try:
```
cell_cfg:
...
  pdcch:
    common:
      ss0_index: 0                    # Set search space zero index to match srsUE capabilities
      coreset0_index: 0            # Set search CORESET Zero index to match srsUE capabilities
```

shirinebadi, 2025-03-07:
With the given parameters, the number of PRBs for PUCCH exceeds the 50% of the BWP PRBs.
srsRAN ERROR: Invalid configuration detected.

pgawlowicz, 2025-03-07:
indeed, please add also:
```
cell_cfg:
...
  pucch:
    f0_or_f1_nof_cell_res_sr: 2
    f2_or_f3_or_f4_nof_cell_res_csi: 2
```

shirinebadi, 2025-03-07:
INI was not able to parse cell_cfg.pucch.f2_or_f3_or_f4_nof_cell_res_csi

pgawlowicz, 2025-03-07:
Please update your srsran repo. The parameter is available here:
https://github.com/srsran/srsRAN_Project/blob/8120ffbce1f802542c9256fd2104f45c86b7e488/apps/units/flexible_o_du/o_du_high/du_high/du_high_config_cli11_schema.cpp#L1001

shirinebadi, 2025-03-07:
I updated repo,
Lower PHY in executor blocking mode.
srsRAN ERROR: Invalid configuration DU cell detected.
> 256QAM MCS table cannot be used for PDSCH with fallback DCI format in SearchSpace#2

shirinebadi, 2025-03-07:
No progress.

My current gnb.confi:
```

cu_cp:
  amf:
    addr: 10.53.1.2                 # The address or hostname of the AMF.
    port: 38412
    bind_addr: 10.53.1.1            # A local IP that the gNB binds to for traffic from the AMF.
    supported_tracking_areas:
      - tac: 7
        plmn_list:
          - plmn: "00101"
            tai_slice_support_list:
              - sst: 1
  inactivity_timer: 7200            # Sets the UE/PDU Session/DRB inactivity timer to 7200 seconds. Supported: [1 - 7200].

ru_sdr:
  device_driver: zmq                # The RF driver name.
  device_args: tx_port=tcp://128.138.75.217:2001,rx_port=tcp://128.138.75.248:2002,base_srate=23.04e6 # Optionally pass arguments to the selected RF driver.
  srate: 23.04                      # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 75                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 75                       # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 368500                  # ARFCN of the downlink carrier (center frequency).
  band: 3                           # The NR band.
  channel_bandwidth_MHz: 5         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 15                    # Subcarrier spacing in kHz used for data.
  plmn: "00101"                     # PLMN broadcasted by the gNB.
  tac: 7                            # Tracking area code (needs to match the core configuration).
  pdcch:
    common:
      ss0_index: 0                  # Set search space zero index to match srsUE capabilities
      coreset0_index: 0          # Set search CORESET Zero index to match srsUE capabilities
    dedicated:
      ss2_type: common              # Search Space type, has to be set to common
      dci_format_0_1_and_1_1: false # Set correct DCI format (fallback)
  prach:
    prach_config_index: 1           # Sets PRACH config to match what is expected by srsUE
  pdsch:
    mcs_table: qam64                # Sets PDSCH MCS to 64 QAM
  pusch:
    mcs_table: qam64
  pucch:
    f0_or_f1_nof_cell_res_sr: 2
    f2_or_f3_or_f4_nof_cell_res_csi: 2                # Sets PUSCH MCS to 64 QAM

log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: info                   # Logging level applied to all layers.
  hex_max_size: 0

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.

```
with no changes to ue.conf

shirinebadi, 2025-03-07:
Just Sometimes I see 
```
2025-03-07T20:05:40.500739 [zmq:tx:0:0] [I] Waiting for request.
2025-03-07T20:05:41.195026 [zmq:rx:0:0] [I] Waiting for data.
2025-03-07T20:05:41.196508 [zmq:rx:0:0] [I] Waiting for reading samples. Completed 300 of 11520 samples.
2025-03-07T20:05:41.500739 [zmq:tx:0:0] [I] Waiting for request.
2025-03-07T20:05:42.195028 [zmq:rx:0:0] [I] Waiting for data.
2025-03-07T20:05:42.196532 [zmq:rx:0:0] [I] Waiting for reading samples. Completed 300 of 11520 samples.
2025-03-07T20:05:42.500741 [zmq:tx:0:0] [I] Waiting for request.
2025-03-07T20:05:43.195030 [zmq:rx:0:0] [I] Waiting for data.
2025-03-07T20:05:43.196586 [zmq:rx:0:0] [I] Waiting for reading samples. Completed 300 of 11520 samples.
```
in gnb logs

pgawlowicz, 2025-03-07:
could you reduce the srate on both sides to 5.76e6?

shirinebadi, 2025-03-08:
2025-03-08T00:00:46.844897 [zmq:tx:0:0] [I] Waiting for request.
2025-03-08T00:00:47.613770 [zmq:rx:0:0] [I] Waiting for data.
2025-03-08T00:00:47.619385 [zmq:rx:0:0] [I] Waiting for reading samples. Completed 75 of 2880 samples.
2025-03-08T00:00:47.844897 [zmq:tx:0:0] [I] Waiting for request.
2025-03-08T00:00:48.613771 [zmq:rx:0:0] [I] Waiting for data.
2025-03-08T00:00:48.619440 [zmq:rx:0:0] [I] Waiting for reading samples. Completed 75 of 2880 samples.
2025-03-08T00:00:48.844897 [zmq:tx:0:0] [I] Waiting for request.
2025-03-08T00:00:49.613773 [zmq:rx:0:0] [I] Waiting for data.
2025-03-08T00:00:49.619447 [zmq:rx:0:0] [I] Waiting for reading samples. Completed 75 of 2880 samples.

I don't know what is causing this. Is there any kind of timeout in gnb side?

shirinebadi, 2025-03-08:
Do you think that it is caused by the insufficient resources? (I think UE is dropping or not processing samples correctly).
My FPGA has 2 CPU and 1 G mem.

This is the result of the top (for ue):
```
Mem: 599488K used, 400836K free, 8928K shrd, 980K buff, 27800K cached
CPU:  13% usr  85% sys   0% nic   1% idle   0% io   0% irq   0% sirq
Load average: 2.34 0.72 0.26 6/66 149
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
  140   133 root     D     813m  83%  80% ./shtest ue.conf

```

shirinebadi, 2025-03-10:
@pgawlowicz Do you think this issue is caused by an improper sample rate or CPU-related problems? I tested it with 5.76, 7.68, and 23.04, but the same problem persisted. I’m planning to test with other sample rates as well (like 24.96 and 11.52), but do you think that would help? Or is it more likely that the issue is related to the kernel or CPU?

pgawlowicz, 2025-03-10:
did you set srate for both gnb and srsue?

AlejandraOliver, 2025-03-10:
Hi! I have the same issue, with this configuration the UE is not attaching the gNB

shirinebadi, 2025-03-10:
@pgawlowicz Yes, I did.

shirinebadi, 2025-03-11:
@pgawlowicz I collected the zmq logs in ue. I couldn't get so much of it. Is there anything unusual in it?

[uedump-fpga.log](https://github.com/user-attachments/files/19173301/uedump-fpga.log)

shirinebadi, 2025-03-12:
Hi @pgawlowicz,

Sorry for spamming. I'm trying to understand why incomplete sample readings are happening in gNB. Instead of using FPGA, I ran srsUE on QEMU, and it connects to gNB with a 23.04 srate but not with 5.76. This doesn't make sense to me—why wouldn't it work with a lower sample rate?

To explain why it’s not working on FPGA, I initially thought it was due to the FPGA being too slow (its frequency is 50 MHz, which is quite low), causing it to send samples very slowly, so gNB doesn’t receive enough samples every millisecond. However, I can't explain why a system that works with a 23.04 srate fails at 5.76. This is very confusing for me.

Sorry, I have no background in wireless communications—I'm just using srsRAN to test my system.

pgawlowicz, 2025-03-12:
could you post your configs you use for 5mhz cell in the setup with fpga?

shirinebadi, 2025-03-12:
[gnb_zmqq.zip](https://github.com/user-attachments/files/19214564/gnb_zmqq.zip)

[ue.zip](https://github.com/user-attachments/files/19214587/ue.zip)

Thanks

pgawlowicz, 2025-03-12:
could you change the following in the srsue:
```
[rat.nr]
bands = 3
nof_carriers = 1
max_nof_prb = 25
nof_prb = 25
```

shirinebadi, 2025-03-12:
Same issue 

[gnb.log](https://github.com/user-attachments/files/19215916/gnb.log)

[uelogs-fpga.zip](https://github.com/user-attachments/files/19216015/uelogs-fpga.zip)

pgawlowicz, 2025-03-14:
Do you have an option to run your setup with usrp instead of zmq?

shirinebadi, 2025-03-14:
Not now, but I'll look into it.
Would you please share your thoughts on what might be causing the issue?

pgawlowicz, 2025-03-14:
The zmq-based RF-device might get stuck. I have experienced such behaviour when using real-time Linux kernel.
What kind of FPGA board do you use?

shirinebadi, 2025-03-14:
The board is Genesys 2 Kintex-7 FPGA. It's quite slow and resource limited. CPU speed is 50 MHz(while laptops work in 2000-2500 order).

pgawlowicz, 2025-03-14:
maybe you could add `low_phy_dl_throttling` to your gnb config to slow down the gnb.
```
ru_sdr:
  device_driver: zmq                # The RF driver name.
  device_args: tx_port=tcp://127.0.0.1:2100,rx_port=tcp://127.0.0.1:2101 # Optionally pass arguments to the selected RF driver.
  srate: 5.76                      # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 75                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 75                       # Receive gain of the RF might need to adjusted to the given situation.
  expert_cfg:
    low_phy_dl_throttling: 0.999
```

shirinebadi, 2025-03-14:
Thanks but UE is running on FPGA. Gnb is on my laptop

pgawlowicz, 2025-03-14:
yes, since the ue is slow, you can try to slow down the gnb so the ue can keep up

shirinebadi, 2025-03-14:
Indeed. Will try it when I get to the lab

shirinebadi, 2025-03-14:
Thanks
No progress. 
I think I should consider alternative approaches.

pgawlowicz, 2025-03-14:
which OS and kernel do you run on the fpga?