srsran/srsRAN_Project #1083: How to connect DU, CU split with nearRT RIC

newt2105, 2025-03-07:
Hi everyone, 
I'm have 4 VM (CU, DU, RIC, Core). I already connected successfully DU, CU and Core together but i'm have some problem when I connect DU or CU with RIC.

Here is the error of CU:

```
--== srsRAN CU (commit 8120ffbce) ==--

N2: Connection to AMF on 10.0.10.96:38412 completed
E2AP: Failed to connect to RIC on 10.0.10.73:36421. error="Connection timed out" timeout=341235ms
F1-C: Listening for new connections on 10.0.10.36:38472...
Segmentation fault
``` 

and their configuration files:
CU:

```
cu_cp:
  f1ap:
    bind_addr:
      10.0.10.36 # Configure the F1AP bind address, this will enable the CU-cp to connect to the DU
  amf:
    addr: 10.0.10.96 # The address or hostname of the AMF.
    bind_addr: 10.0.10.36 # A local IP that the gNB binds to for traffic from the AMF.
    supported_tracking_areas: # Configure the TA associated with the CU-CP
      - tac: 1
        plmn_list:
          - plmn: "00101"
            tai_slice_support_list:
              - sst: 1

e2:
  enable_cu_cp_e2: true                
  enable_cu_up_e2: true                 
  addr: 10.0.10.73
  port: 36421                         
  bind_addr: 10.0.10.36               
  e2sm_kpm_enabled: true              
  e2sm_rc_enabled: true 
``` 

RIC:
```
SM_DIR = "/usr/local/lib/flexric/"

# supported name = NearRT_RIC, E2_Agent, E2_Proxy_Agent, xApp
Name = "NearRT_RIC"
NearRT_RIC_IP = "10.0.10.73"
E2_Port = 36421
E42_Port = 36422
``` 

Can someone deal with this problem before?

pgawlowicz, 2025-03-07:
Could you check if the CU-CP E2 agent connects when running gnb app?

newt2105, 2025-03-08:
> Could you check if the CU-CP E2 agent connects when running gnb app?

Hi [pgawlowicz](https://github.com/pgawlowicz), 
I try to connect gnb with RIC and it has the same error with CU connect with RIC:
```
E2AP: Failed to connect to RIC on 10.0.10.73:36425. error="Operation already in progress" timeout=9598ms
``` 

But when I'm connect gnb and RIC in the same host, it worked

newt2105, 2025-03-11:
Hi [pgawlowicz](https://github.com/pgawlowicz),
I just want to tell that it worked, I can connect CU, DU split with nearRT RIC on different VM. 

I have another question, I deploy CU, DU , RIC on server openstack, I also connectd USRP N321 with the server, I can ssh to usrp. Can I connect CU, DU with USRP on server? without cable? Because I read some tutorial and they use cable.

Thanks!

Keilwerth11270, 2025-03-13:
Hi @newt2105 what changes did you make to fix the connection problem? Did you have some changes in CU/DU configs or in the O-RAN SC RIC from srsRAN docker-compose file?

I am facing similar issue with attaching CU to O-RAN SC RIC. I am using srsRAN commit 9d5dd742a. 

I am using single machine, not OpenStack. Core, CU/DU, and RIC are all colocated.

newt2105, 2025-03-13:
> Hi [@newt2105](https://github.com/newt2105) what changes did you make to fix the connection problem? Did you have some changes in CU/DU configs or in the O-RAN SC RIC from srsRAN docker-compose file?
> 
> I am facing similar issue with attaching CU to O-RAN SC RIC. I am using srsRAN commit [9d5dd74](https://github.com/srsran/srsRAN_Project/commit/9d5dd742a70e82c0813c34f57982f9507f1b6d5d). Are you using most recent commit of srsRAN?
> 
> I am using single machine, not OpenStack. Core, CU/DU, and RIC are all colocated.

Hi [Keilwerth11270](https://github.com/Keilwerth11270),
I use flexric to deploy RIC so I think i can't help much, but you can try to bind correct IP. It's worked for me

pgawlowicz, 2025-06-12:
@Keilwerth11270 any update on this issue?