srsran/srsRAN_Project #1089: Grafana not working

AlejandraOliver, 2025-03-09:
Hi, I follow this tutorial: https://docs.srsran.com/projects/project/en/latest/tutorials/source/near-rt-ric/source/index.html

and also I follow this in order to view results on grafana: https://docs.srsran.com/projects/project/en/latest/user_manuals/source/grafana_gui.html

Everything works fine excepto Grafana, the metric server is receiving the data but the infludb receive 0 in ul_brate when I'm doing:
_________________________
[  3] WARNING: did not receive ack of last datagram after 10 tries.

srsran@srsran-VirtualBox:~$ sudo ip netns exec ue1 iperf -c 10.45.1.1 -u -b 10M -i 1 -t 60

------------------------------------------------------------

Client connecting to 10.45.1.1, UDP port 5001

Sending 1470 byte datagrams, IPG target: 1121.52 us (kalman adjust)

UDP buffer size:  208 KByte (default)

------------------------------------------------------------

[  1] local 10.45.1.2 port 43153 connected with 10.45.1.1 port 5001

[ ID] Interval       Transfer     Bandwidth

[  1] 0.0000-1.0000 sec  1.25 MBytes  10.5 Mbits/sec

[  1] 1.0000-2.0000 sec  1.25 MBytes  10.5 Mbits/sec

[  1] 2.0000-3.0000 sec  1.25 MBytes  10.5 Mbits/sec

^C[  1] 0.0000-3.3467 sec  4.19 MBytes  10.5 Mbits/sec

[  1] Sent 2987 datagrams

read failed: Connection refused

read failed: Connection refused
___________________________________________________________________

Here I show that the metric server is catching data but influxdb no, so in Grafana dashboard I see 0bps:
___________________________________________________________________________

srsran@srsran-VirtualBox:~$ sudo tcpdump -i any udp port 55555 -A

tcpdump: data link type LINUX_SLL2

tcpdump: verbose output suppressed, use -v[v]... for full protocol decode

listening on any, link-type LINUX_SLL2 (Linux cooked v2), snapshot length 262144 bytes

16:43:33.677414 br-1346fa3c860b Out IP srsran-VirtualBox.49515 > 172.19.1.4.55555: UDP, length 794

E..6. @.@.?k.........k..."]_{

  "timestamp": 1741535013.676,

  "cell_metrics": {

    "error_indication_count": 0,

    "average_latency": 113,

    "latency_histogram": [45, 584, 217, 75, 31, 24, 5, 3, 3, 13]

  },

  "ue_list": [

    {

[... 526 lines cut ...]
release = 15

ue_category = 4

[nas]

apn = srsapn

apn_protocol = ipv4

[gw]

netns = ue1

ip_devname = tun_srsue

ip_netmask = 255.255.255.0

[gui]

enable = false

__________________________________________
(routing is configured fine too).
Thanks!!!

BGarcs, 2025-03-11:
I have the same problem, were you able to solve it?

AlejandraOliver, 2025-03-12:
Hi!! suddenly, turning it back on worked for me, but nothing special!

Sorry!!

closed 2025-03-13 by AlejandraOliver