srsran/srsRAN_Project #1096: srsRAN 7.2 split with Foxconn 7801 O-RU: DL U-plane messages are early.

mooncake-22, 2025-03-12:
## Description
I follow the srsRAN documents to setup srsRAN with DPDK support and integrate with Foxconn RPQN 7801 O-RU. From the open fronhaul(O-FH) statistics, the O-FH control plane packets arrived RU on time. The O-FH UL user plane packets arrived RU on time. However, the O-FH DL user plane packets arrived RU too early. 

I have tried to adjust the `t1a_max_up` and `t1a_min_up` values, here are the results:
- Using srsRAN default value:
  - srsRAN config:
    ```=
    ru_ofh:
      t1a_max_up: 196
      t1a_min_up: 80
    ```
  - O-RU statistices:
    ```=
    [2025-03-12 01:54:16.027] xRN: total=362200 c_early=0 c_on=29468 c_late=0 err_tci=0 err_ecpri=0 err_port=0 err_sct=0 err_total=182864
    [2025-03-12 01:54:16.228] xRN: u-plane u_early=107569 u_on=157453 u_late=84496
    [2025-03-12 01:54:17.030] xRN: total=445804 c_early=0 c_on=36268 c_late=0 err_tci=0 err_ecpri=0 err_port=0 err_sct=0 err_total=225034
    [2025-03-12 01:54:17.231] xRN: u-plane u_early=130797 u_on=191554 u_late=102763
    ```
  - Result:
    The packets will distribute on early, ontime and late.
 
- Increase `t1a_max_up` to prevent late:
  - srsRAN config:
    ```=
    ru_ofh:
      t1a_max_up: 460
      t1a_min_up: 80
    ```
  - O-RU statistices:
    ```=
    [2025-03-12 02:01:09.049] xRN: total=3109604 c_early=0 c_on=252936 c_late=0 err_tci=0 err_ecpri=0 err_port=0 err_sct=0 err_total=2856669
    [2025-03-12 02:01:09.150] xRN: u-plane u_early=2866458 u_on=0 u_late=840
    [2025-03-12 02:01:10.001] xRN: total=3193196 c_early=0 c_on=259736 c_late=0 err_tci=0 err_ecpri=0 err_port=0 err_sct=0 err_total=2933460
    [2025-03-12 02:01:10.153] xRN: u-plane u_early=2943284 u_on=0 u_late=840
    ```
  - Result:
    Most of the packets distributed on early, small portion on late, and nothing on ontime.

- Decrease `t1a_min_up` to prevent early:
  - srsRAN config:
    ```=
    ru_ofh:
      t1a_max_up: 460
      t1a_min_up: 0
    ```
  - O-RU statistices:
    ```=
    [2025-03-12 02:08:22.034] xRN: total=3035817 c_early=0 c_on=246936 c_late=0 err_tci=0 err_ecpri=0 err_port=0 err_sct=0 err_total=2788881
    [2025-03-12 02:08:22.185] xRN: u-plane u_early=2798712 u_on=0 u_late=840
    [2025-03-12 02:08:23.037] xRN: total=3119417 c_early=0 c_on=253736 c_late=0 err_tci=0 err_ecpri=0 err_port=0 err_sct=0 err_total=2865681
    [2025-03-12 02:08:23.188] xRN: u-plane u_early=2875512 u_on=0 u_late=840
    ```
  - Result:
    Most of the packets distributed on early, small portion on late, and nothing on ontime.

UE can detect MIB and SIB1, attach to the RAN and Core, and complete the signalling steps but not stable.

[image]

I would like to ask, do you encounter the similar situation and is there any method to solve the issue?  

## System Architecture

[image]

## Appendix
- srsRAN Running Log
  [gnb.log](https://github.com/user-attachments/files/19201476/gnb.log)
- srsRAN Configuration
  ```=
[... 184 lines cut ...]
  <!--            PTPV2 Related  -->
  <!--                           -->
  <!-- RRH_PTPV2_GRAND_MASTER_MODE: 0: Unicast over 1G, 1:Multicast over 1G; 2: Unicast over 10G; 3: Multicast over 10G -->
  RRH_PTPV2_GRAND_MASTER_MODE = 3
  <!-- RRH_PTPV2_JITTER_LEVEL: The estimated jitter of PTP time packets. 0:direct connection to GM/BC, 1:light, 2:medium, 3:heavy -->
  RRH_PTPV2_JITTER_LEVEL = 0
  <!-- RRH_PTPV2_VLAN_ID: VLAN ID of PTPv2.  0/1: No VLAN of PTPv2; [2~4092]: valid VLAN of PTPv2; >4092: Invalid and no VLAN will be applied -->
  RRH_PTPV2_VLAN_ID = 0
  <!-- RRH_PTPV2_IP_MODE: 4: IPv4 (default); 6: IPv6.  Note: this configuration must be defined before RRH_PTPV2_GRAND_MASTER_IP -->
  RRH_PTPV2_IP_MODE = 4
  <!-- RRH_PTPV2_GRAND_MASTER_IP: IP address of grand-master -->
  RRH_PTPV2_GRAND_MASTER_IP = 192.167.27.150
  <!-- RRH_PTPV2_SUB_DOMAIN_NUM: The sub-domain number -->
  RRH_PTPV2_SUB_DOMAIN_NUM = 24
  <!-- RRH_PTPV2_ACCEPTED_CLOCK_CLASS: The acceptable clockClass threshold [6~248] -->
  RRH_PTPV2_ACCEPTED_CLOCK_CLASS = 135
  <!-- RRH_TRACE_PERIOD: The period of trace log. It can be applied only when the start-trace-logs of M-plane is enabled -->
  RRH_TRACE_PERIOD = 10
  <!-- RRH_CLGC_EXPECTED_LOOP_GAIN: The clgc excepted loop gain: 0xFF: enable clgc after receive cu-plane data 300s, other value: use this value to enable clgc on dpd init -->
  RRH_CLGC_EXPECTED_LOOP_GAIN = 0xFF, 0xFF, 0xFF, 0xFF
  <!-- RRH_DL_IQ_SCALING: Set DL iq scaling value that 0x0: disable dl iq scaling-->
  RRH_DL_IQ_SCALING = 0x10001
  <!-- RRH_CFR_PEAK_THRESHOLD: Set CFR Peak Threshold -->
  RRH_CFR_PEAK_THRESHOLD = 0.5
  ```

sauka, 2025-03-14:
Hi @mooncake-22 

When you see early/late/on-time simultaneously it is a sign of a PTP not working correctly. From your description it seems that on the srsRAN side you are using a VF for O-FH traffic and a PF for the PTP  - this should help avoiding any synchronization issues, but please share an output of `ptp4l` and `phc2sys` processes **while** you are running the gNB to be sure it is working correctly. The expected `rms` values of correctly working PTP should look as described in this [tutorial](https://docs.srsran.com/projects/project/en/latest/tutorials/source/oranRU/source/index.html#ptp-configuration). 

Please make sure also that PTP is working fine on the Foxconn side.

Assuming the PTP is working correctly, you could try to tune the timing window. The values provided in the example config may not work ideally for you as they depend on many setup parameters. 

When tuning the `t1a_max_up` please bear in mind that the max t1a value for the U-Plane normally shouldn't be set to a value higher than the min value of respective C-Plane.

Even though the C-Plane is received on-time in your tests, I'd try to find the `t1a_max_cp_dl`/`t1a_min_cp_dl` making the window as wide as possible (and do the same for UL). The strategy you could take is described here https://github.com/srsran/srsRAN_Project/discussions/517#discussioncomment-9506259. Once you determine the `t1a_min_cp_dl` value, try setting the `t1a_max_up` to the same value or close to it. Just to give you another example, here are the values I used in the one of the setups with the same O-RU:

```
  t1a_max_cp_dl: 520
  t1a_min_cp_dl: 350
  t1a_max_cp_ul: 520
  t1a_min_cp_ul: 350
  t1a_max_up: 320
  t1a_min_up: 80
```

mooncake-22, 2025-03-14:
Hi @sauka ,
Thanks for your output. Here are the log for ptp4l and phc2sys.

When srsRAN is not running:
- ptp4l:
  ```=
  ● ptp4l.service - Precision Time Protocol (PTP) service
       Loaded: loaded (/lib/systemd/system/ptp4l.service; enabled; vendor preset: enabled)
       Active: active (running) since Wed 2025-03-12 09:49:52 CST; 2 days ago
     Main PID: 101206 (ptp4l)
        Tasks: 1 (limit: 134464)
       Memory: 284.0K
       CGroup: /system.slice/ptp4l.service
               └─101206 /usr/sbin/ptp4l -f /etc/ptp4l.conf
  
   三  14 21:53:10 srs72-PowerEdge-R740 ptp4l[101206]: [461006.139] rms    7 max   14 freq -13854 +/-  11 delay   141 +/-   0
   三  14 21:53:11 srs72-PowerEdge-R740 ptp4l[101206]: [461007.139] rms    8 max   17 freq -13869 +/-   9 delay   140 +/-   1
   三  14 21:53:12 srs72-PowerEdge-R740 ptp4l[101206]: [461008.139] rms    4 max    9 freq -13863 +/-   6 delay   140 +/-   1
   三  14 21:53:13 srs72-PowerEdge-R740 ptp4l[101206]: [461009.139] rms    6 max   14 freq -13868 +/-  10 delay   141 +/-   1
   三  14 21:53:14 srs72-PowerEdge-R740 ptp4l[101206]: [461010.139] rms    5 max   10 freq -13861 +/-   8 delay   140 +/-   1
   三  14 21:53:15 srs72-PowerEdge-R740 ptp4l[101206]: [461011.139] rms    5 max   11 freq -13865 +/-   9 delay   140 +/-   1
   三  14 21:53:16 srs72-PowerEdge-R740 ptp4l[101206]: [461012.139] rms    5 max   12 freq -13862 +/-   8 delay   141 +/-   1
   三  14 21:53:17 srs72-PowerEdge-R740 ptp4l[101206]: [461013.139] rms    8 max   13 freq -13865 +/-  12 delay   140 +/-   1
   三  14 21:53:18 srs72-PowerEdge-R740 ptp4l[101206]: [461014.139] rms    6 max   13 freq -13861 +/-  10 delay   140 +/-   1
   三  14 21:53:19 srs72-PowerEdge-R740 ptp4l[101206]: [461015.139] rms    8 max   16 freq -13850 +/-  11 delay   142 +/-   0
  ```
- phc2sys:
  ```=
  ● phc2sys.service - Synchronize system clock or PTP hardware clock (PHC)
       Loaded: loaded (/lib/systemd/system/phc2sys.service; enabled; vendor preset: enabled)
       Active: active (running) since Wed 2025-03-12 09:49:58 CST; 2 days ago
     Main PID: 101233 (phc2sys)
        Tasks: 1 (limit: 134464)
       Memory: 360.0K
       CGroup: /system.slice/phc2sys.service
               └─101233 /usr/sbin/phc2sys -a -r -r -n 24
  
   三  14 21:53:21 srs72-PowerEdge-R740 phc2sys[101233]: [461017.204] CLOCK_REALTIME phc offset       -10 s2 freq   +8048 delay    560
   三  14 21:53:22 srs72-PowerEdge-R740 phc2sys[101233]: [461018.204] CLOCK_REALTIME phc offset       -19 s2 freq   +8036 delay    571
   三  14 21:53:23 srs72-PowerEdge-R740 phc2sys[101233]: [461019.204] CLOCK_REALTIME phc offset       -24 s2 freq   +8025 delay    561
   三  14 21:53:24 srs72-PowerEdge-R740 phc2sys[101233]: [461020.204] CLOCK_REALTIME phc offset        -5 s2 freq   +8037 delay    563
   三  14 21:53:25 srs72-PowerEdge-R740 phc2sys[101233]: [461021.204] CLOCK_REALTIME phc offset         9 s2 freq   +8050 delay    562
   三  14 21:53:26 srs72-PowerEdge-R740 phc2sys[101233]: [461022.204] CLOCK_REALTIME phc offset        10 s2 freq   +8053 delay    564
   三  14 21:53:27 srs72-PowerEdge-R740 phc2sys[101233]: [461023.204] CLOCK_REALTIME phc offset       -20 s2 freq   +8026 delay    562
   三  14 21:53:28 srs72-PowerEdge-R740 phc2sys[101233]: [461024.204] CLOCK_REALTIME phc offset        10 s2 freq   +8050 delay    563
   三  14 21:53:29 srs72-PowerEdge-R740 phc2sys[101233]: [461025.205] CLOCK_REALTIME phc offset        21 s2 freq   +8064 delay    554
   三  14 21:53:30 srs72-PowerEdge-R740 phc2sys[101233]: [461026.205] CLOCK_REALTIME phc offset        -1 s2 freq   +8049 delay    564
  ```

When srsRAN is running:
- ptp4l:
  ```=
  ● ptp4l.service - Precision Time Protocol (PTP) service
       Loaded: loaded (/lib/systemd/system/ptp4l.service; enabled; vendor preset: enabled)
       Active: active (running) since Wed 2025-03-12 09:49:52 CST; 2 days ago
     Main PID: 101206 (ptp4l)
        Tasks: 1 (limit: 134464)
       Memory: 528.0K
       CGroup: /system.slice/ptp4l.service
               └─101206 /usr/sbin/ptp4l -f /etc/ptp4l.conf
  
   三  14 21:55:13 srs72-PowerEdge-R740 ptp4l[101206]: [461129.202] rms   37 max   59 freq -13877 +/-  48 delay   169 +/-  30
   三  14 21:55:14 srs72-PowerEdge-R740 ptp4l[101206]: [461130.202] rms   17 max   28 freq -13889 +/-  27 delay   151 +/-  15
   三  14 21:55:15 srs72-PowerEdge-R740 ptp4l[101206]: [461131.202] rms   38 max   63 freq -13965 +/-  42 delay   185 +/-  28
   三  14 21:55:16 srs72-PowerEdge-R740 ptp4l[101206]: [461132.202] rms   28 max   51 freq -13886 +/-  34 delay   159 +/-  20
   三  14 21:55:17 srs72-PowerEdge-R740 ptp4l[101206]: [461133.202] rms   26 max   42 freq -13949 +/-  32 delay   187 +/-  25
   三  14 21:55:18 srs72-PowerEdge-R740 ptp4l[101206]: [461134.202] rms   30 max   66 freq -13903 +/-  48 delay   195 +/-  24
   三  14 21:55:19 srs72-PowerEdge-R740 ptp4l[101206]: [461135.202] rms   26 max   54 freq -13859 +/-  27 delay   149 +/-  12
   三  14 21:55:20 srs72-PowerEdge-R740 ptp4l[101206]: [461136.202] rms   12 max   32 freq -13896 +/-  17 delay   149 +/-  12
   三  14 21:55:21 srs72-PowerEdge-R740 ptp4l[101206]: [461137.202] rms   23 max   38 freq -13918 +/-  34 delay   155 +/-  16
[... 27 lines cut ...]
For the ptp4l on the foxconn RU:
```=
root@arria10:~/test/firmware# systemctl status ptp4l
● ptp4l.service - Run ptp4l
   Loaded: loaded (/etc/systemd/system/ptp4l.service; static; vendor preset: enabled)
   Active: active (running) since Wed 2025-03-12 01:51:43 UTC; 2 days ago
[... cut ...]
Mar 14 13:58:42 arria10 ptp4l[7355]: [0.000] state=3 rms    6 max    9 freq   -290 +/-   0 delay    ...   0
Mar 14 13:58:43 arria10 run_ptp4l.qse-eth.sh[7350]: ptp4l[2025/03/14/13:58:43]: state=3 rms    7 max ...  0
Mar 14 13:58:43 arria10 ptp4l[7355]: [0.000] state=3 rms    7 max   11 freq   -290 +/-   0 delay    ...   0
```

Observation:
When srsRAN is running, the rms of ptp4l is increased. On the otherhand, the rms of ptp4l is smaller than 10. For the rms of ptp4l on the RU side, it seems to be stable.

mooncake-22, 2025-04-23:
Hi @sauka ,

The O-FH timing issue persists regardless of configuration changes. However, after adjusting the iq_scaling parameter, the UE is now able to attach to the cell immediately.

I also tested DL/UL throughput using iperf3. Please see the results below:

- DL:
  [image]

- UL:
  [image]

The downlink throughput reaches around 920 Mbps. The channel quality appears good, and the MCS hovers around 20. However, it doesn't scale up to 256QAM.

Do you think this limitation could be related to CPU frequency, or might there be other parameters worth checking?

sauka, 2025-04-24:
> Hi [@sauka](https://github.com/sauka) ,
>
> The O-FH timing issue persists regardless of configuration changes. However, after adjusting the iq_scaling parameter, the UE is now able to attach to the [...]

Hi @mooncake-22 ,

I assume you have been able to resolve the timing problem and all the OFH traffic arrives on-time at the RU. Could you share your latest gNB config and the log please? We can double check whether there are warnings related to real-time performance of your server. Based on the CQI reported when doing DL iperf I'd say the BLER is caused by poor RF conditions. But it may also be caused by the wrong `iq_scaling`

sauka, 2025-05-02:
Hi @mooncake-22 , 

I was looking for the RPQN-7801 config that we have been using successfully the last time, and I noticed a few differences with the config you shared. The FW version in our test was `v3.1.15q.551v0706`. 
These are the parameters set differently, you may give it a try:

`RRH_TX_POWER = 24, 24`

`RRH_RX_ATTENUATION = 0.0, 0.0, 0.0, 0.0`

Actually the important one was `RRH_DL_IQ_SCALING`, by default it was set to 0 and the UE was not able to attach. Setting it to 0x10001 fixed the problem. This value was working perfectly for 2T2R mode, however not for 4T. I'd suggest contacting Foxconn regarding the recommended config for 4T4R mode.

mooncake-22, 2025-05-03:
Hi @sauka ,

Thanks for the information. Currently, our Foxconn RPQN O-RU is being used by another team, so I’m unable to test it right away. I will perform the test and update you as soon as the O-RU becomes available. 

By the way, I’m curious about the use of `RRH_DL_IQ_SCALING`. I noticed there’s an `iq_scaling` parameter in the srsRAN configuration. Does this mean we need to configure it on both sides?

sauka, 2025-06-04:
> Hi [@sauka](https://github.com/sauka) ,
>
> Thanks for the information. Currently, our Foxconn RPQN O-RU is being used by another team, so I’m unable to test it right away. I will perform the test and [...]

Hey @mooncake-22 ,

Sorry I missed your question - yes the `iq_scaling` has to be set as well, we normally set it to 1.0. It gave us good results in 2x2 tests when we've set the `RRH_DL_IQ_SCALING` to `0x10001`

mooncake-22, 2025-06-09:
Hi @sauka ,

Thanks for the reply.

By the way.The RU is back, will attach the config and log soon.

mooncake-22, 2025-06-11:
Hi @sauka,
I’ve retested the setup with srsRAN and the Foxconn RU, but unfortunately, the issue still persists.
As the log file is too large to attach directly, I’ve uploaded both the gNB configuration and the log files to Google Drive. Please find them [here](https://drive.google.com/drive/folders/1An_nWt3d8Nl_AxKKx_C5p00Dd6qqUA7I?usp=sharing).

[image]

Let me know if you need any further information.