srsran/srsRAN_Project #1097: Testing Traffic of srsRAN gNB and srsUE ZMQ Radios under Channel Effects

womper25, 2025-03-12:
I am trying to study the effects of different channels to traffic generated by the stack (using srsrRAN gNB, srsUE, and ZMQ) which can be characterized by filter taps. With that, I have implemented the channel block as shown below and have only changed the filter taps variable to model a multi-path channel. 

[image]

The filter taps were generated by using a 5G Channel Simulator (e.g., PyWiCh) to emulate a certain scenario, for this case, Urban Micro. However upon running the stack, when the block is placed in the Downlink direction, the UE fails to connect with the gNB as shown below (even while the flowgraph is running).  (Note: I have already verified that the ideal scenario works and that the UE can connect with the gNB when bypassing the channel block) 

[image]

I have tried creating an Embedded Python block that emulates an ideal channel for a set amount of time just for the UE to connect, then switches to the Channel using the same filter taps. However, this method also causes the UE to disconnect from the gNB as shown below.
 
[image]

I would like to know what are some fixes or workarounds for the UE to still connect despite being affected by a channel.

---------------------------------------------------------------------------------------------------------------------------

Here are the configuration files as well as the GRC Flowgraph with the embedded python block:

UE: [ue1_zmq.conf.txt](https://github.com/user-attachments/files/19213405/ue1_zmq.conf.txt)
gNB: [gnb_zmq.yaml.txt](https://github.com/user-attachments/files/19213409/gnb_zmq.yaml.txt)
Flowgraph: [GRC_flowgraph_channel.txt](https://github.com/user-attachments/files/19213427/GRC_flowgraph_channel.txt)

Additional Information:
- the filter taps were generated using a UE distance of 14 m from the BS
- the filter taps are: `[-0.00971564+0.01062702j, -0.04883288-0.05761252j, -0.04813277+0.01374759j,  0.09444031+0.05342406j, -0.0653316 -0.06339766j,  0.05448993-0.05245771j, -0.00529209-0.04225221j,  0.07695623-0.07572819j, -0.03811407+0.05733179j]`

pgawlowicz, 2025-03-12:
could you check with this patch?
https://github.com/srsran/srsRAN_Project/issues/826#issuecomment-2370588502

womper25, 2025-03-13:
I have tried applying the patch (and verified it by checking the modified library files), but it still fails to connect in the downlink direction. Here are the logs for the gNB and UE:

UE log: [ue1.log](https://github.com/user-attachments/files/19222562/ue1.log)
gNB log:  [gnb.log](https://github.com/user-attachments/files/19222559/gnb.log) 

> could you check with this patch? [#826 (comment)](https://github.com/srsran/srsRAN_Project/issues/826#issuecomment-2370588502)

jcdagami13, 2025-03-23:
Hi, may I ask if there's an update on this issue? I'm also experiencing a similar case where srsUE cannot connect to srsgnb in the presence of multipath. Our team believes that it might be related to the tx power of the gnb? We've already increased tx gain in the gnb config file but the issue still persists. Is there a way to manually set the transmit power, instead of the gain? We've observed that the average tx power in ru_metrics is not changing when modifying the tx gain in the gnb config file.

[image]