srsran/srsRAN_Project #1102: COTS UE having problem attaching to network, UE Release (5G SA COTS UE)

meong3000, 2025-03-13:
## Issue Desc 
AMF - gnB connection established, but cannot successfully attach

## Setup Details

 UE: Oppo Reno 8
OS: Ubuntu 22.04 (2 PC for RAN & Core)
RAN: srsrAN 5G
Core: Open5Gs 
RF Frontend: B205mini-i (UHD 4.7.0)

## Expected Behavior
UE attached and gets a stable connection, PDU session established, etc

## Actual Behavior
Network shown on UE interface, but cannot attach to it

## Steps to reproduce the problem
Log & configuration files attached
[conflog_gNB_UE_AMF.zip](https://github.com/user-attachments/files/19229721/conflog_gNB_UE_AMF.zip)

## Additional Info
Already done srsran_performance script on the srsRAN machine.
Also when i used debug for log, i get this continuous log info

``` bash 2025-03-10T05:58:34.814117 [FAPI    ] [D] [   156.0] Slot.indication
2025-03-10T05:58:34.814138 [FAPI    ] [D] [   156.0] DL_TTI.request slot=156.0, is_last_message_in_slot=true
2025-03-10T05:58:34.815146 [FAPI    ] [D] [   156.1] Slot.indication
2025-03-10T05:58:34.815161 [FAPI    ] [D] [   156.1] DL_TTI.request slot=156.1, is_last_message_in_slot=true
2025-03-10T05:58:34.816154 [FAPI    ] [D] [   156.2] Slot.indication
2025-03-10T05:58:34.816187 [FAPI    ] [D] [   156.2] DL_TTI.request slot=156.2, is_last_message_in_slot=true
- NZP-CSI-RS crbs=0:52 row=1 symbL0=4 symbL1=2 scramb_id=1
- NZP-CSI-RS crbs=0:52 row=1 symbL0=8 symbL1=2 scramb_id=1
2025-03-10T05:58:34.817174 [FAPI    ] [D] [   156.3] Slot.indication
```

Then i rebuild the srsRAN to use the relwithdebinfo, but the attached logs still appear. Thanks!

robertfalkenberg, 2025-03-14:
please repeat the experiment with mac+ngap pcaps enabled and share them (and the logs of open5gs logs and gnb)

```yml
pcap:
  mac_enable: true
  mac_filename: /tmp/gnb_mac.pcap
  ngap_enable: true
  ngap_filename: /tmp/gnb_ngap.pcap
```

meong3000, 2025-03-19:
Also this is the trace output, could it be the problem is related to timing advance? Ty!

meong3000, 2025-03-20:
Nevermind, b205mini-i works fine, the problem was my programmable sim card did not support the 5G network, i changed it and it works fine!

Alan1321, 2025-03-25:
> Nevermind, b205mini-i works fine, the problem was my programmable sim card did not support the 5G network, i changed it and it works fine!

What sim card did you use after?

meong3000, 2025-03-26:
> What sim card did you use after?

at first i was using OYEITIMES (but i used the 4G version), then i change to a blank SIM JavaCard from a local e-commerce and it works @Alan1321