srsran/srsRAN_Project #1113: This is my `cu` configuration for mobility

h57jafari, 2025-03-18:
This is my `cu` configuration for mobility

```
gnb_id: 531
gnb_id_bit_length: 32
ran_node_name: cu_cp_01

#slicing:                           
#  -                                   
#    sst: 1                               
#    sd: 0                              

amf:
  addr:  192.168.68.42
  bind_addr: 192.168.68.210

cu_cp:
  inactivity_timer: 7200
  mobility:
    trigger_handover_from_measurements: true
    cells:
      - nr_cell_id: 0x2130
        periodic_report_cfg_id: 1
        ncells:
          - nr_cell_id: 0x2131
            report_configs: [ 2 ]
      - nr_cell_id: 0x2131
        periodic_report_cfg_id: 1
        ncells:
          - nr_cell_id: 0x2130
            report_configs: [ 2 ]
    report_configs:
      - report_cfg_id: 1
        report_type: periodical
        report_interval_ms: 1024
      - report_cfg_id: 2
        report_type: event_triggered
        event_triggered_report_type: a3
        meas_trigger_quantity: rsrp
        meas_trigger_quantity_offset_db: 2
        hysteresis_db: 0
        time_to_trigger_ms: 100
        report_interval_ms: 1024

  f1ap:
    bind_addr: 192.168.68.210

cu_up:
  nru:
    bind_addr: 192.168.68.212
    udp_max_rx_msgs: 256

log:
  filename: /tmp/cu.log
  all_level: debug

pcap:
  ngap_enable: false
  ngap_filename: /tmp/cu_ngap.pcap
```

and 2 `DUs`

DU1

```
gnb_id: 531
gnb_id_bit_length: 32
gnb_du_id: 0

f1ap:
  cu_cp_addr: 192.168.68.210
  bind_addr: 192.168.68.19
nru:
  bind_addr: 192.168.68.19

ru_sdr:
  device_driver: uhd
  device_args: type=b200,num_recv_frames=64,num_send_frames=64
  srate: 23.04
  otw_format: sc12
  tx_gain: 53
  rx_gain: 40

cell_cfg:
  dl_arfcn: 632628                                                # ARFCN of the downlink carrier>
  band: 78                                                        # The NR band.
  channel_bandwidth_MHz: 20                                       # Bandwith in MHz. Number of PR>
  common_scs: 30                                                  # Subcarrier spacing in kHz use>
  plmn: "00101"                                                   # PLMN broadcasted by the gNB.
  tac: 7                                                          # Tracking area code (needs to >
  pci: 1                                                          # Physical cell ID.
  nof_antennas_dl: 1
  nof_antennas_ul: 1
  sector_id: 0
  pdsch:
    mcs_table: qam256

log:
  filename: /tmp/du.log
  all_level: warning

pcap:
  mac_enable: false
  mac_filename: /tmp/du_mac.pcap
  f1ap_enable: false
  f1ap_filename: /tmp/du_f1ap.pcap
  f1u_enable: false
  f1u_filename: /tmp/du_f1u.pcap
```

DU2

```
gnb_id: 531
gnb_id_bit_length: 32
gnb_du_id: 1

f1ap:
  cu_cp_addr: 192.168.68.210
  bind_addr: 192.168.68.152
nru:
  bind_addr: 192.168.68.152

ru_sdr:
  device_driver: uhd
  device_args: type=b200,num_recv_frames=64,num_send_frames=64
  srate: 23.04
  otw_format: sc12
  tx_gain: 53
  rx_gain: 40

cell_cfg:
  dl_arfcn: 632628                                                # ARFCN of the downlink carrier>
  band: 78                                                        # The NR band.
  channel_bandwidth_MHz: 20                                       # Bandwith in MHz. Number of PR>
  common_scs: 30                                                  # Subcarrier spacing in kHz use>
  plmn: "00101"                                                   # PLMN broadcasted by the gNB.
  tac: 7                                                          # Tracking area code (needs to >
  pci: 2                                                          # Physical cell ID.
  nof_antennas_dl: 1
  nof_antennas_ul: 1
  sector_id: 1
  pdsch:
    mcs_table: qam256

log:
  filename: /tmp/du.log
  all_level: warning

pcap:
  mac_enable: false
  mac_filename: /tmp/du_mac.pcap
  f1ap_enable: false
  f1ap_filename: /tmp/du_f1ap.pcap
  f1u_enable: false
  f1u_filename: /tmp/du_f1u.pcap
```

I put two B210s in separated  room and I set ` meas_trigger_quantity_offset_db` to 2. I also reduce tx and rx gain.  It can do handover. I test many times by moving the UE (Samsung A33 5G) between two rooms. However, I don't know why the UE cannot connect to the internet using this configurations while the UE can connect to the internet when using CU-DU split without handover.

_Originally posted by @fllay in https://github.com/srsran/srsRAN_Project/discussions/827#discussioncomment-10748247_

h57jafari, 2025-03-18:
For this CU+DU1+DU2 split setup, what other configurations need to be done in amf.yml and upf.yml or else. Thank you.

h57jafari, 2025-03-31:
Since you are using USRP, did you use external clock like OctoClock or GPSDO?

joelsernamoreno, 2025-03-31:
No, I have not used it, but I can use the Leo Bodnar LBE-1420

h57jafari, 2025-03-31:
Yes, try to use it. In my case even without external clock, UE did not attache to the network, you are one step ahead.

h57jafari, 2025-03-31:
also, please share the IP of your machines, CU, DU1, DU2, ....