srsran/srsRAN_Project #1116: Error when connect split8 srsRAN DU with USRP N321

Thanhlongnguyen1012, 2025-03-19:
Hi everyone, I connect srsDU with USRP N321, it's worked but after a short time running, it's has the error below:

```
ubuntu@du:~/srsRAN_Project/build/apps/du$ sudo ./srsdu -c du.yml

--== srsRAN DU (commit 2be82d8ea) ==--

Lower PHY in quad executor mode.
Available radio types: uhd and zmq.
[INFO] [UHD] linux; GNU C++ version 11.4.0; Boost_107400; UHD_3.15.0.0-74-ge35f66e8
[INFO] [LOGGING] Fastpath logging disabled at runtime.
No device type given, found device with address 'mgmt_addr=192.168.180.94,type=n3xx,product=n320,serial=3258436,claimed=False'
Making USRP object with args 'addr=192.168.180.94,clock_source=internal,send_frame_size=1472,recv_frame_size=1472,master_clock_rate=245.76e6,type=n3xx'
[INFO] [MPMD] Initializing 1 device(s) in parallel with args: mgmt_addr=192.168.180.94,type=n3xx,product=n320,serial=3258436,claimed=False,addr=192.168.180.94,clock_source=internal,send_frame_size=1472,recv_frame_size=1472,master_clock_rate=245.76e6
[INFO] [MPM.PeriphManager] init() called with device args `product=n320,mgmt_addr=192.168.180.94,recv_frame_size=1472,master_clock_rate=245.76e6,time_source=internal,clock_source=internal,send_frame_size=1472'.
[INFO] [MPM.Rhodium-0] init() called with args `product=n320,mgmt_addr=192.168.180.94,recv_frame_size=1472,master_clock_rate=245.76e6,time_source=internal,clock_source=internal,send_frame_size=1472'
[INFO] [MPM.Rhodium-1] init() called with args `product=n320,mgmt_addr=192.168.180.94,recv_frame_size=1472,master_clock_rate=245.76e6,time_source=internal,clock_source=internal,send_frame_size=1472'
[INFO] [MPM.Rhodium-0.init.LMK04828] LMK initialized and locked!
[INFO] [MPM.Rhodium-1.init.LMK04828] LMK initialized and locked!
[INFO] [MPM.Rhodium-0.DAC37J82] DAC PLL Locked!
[INFO] [MPM.Rhodium-0.AD9695] ADC PLL Locked!
[INFO] [MPM.Rhodium-0.init] JESD204B Link Initialization & Training Complete
[INFO] [MPM.Rhodium-1.DAC37J82] DAC PLL Locked!
[WARNING] [UDP] The recv buffer could not be resized sufficiently.
Target sock buff size: 6250000 bytes.
Actual sock buff size: 212992 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.rmem_max=6250000
[WARNING] [UDP] The current recv_buff_size of 212992 is less than the minimum recommended size of 816000 and may result in dropped packets on some NICs
[WARNING] [UDP] The recv buffer could not be resized sufficiently.
Target sock buff size: 6250000 bytes.
Actual sock buff size: 212992 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.rmem_max=6250000
[WARNING] [UDP] The current recv_buff_size of 212992 is less than the minimum recommended size of 816000 and may result in dropped packets on some NICs
[WARNING] [UDP] The recv buffer could not be resized sufficiently.
Target sock buff size: 6250000 bytes.
Actual sock buff size: 212992 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.rmem_max=6250000
[WARNING] [UDP] The current recv_buff_size of 212992 is less than the minimum recommended size of 816000 and may result in dropped packets on some NICs
[INFO] [MPM.Rhodium-1.AD9695] ADC PLL Locked!
[INFO] [MPM.Rhodium-1.init] JESD204B Link Initialization & Training Complete
[WARNING] [UDP] The recv buffer could not be resized sufficiently.
Target sock buff size: 6250000 bytes.
Actual sock buff size: 212992 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.rmem_max=6250000
[WARNING] [UDP] The current recv_buff_size of 212992 is less than the minimum recommended size of 816000 and may result in dropped packets on some NICs
[INFO] [0/Replay_0] Initializing block control (NOC ID: 0x4E91A00000000004)
[WARNING] [UDP] The recv buffer could not be resized sufficiently.
Target sock buff size: 6250000 bytes.
Actual sock buff size: 212992 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.rmem_max=6250000
[WARNING] [UDP] The current recv_buff_size of 212992 is less than the minimum recommended size of 816000 and may result in dropped packets on some NICs
[INFO] [0/Radio_0] Initializing block control (NOC ID: 0x12AD100000000320)
[WARNING] [UDP] The recv buffer could not be resized sufficiently.
Target sock buff size: 6250000 bytes.
Actual sock buff size: 212992 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.rmem_max=6250000
[WARNING] [UDP] The current recv_buff_size of 212992 is less than the minimum recommended size of 816000 and may result in dropped packets on some NICs
[INFO] [0/Radio_1] Initializing block control (NOC ID: 0x12AD100000000320)
[WARNING] [UDP] The recv buffer could not be resized sufficiently.
Target sock buff size: 6250000 bytes.
Actual sock buff size: 212992 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.rmem_max=6250000
[WARNING] [UDP] The current recv_buff_size of 212992 is less than the minimum recommended size of 816000 and may result in dropped packets on some NICs
[... 52 lines cut ...]
Target sock buff size: 6250000 bytes.
Actual sock buff size: 212992 bytes.
See the transport application notes on buffer resizing.
Please run: sudo sysctl -w net.core.rmem_max=6250000
Cell pci=1, bw=20 MHz, 1T1R, dl_arfcn=499200 (n41), dl_freq=2496 MHz, dl_ssb_arfcn=498270, ul_freq=2496 MHz

F1-C: Connection to CU-CP on 10.0.10.92:38472 completed
==== DU started ===
Type <h> to view help

Late: 11492; Underflow: 4795; Overflow: 78;
Late: 30587; Underflow: 13058; Overflow: 93;
Late: 29366; Underflow: 13265; Overflow: 104;
Late: 32855; Underflow: 14503; Overflow: 111;
Late: 36532; Underflow: 15637; Overflow: 131;
Late: 35815; Underflow: 16088; Overflow: 111;
Late: 37133; Underflow: 15517; Overflow: 125;
Late: 24444; Underflow: 10460; Overflow: 108;
Late: 33431; Underflow: 13095; Overflow: 120;
srsRAN FATAL ERROR: Failed to execute uplink processing task.
Aborted

``` 
Do anyone have same error? 
Thanks

asheeshtripathi, 2025-03-20:
Please run the performance script before you start the DU. The srsran_performance script is in srsRAN_Project/scripts folder. There are a lot of lates and underflows. Also, make sure the MTU size on the DU to USRP (10G) interface is set to 9000.

Thanhlongnguyen1012, 2025-03-26:
It's not optimized, do you have any other way?

[image]

asheeshtripathi, 2025-03-26:
There are a lot of late and underflows still. More optimizations will be needed. Please refer to other issues/discussions to see similar issues and how they are resolved.