srsran/srsRAN_Project #1118: UE not able to attach to Benetel RAN 650

Keilwerth11270, 2025-03-19:
Our Setup:
We have a Benetel O-RU RAN 650 which we are integrating with srsRAN CU+DU. We are able to get PTP sync locked on the O-RU. After running srsRAN gNB, the UE is able to see the network but not able to attach.

Testbench and Connectivity:

Server machine: 
Intel(R) Xeon(R) Silver 4416+, 40 cores, Ubuntu 22.04.5 LTS with kernel 5.15.0-1032-realtime
Core: Open5GS
CU+DU: srsRAN: 4X2 configuration, 100/40/20MHz bandwidth, DPDK is not enabled.
O-RU: RAN 650, RAN650-1v1.2.2-2fa04bc, Band n78

The Core and CU-DU are co-located in the server. The DU and O-RU are connected via Intel NIC E810XXVDA4TGG1, which provides a hardware timestamp for PTP sync. We have set the MTU on the interface to 9600. It is connected via a multimode fiber with full-duplex 10G SFP. The O-RU is powered on and is in sync state. The RF ports of O-RU are connected with 10-Watt 30 dB attenuators, and the attenuator output is connected to sub-6 multiband antennas. We are using COTS UEs (OnePlus 8T and Samsung A23, both previously connected to USRP X310, B210, and N310).

PTP status:

[image]

[image]

[image]

O-RU side:

[image]

[image]

[image]

With bandwidth configurations of 20, 40, and 100 MHz, the UE can see the network but is not able to attach. The gNB logs do not show anything in info mode, but when set to debug, we see the following:

[image]

If only PRACH is grepped:

[image]

Attached are the gNB and O-RU config files and gNB log files.

Other than this, on FH interface we have set MTU to be 9600 but when we start the gNB it goes back to 9000. If we again set the MTU size to be 9600 while the gNB is running, we sometimes see an error on ptp4l process regarding VLAN disabled, though sometimes it works.

ninjab3s, 2025-04-22:
Hello @Keilwerth11270,

the RAN seems to be configured properly from what I can see here. Can you share the AMF logs when attaching a UE? Also, to clarify - Do you use a VF for the gnb or the same interface for PTP and gnB?

Thanks!

Keilwerth11270, 2025-04-22:
We have 30 dB 10W attenuators connected to the antenna ports and then output is fed to RF Chamber where we keep UE. Do we still need to lower the tx_power_dbm? 

We tried with IQ scaling of 16 with 100MHz BW, 4x2 configuration, and long PRACH. UE was able to scan the network, but failed to attach. We do not see any PRACH messages in the gNB logs. Neither in the Core console prints. 

However, we have **not** set register for the long PRACH in the O-RU [as mentioned in the srsRAN 7.2 documentation for Benetel](https://docs.srsran.com/projects/project/en/latest/tutorials/source/oranRU/source/rus/r550.html):

[image]

Then we tried short PRACH format with 2x2 and 20MHz BW. UE was unable to scan the network. Same observation with 2x2 100MHz BW and 4x2 100MHz BW with short PRACH format. We only had UE able to scan the network with the configuration file shared earlier.

We are using E810 which is directly connected to the O-RU via multimode fiber over 10G interface. We have set MTU size to be 9600 but starting the gNB results in MTU size reverting to 9000 and we manually change it back to 9600 after starting gNB each time. There is no DPDK or external GPS however the O-RU gets PTP Sync via E810 hardware timestamp.

## Setup Diagrams

### Network Architecture
[image]

### Physical Setup
[image]

Keilwerth11270, 2025-04-23:
We upgraded the firmware version of the Benetel O-RU as recommended:

```
root@benetelru:~# cat /etc/benetel-rootfs-version
RAN650-1v1.3.11-M-1c0e9d8
root@benetelru:~#
```

We tried with tx_power 27, 24, and 21 with IQ scaling set to 8. We encountered similar issue, UE is able to scan the network but cannot attach. We don't see any PRACH message in the gNB logs as well as in the Core logs. We also tried changing the PLMN from 00101 to 90170 and found that UE is able to scan the network with corresponding PLMN but still could not attach. We tried to attach to the network with a OnePlus 8T, OnePlus Nord 5G, and Quectel RM510 Modem. All of them have been tested with SDR-based srsRAN 5G networks. The Open5GS configuration is identical to the SDR-based network configurations. 

The below screenshot shows the Open5GS Console print.
[image]

Currently we are using the same OFH interface with no virtual function (there is no switch, NIC interface is directly connected to the O-RU via fiber), but we have tried using VLAN Tag in the past. But we get similar observation for UE attach even with this.
```
echo 1 > /sys/class/net/enp13s0f0/device/sriov_numvfs
root@cci-creek-01:/home/cci-creek-01/github/Benetel# ip link show enp13s0f0
14: enp13s0f0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 9000 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether 50:7c:6f:58:19:98 brd ff:ff:ff:ff:ff:ff
    vf 0     link/ether 6e:68:9f:fd:d4:75 brd ff:ff:ff:ff:ff:ff, spoof checking on, link-state auto, trust off
```
[Attached](https://github.com/user-attachments/files/19875442/ran650_srsran_open5gs.zip) are the configuration files for the RAN650, Open5GS Core, and gNB. 

Our UEs/Modem are not rooted. But they have been tested with srsRAN. To this point we have not tested with DPDK.

Keilwerth11270, 2025-05-01:
Hi @ninjab3s do you have any further recommendations for troubleshooting on this?

ninjab3s, 2025-05-07:
It seems like a power issue to me! Please have a look at the following two configs. Its from cabled setup, modem cabled directly to the R650 with 40 or 50db attenuation. Your setup shouldn't be too different. Try increasing the power on the DU side until you see a cell.

[configs.zip](https://github.com/user-attachments/files/20080357/configs.zip)

Also, I have seen that you are using 90170 as PLMN. I have see some phones not attaching to this PLMN. We had very good results using 00101 so far. I would also change this in your setup or use real PLMNs since you are working with a shielded environment.

Let me know how it goes and please send all config files and logs in case it doesnt work as you did the last time. Thanks!

ninjab3s, 2025-05-12:
Hey @Keilwerth11270!

After our last email exchange the UE was able to see the cell. Can we close this issue? Please open another one in case you have different issues. Thanks!

Keilwerth11270, 2025-05-12:
We've successfully achieved UE attachment using srsRAN with Benetel RAN650 O-RU:

- Stable UE attach with 100 MHz BW with both UL-heavy and DL-heavy TDD patterns.
- With 20MHz BW and DL-heavy TDD pattern, UE was attached but kept connecting and re-connecting (could be due to some RF issue)

## How we resolved the problem:
Overall, our issue was resolved after doing the following:
1. Upgrading srsRAN to latest version (commit starting with 122a...)
2. Upgrading O-RU firmware to RAN650-1v1.4.0
3. Enabling DPDK

closed 2025-05-12 by Keilwerth11270

alektoja, 2025-08-15:
How did you change the setup so that DPDK and PTP work simultaneously?