srsran/srsRAN_Project #1119: Multiple UE's and gNODEb in different machines

j3les-0, 2025-03-19:
im trying to connect multiple UE's in a IP A to a gNb running in a different machine with IP B. gnuradio did not work, as was expected. i tried using a proxy, but it wasnt enough either. Is there a config example of this documented or any form or tutorial to make this system functional? if not, what would be the possible requirements for it to happen?

pgawlowicz, 2025-03-24:
with zmq? you need to update the IPs in all configs and also gnuradio graph. See here:
https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html#run-zmq-based-setup-on-two-host-machines

closed 2025-03-24 by pgawlowicz

j3les-0, 2025-03-25:
First of all, thanks for the response.
I apologize, I wasn't clear. Gnuradio wasn't enough for communication beetwen multiples UE's in a computer to a gnb running in another computer. I did test every normal option before starting this issue. let me explain further: 

running multiple UE's and a gNODEb in the same machine works fine with gnuradio.

running only one UE and a gNODEb in different machines works fine, following the tutorial in the documentation.

running multiple UE's and a gNODEb in different machines does not work using gnuradio.

That's the reason why I was asking for any form of tutorial regarding this last possible setup. As mentioned, I tried running a different broker, but it was not enough.

pgawlowicz, 2025-03-27:
Did you set the addresses in gnuradio ZMQ blocks correctly?

[image]