srsran/srsRAN_Project #1127: simple_RC_xapp and simple_xapp do not work, disconnecting UE

h57jafari, 2025-03-24:
Hi,
I am working with OS xApps. KPM xApp works fine, it seems other two xApps have problem and make DL parameters including "brate", "mcs", "ok" go to zeros.

newt2105, 2025-03-24:
hi [h57jafari](https://github.com/h57jafari), 
Do you use USRP to connect with UE?

h57jafari, 2025-03-24:
Hi, No. I am using ZMQ between UE and gNB.

h57jafari, 2025-03-24:
here is the output of gNB terminal:

[image]

h57jafari, 2025-03-24:
it seems the problem related to RMR_ERR_NOENDPT.
the ric_e2term reporting issue that can not find the endpoint. 
the ric_submgr report "No E2 connection for ranName gnbd_001_001_00019b_0. 
where can modify it.

h57jafari, 2025-03-24:
I am not sure if these logs also related:

gNB_DU_ID is not null
cuupid =-1
duid =0
ranName =gnbd_001_001_00019b
1742795187955 1/RMR [INFO] ric message routing library on SI95 p=38000 mv=3 flg=00 id=a (f447e29 4.9.4 built: Dec 13 2023)
1742795332294 1/RMR [INFO] ric message routing library on SI95 p=38000 mv=3 flg=00 id=a (f447e29 4.9.4 built: Dec 13 2023)

h57jafari, 2025-03-24:
and this one:

gNB_DU_ID is not null
cuupid =-1
duid =0
ranName =gnbd_001_001_00019b
1742756126987 1/RMR [INFO] ric message routing library on SI95 p=38000 mv=3 flg=00 id=a (f447e29 4.9.4 built: Dec 13 2023)
{"ts":1742756200228,"crit":"ERROR","id":"E2Terminator","mdc":{"PID":"125770154309376","POD_NAME":"e2term_pod","CONTAINER_NAME":"ric_e2term","SERVICE_NAME":"ric_e2term_service","HOST_NAME":"ric_e2term_host","SYSTEM_NAME":"oran_sc_ric"},"msg":"Read error, Bad file descriptor "}

h57jafari, 2025-03-24:
this is the output of "$ sudo docker logs ric_rtmgr_sim" that I have now:

2025/03/24 13:55:17 Server started on port 12020
2025/03/24 13:55:18 POST /ric/v1/handles/e2t  body: {"E2TAddress":"10.0.2.10:38000"} elapsed: 65.706µs
2025/03/24 13:55:29 POST /ric/v1/handles/associate-ran-to-e2t  body: [{"E2TAddress":"10.0.2.10:38000","ranNamelist":["gnbd_001_001_00019b_0"]}] elapsed: 34.929µs
2025/03/24 14:00:10 POST /ric/v1/handles/xapp-subscription-handle  body: {"address":"10.0.2.20","port":4562,"subscription_id":1} elapsed: 33.306µs
2025/03/24 14:00:38 DELETE /ric/v1/handles/xapp-subscription-handle  body: {"address":"10.0.2.20","port":4562,"subscription_id":1} elapsed: 38.869µs
2025/03/24 14:00:49 POST /ric/v1/handles/xapp-subscription-handle  body: {"address":"10.0.2.20","port":4560,"subscription_id":2} elapsed: 33.566µs
2025/03/24 14:28:47 DELETE /ric/v1/handles/xapp-subscription-handle  body: {"address":"10.0.2.20","port":4560,"subscription_id":2} elapsed: 35.522µs
2025/03/24 14:28:47 DELETE /ric/v1/handles/e2t  body: {"E2TAddress":"10.0.2.10:38000","ranNamelistTobeDissociated":["gnbd_001_001_00019b_0"]} elapsed: 211.705µs

h57jafari, 2025-03-25:
as I checked in all my config files there is only "gnbd_001_001_00019b_0", not "gnbd_001_001_00019b".
and when I look the logs as following, there are two "gnbd_001_001_00019b_0" and "gnbd_001_001_00019b". it may causing problem:

gNB_DU_ID is not null
cuupid =-1
duid =0
ranName =gnbd_001_001_00019b
1742795187955 1/RMR [INFO] ric message routing library on SI95 p=38000 mv=3 flg=00 id=a (f447e29 4.9.4 built: Dec 13 2023)
1742795332294 1/RMR [INFO] ric message routing library on SI95 p=38000 mv=3 flg=00 id=a (f447e29 4.9.4 built: Dec 13 2023)

gNB_DU_ID is not null
cuupid =-1
duid =0
ranName =gnbd_001_001_00019b
1742707850403 1/RMR [INFO] ric message routing library on SI95 p=38000 mv=3 flg=00 id=a (f447e29 4.9.4 built: Dec 13 2023)
{"ts":1742708053639,"crit":"ERROR","id":"E2Terminator","mdc":{"PID":"140669708400384","POD_NAME":"e2term_pod","CONTAINER_NAME":"ric_e2term","SERVICE_NAME":"ric_e2term_service","HOST_NAME":"ric_e2term_host","SYSTEM_NAME":"oran_sc_ric"},"msg":"Read error, Bad file descriptor "}

{"ts":1742841777661,"crit":"ERROR","id":"E2Terminator","mdc":{"PID":"124777082701568","POD_NAME":"e2term_pod","CONTAINER_NAME":"ric_e2term","SERVICE_NAME":"ric_e2term_service","HOST_NAME":"ric_e2term_host","SYSTEM_NAME":"oran_sc_ric"},"msg":"Failed to send message no CU entry gnbd_001_001_00019b_0"}
{"ts":1742841777661,"crit":"ERROR","id":"E2Terminator","mdc":{"PID":"124777082701568","POD_NAME":"e2term_pod","CONTAINER_NAME":"ric_e2term","SERVICE_NAME":"ric_e2term_service","HOST_NAME":"ric_e2term_host","SYSTEM_NAME":"oran_sc_ric"},"msg":"Error handling Xapp message"}

AlejandraOliver, 2025-03-26:
Hi, Which command you use to view? :

gNB_DU_ID is not null
cuupid =-1
duid =0
ranName =gnbd_001_001_00019b
1742795187955 1/RMR [INFO] ric message routing library on SI95 p=38000 mv=3 flg=00 id=a (f447e29 4.9.4 built: Dec 13 2023)
1742795332294 1/RMR [INFO] ric message routing library on SI95 p=38000 mv=3 flg=00 id=a (f447e29 4.9.4 built: Dec 13 2023)

gNB_DU_ID is not null
cuupid =-1
duid =0
ranName =gnbd_001_001_00019b
1742707850403 1/RMR [INFO] ric message routing library on SI95 p=38000 mv=3 flg=00 id=a (f447e29 4.9.4 built: Dec 13 2023)
{"ts":1742708053639,"crit":"ERROR","id":"E2Terminator","mdc":{"PID":"140669708400384","POD_NAME":"e2term_pod","CONTAINER_NAME":"ric_e2term","SERVICE_NAME":"ric_e2term_service","HOST_NAME":"ric_e2term_host","SYSTEM_NAME":"oran_sc_ric"},"msg":"Read error, Bad file descriptor "}

h57jafari, 2025-03-26:
sure, you can use these two to see logs:
$ sudo docker logs ric_e2term
$ sudo docker logs ric_rtmgr_sim

AlejandraOliver, 2025-03-26:
Cpuld you solved the problem?

h57jafari, 2025-03-26:
you can run the following command in your machine or the folder you want to find which files have "001_001_00019b":

grep -r "gnbd_001_001_00019b"

in my case in my laptop all I have is "gnbd_001_001_00019b_0" in several files.

AlejandraOliver, 2025-03-26:
Yes, I have run this command but all I get is gnbd_001_001_00019b_0. But in some place the ranName is getting gnbd_001_001_00019b. I don´t know why

h57jafari, 2025-03-26:
It seems they must be all same. For me, all four xApp files has "gnbd_001_001_00019b_0", even the kpm_mon_xapp.py, has "gnbd_001_001_00019b_0", and  it run successfully for me. Although other 3 xApp py files have "gnbd_001_001_00019b_0". but I can not run them successfully.

AlejandraOliver, 2025-03-26:
Anyone has implemented with success the RC xApps of the oran-sc-ric with the lastest version of both srsrran_proyect and oran-sc-ric? Please help us!