srsran/srsRAN_Project #1129: UE doesn't see the network

Alan1321, 2025-03-24:
**I have the following setup:**

B210 with an octoclock
Intel core i7 cpu with 32 gigs of RAM
UE's: 
1. Motorola one 5G ace
2. Sony Xperia 10 III

**gNB and EPC** 

gNB and Open5GS, both of them running inside a docker container
I have them successfully connected and running, BUT I just can't see the network on the phone.

I have tested these things:
mcc, mnc -> (999,99) (001,01) (901,70)

Tested all Bands supported by each of these phones. Nothing has worked yet. Would really really appreciate if someone could help.

**Logs**

**gNB Logs**
root@971de80fcf90:/home/ubuntu# gnb -c /home/ubuntu/gnb_rf_b200_tdd_n78_20mhz.yml 

--== srsRAN gNB (commit 8120ffbce) ==--

srsLog error - Unable to create log file "/tmp/gnb.log": Permission denied
Lower PHY in quad executor mode.
Available radio types: uhd and zmq.
[INFO] [UHD] linux; GNU C++ version 14.2.0; Boost_108300; UHD_4.8.0.0-64-g0dede88c
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=b200,num_recv_frames=64,num_send_frames=64'
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [MULTI_USRP] Setting master clock rate selection to 'manual'.
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Cell pci=1, bw=20 MHz, 1T1R, dl_arfcn=627340 (n78), dl_freq=3410.1 MHz, dl_ssb_arfcn=626976, ul_freq=3410.1 MHz

N2: Connection to AMF on 10.53.1.2:38412 completed
==== gNB started ===
Type <h> to view help
Late: 0; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;
Late: 5; Underflow: 1; Overflow: 0;
Late: 0; Underflow: 1; Overflow: 0;

**Open5GS Logs**
03/24 17:08:45.553: [app] INFO: Configuration: 'open5gs-5gc.yml' (../lib/app/ogs-init.c:133)
03/24 17:08:45.556: [sbi] INFO: NF EndPoint(addr) setup [127.0.0.22:7777] (../lib/sbi/context.c:474)
03/24 17:08:45.557: [dbi] INFO: MongoDB URI: 'mongodb://127.0.0.1/open5gs' (../lib/dbi/ogs-mongoc.c:130)
03/24 17:08:45.557: [sbi] INFO: NF Service [nudr-dr] (../lib/sbi/context.c:1829)
03/24 17:08:45.558: [sbi] INFO: nghttp2_server() [http://127.0.0.20]:7777 (../lib/sbi/nghttp2-server.c:424)
03/24 17:08:45.558: [app] INFO: UDR initialize...done (../src/udr/app.c:31)
03/24 17:08:45.559: [sbi] INFO: NF EndPoint(addr) setup [127.0.0.20:80] (../lib/sbi/context.c:2195)
03/24 17:08:45.559: [sbi] INFO: NF EndPoint(addr) setup [127.0.0.20:7777] (../lib/sbi/context.c:1934)
03/24 17:08:45.559: [nrf] INFO: [43648220-08ca-41f0-a802-25d5147420a5] NF registered [Heartbeat:10s] (../src/nrf/nf-sm.c:202)
03/24 17:08:45.560: [sbi] INFO: [43648220-08ca-41f0-a802-25d5147420a5] (NRF-notify) NF registered (../lib/sbi/nnrf-handler.c:1133)
03/24 17:08:45.560: [sbi] INFO: [43648220-08ca-41f0-a802-25d5147420a5] (NRF-notify) NF Profile updated [type:UDR] (../lib/sbi/nnrf-handler.c:1147)
03/24 17:08:45.560: [sbi] INFO: NF EndPoint(addr) setup [127.0.0.20:80] (../lib/sbi/context.c:2195)
03/24 17:08:45.560: [sbi] INFO: NF EndPoint(addr) setup [127.0.0.20:7777] (../lib/sbi/context.c:1934)
03/24 17:08:45.560: [scp] INFO: NF EndPoint(addr) setup [127.0.0.12:7777] (../src/scp/sbi-path.c:429)
03/24 17:08:45.560: [scp] INFO: NF EndPoint(addr) setup [127.0.0.13:7777] (../src/scp/sbi-path.c:429)
03/24 17:08:45.561: [sbi] INFO: [43648220-08ca-41f0-a802-25d5147420a5] (NRF-notify) NF registered (../lib/sbi/nnrf-handler.c:1133)
03/24 17:08:45.561: [sbi] INFO: [43648220-08ca-41f0-a802-25d5147420a5] (NRF-notify) NF Profile updated [type:UDR] (../lib/sbi/nnrf-handler.c:1147)
03/24 17:08:45.561: [sbi] INFO: [43648220-08ca-41f0-a802-25d5147420a5] (NRF-notify) NF registered (../lib/sbi/nnrf-handler.c:1133)
03/24 17:08:45.561: [sbi] INFO: [43648220-08ca-41f0-a802-25d5147420a5] (NRF-notify) NF Profile updated [type:UDR] (../lib/sbi/nnrf-handler.c:1147)
03/24 17:08:45.561: [sbi] INFO: NF EndPoint(addr) setup [127.0.0.20:80] (../lib/sbi/context.c:2195)
03/24 17:08:45.561: [sbi] INFO: NF EndPoint(addr) setup [127.0.0.20:7777] (../lib/sbi/context.c:1934)
03/24 17:08:45.561: [sbi] INFO: NF EndPoint(addr) setup [127.0.0.20:80] (../lib/sbi/context.c:2195)
03/24 17:08:45.561: [sbi] INFO: NF EndPoint(addr) setup [127.0.0.20:7777] (../lib/sbi/context.c:1934)
03/24 17:08:45.561: [sbi] INFO: [43648220-08ca-41f0-a802-25d5147420a5] NF registered [Heartbeat:10s] (../lib/sbi/nf-sm.c:208)
03/24 17:08:45.562: [nrf] INFO: NF EndPoint(addr) setup [127.0.0.20:7777] (../src/nrf/nnrf-handler.c:418)
03/24 17:08:45.562: [nrf] INFO: [43656f50-08ca-41f0-87fd-814699fa3cb7] Subscription created until 2025-03-25T17:08:45.562288+01:00 [duration:86400000000,validity:86400.000000] (../src/nrf/nnrf-handler.c:446)
03/24 17:08:45.562: [sbi] INFO: NF EndPoint(addr) setup [127.0.0.10:7777] (../lib/sbi/nnrf-handler.c:949)
03/24 17:08:45.562: [sbi] INFO: [43656f50-08ca-41f0-87fd-814699fa3cb7] Subscription created until 2025-03-25T17:08:45.562288+01:00 [duration:86400000000,validity:86400.000000,patch:43200.000000] (../lib/sbi/nnrf-handler.c:868)
03/24 17:08:56.382: [amf] INFO: gNB-N2 accepted[10.53.1.3]:59440 in ng-path module (../src/amf/ngap-sctp.c:113)
03/24 17:08:56.382: [amf] INFO: gNB-N2 accepted[10.53.1.3] in master_sm module (../src/amf/amf-sm.c:796)
03/24 17:08:56.390: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:1238)
03/24 17:08:56.391: [amf] INFO: gNB-N2[10.53.1.3] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:842)
03/24 17:09:16.930: [amf] INFO: gNB-N2[10.53.1.3] connection refused!!! (../src/amf/amf-sm.c:855)
03/24 17:09:16.930: [amf] INFO: [Removed] Number of gNBs is now 0 (../src/amf/context.c:1265)
03/24 17:10:51.667: [amf] INFO: gNB-N2 accepted[10.53.1.3]:59636 in ng-path module (../src/amf/ngap-sctp.c:113)
03/24 17:10:51.667: [amf] INFO: gNB-N2 accepted[10.53.1.3] in master_sm module (../src/amf/amf-sm.c:796)
03/24 17:10:51.673: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:1238)
03/24 17:10:51.673: [amf] INFO: gNB-N2[10.53.1.3] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:842)
03/24 17:15:57.395: [app] INFO: Signal-NUM[28] received (Window changed) (../src/main.c:63)
03/24 17:15:57.395: [app] INFO: Signal-NUM[28] received (Window changed) (../src/main.c:63)
03/24 17:15:57.395: [app] INFO: Signal-NUM[28] received (Window changed) (../src/main.c:63)
03/24 17:15:57.395: [app] INFO: Signal-NUM[28] received (Window changed) (../src/main.c:63)
03/24 17:15:57.395: [app] INFO: Signal-NUM[28] received (Window changed) (../src/main.c:63)
03/24 17:15:57.395: [app] INFO: Signal-NUM[28] received (Window changed) (../src/main.c:63)
03/24 17:15:57.395: [app] INFO: Signal-NUM[28] received (Window changed) (../src/main.c:63)

**gNB Configs**

[gnb_rf_b200_tdd_n78_20mhz.txt](https://github.com/user-attachments/files/19434032/gnb_rf_b200_tdd_n78_20mhz.txt)

**Open5GS Configs**

[open5gs-5gc.txt](https://github.com/user-attachments/files/19434065/open5gs-5gc.txt)

robertfalkenberg, 2025-03-25:
COTS UE not seeing the 5G network is a common issue that is often caused by restrictive configuration of the UE vendors.

Try using PLMN 001/01 for the network and also program the uSIM with an IMSI that starts with 00101...

If it still doesen't work: On some phones you need to enable 5G-SA explicitly in the networking menu. Sometimes that menu is hidden and requires apps with root permissions to unlock that menu item...

Alan1321, 2025-03-25:
`./pySim-prog.py -p0 -s 8988211000000689615 --mcc=001 --mnc=01 -a 77190612 --imsi=001010123456789  -k 41B7157E3337F0ADD8DA89210D89E17F --opc=1CD638FC96E02EBD35AA0D41EB6F812F`

^ This is my sim card setting. I have already tried using 001, 01 for mcc and mnc code and imsi starting with 00101.
I have enabled 5G-SA as well.

robertfalkenberg, 2025-03-27:
In your config I noticed the `rx_gain` being quite high:

```yml
  rx_gain: 76
```

Can you try with lower value, e.g. 40 as in our examples.

robertfalkenberg, 2025-03-27:
Also take a look here: https://github.com/srsran/srsRAN_Project/discussions/146#discussioncomment-6617621
  
Maybe the PLMN 00101 wasn't fully applied to all fields?

Alan1321, 2025-03-30:
I did all those and it still didn't work.
@robertfalkenberg Do you already have a working srsran? If so, could you share with me your OS with specific version, along with what commit of srsRAN_Project you are using?

I was looking through the build today and saw that 3 of the test cases were not passing. 
`The following tests FAILED:
	770 - transport_layer_address_test.ipv6_transport_layer_address_to_asn1 (Subprocess aborted)
	772 - transport_layer_address_test.asn1_to_ipv6_transport_layer_address (Subprocess aborted)
	5088 - network_test (Subprocess aborted)`

Alan1321, 2025-04-26:
Closing. It was a SIM card issue.

closed 2025-04-26 by Alan1321

luyuanfan, 2025-10-28:
> COTS UE not seeing the 5G network is a common issue that is often caused by restrictive configuration of the UE vendors.
>
> Try using PLMN 001/01 for the network and also program the uSIM with an IMSI [...]

Hi! Is this problem specific to 5G network? I remember using the LTE version of srsRAN with PLMN `99999` and the UE was able to discover the network.

robertfalkenberg, 2025-10-31:
Hi @luyuanfan,

The difference of network PLMN and IMSI mentioned in [the guide from your screenshot](https://docs.srsran.com/projects/project/en/latest/tutorials/source/cotsUE/source/index.html#sim-programming) especially targets the OnePlus 8t device, which is a very "stubborn" device and requires a "roaming hack" (mentioned [here](https://docs.srsran.com/projects/project/en/latest/knowledge_base/source/cots_ues/source/index.html#tested-cots-ues)) to connect. The device only connected to a 5G-SA network using an IMSI that starts with 00101 and configuring the network to a non-00101 PLMN. In addition that device required root and NSG to enfore 5G-SA.

My personal advice is to use a less-restricted UE of a different vendor and get the whole setup working.
For example, I had good experience with Pixel 6 smartphone (set all to 00101); or mobile broadband modems Quectel RM500Q and Telit FN990A28 (both accept any PLMN).