srsran/srsRAN_Project #1133: Invalid config for gnb?

dominikheinz, 2025-03-26:
Hi,

I am trying to start the gnb with the following configuration file, but it seems like it is not able to parse it?
I find this strange because in the past this configuration worked fine.

What is the problem in my configuration? The error message does not tell me what parameter is the issue.

**Error message:**

```
lab@gnb:~/srsRAN_Project/build/apps/gnb$ sudo ./gnb -c n78_siso_100mhz.yml

--== srsRAN gNB (commit 2be82d8ea) ==--

INI was not able to parse amf.++
Run with --help for more information.
```

**Config:**
```
lab@gnb:~/srsRAN_Project/build/apps/gnb$ sudo ./gnb -c n78_siso_100mhz.yml --help

--== srsRAN gNB (commit 2be82d8ea) ==--

INI was not able to parse amf.++
Run with --help for more information.
lab@gnb:~/srsRAN_Project/build/apps/gnb$ cat n78_siso_100mhz.yml
amf:
  addr: 127.0.0.5                # The address or hostname of the AMF.
  bind_addr: 127.0.0.1              # A local IP that the gNB binds to for traffic from the AMF.

ru_sdr:
  device_driver: uhd                # The RF driver name.
  device_args: type=x300            # Optionally pass arguments to the selected RF driver.
  clock: internal # gpsdo # external                           # Specify the clock source used by the RF.
  sync: internal # gpsdo # external                            # Specify the sync source used by the RF.
  srate: 184.32                    # RF sample rate might need to be adjusted according to selected bandwidth.
  #srate: 92.16
  tx_gain: 30                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 26                      # Receive gain of the RF might need to adjusted to the given situation.
  #time_alignment_calibration: -54

cell_cfg:
  dl_arfcn: 650000                  # ARFCN of the downlink carrier (center frequency).
  band: 78                          # The NR band.
  channel_bandwidth_MHz: 100        # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                    # Subcarrier spacing in kHz used for data.
  plmn: "99970"                     # PLMN broadcasted by the gNB.
  tac: 7                            # Tracking area code (needs to match the core configuration).
  pci: 1                            # Physical cell ID.

log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: debug                # Logging level applied to all layers.

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.
```

Open5Gs AMF Config:

```
lab@gnb:~/srsRAN_Project/build/apps/gnb$ cat /etc/open5gs/amf.yaml
logger:
  file:
    path: /var/log/open5gs/amf.log
#  level: info   # fatal|error|warn|info(default)|debug|trace

global:
  max:
    ue: 1024  # The number of UE can be increased depending on memory size.
#    peer: 64

amf:
  sbi:
    server:
      - address: 127.0.0.5
        port: 7777
    client:
#      nrf:
#        - uri: http://127.0.0.10:7777
      scp:
        - uri: http://127.0.0.200:7777
  ngap:
    server:
      - address: 127.0.0.5
  metrics:
    server:
      - address: 127.0.0.5
        port: 9090
  guami:
    - plmn_id:
        mcc: 999
        mnc: 70
      amf_id:
        region: 2
        set: 1
  tai:
    - plmn_id:
        mcc: 999
        mnc: 70
      tac: 1
  plmn_support:
    - plmn_id:
        mcc: 999
        mnc: 70
      s_nssai:
        - sst: 1
  security:
    integrity_order : [ NIA2, NIA1, NIA0 ]
    ciphering_order : [ NEA0, NEA1, NEA2 ]
  network_name:
    full: Open5GS
    short: Next
  amf_name: open5gs-amf0
  time:
#    t3502:
#      value: 720   # 12 minutes * 60 = 720 seconds
    t3512:
      value: 540    # 9 minutes * 60 = 540 seconds
```

brendan-mcauliffe, 2025-03-26:
Hi @dominikheinz, your configuration is out of date with the latest version of the code. The structure of the configuration file has been updated. Take a look at the [configuration reference file](https://docs.srsran.com/projects/project/en/latest/user_manuals/source/config_ref.html) in the documentation. The ``amf`` is now configured within the ``cu_cp``.

dominikheinz, 2025-03-27:
@brendan-mcauliffe Thanks for the reply. This already got me already a bit further, but I still have some issues with the configuration file format.

I moved the `amf` within the `cu_cp`, so, the initial error is now gone.
However, I now have the following error:

```
N2: Connection to AMF on 127.0.0.5:38412 completed
"NG Setup Procedure" failed. AMF NGAP cause: "unknown-PLMN-or-SNPN"
srsRAN ERROR: CU-CP failed to connect to AMF
```

I don't understand why this happens. The PLMN in both, my gnb config file, and in the amf.yaml of Open5GS is set to 99970.

Configs attached below:

**gnb_config.yaml:**

```
cu_cp:
  amf:
    addr: 127.0.0.5                # The address or hostname of the AMF.
    bind_addr: 127.0.0.1              # A local IP that the gNB binds to for traffic from the AMF.
    supported_tracking_areas:
      - tac: 7
        plmn_list:
          - plmn: "99970"
            tai_slice_support_list:
              - sst: 1
ru_sdr:
  device_driver: uhd                # The RF driver name.
  device_args: type=x300            # Optionally pass arguments to the selected RF driver.
  clock: internal # gpsdo # external                           # Specify the clock source used by the RF.
  sync: internal # gpsdo # external                            # Specify the sync source used by the RF.
  srate: 184.32                    # RF sample rate might need to be adjusted according to selected bandwidth.
  #srate: 92.16
  tx_gain: 30                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 26                      # Receive gain of the RF might need to adjusted to the given situation.
  #time_alignment_calibration: -54

cell_cfg:
  dl_arfcn: 650000                  # ARFCN of the downlink carrier (center frequency).
  band: 78                          # The NR band.
  channel_bandwidth_MHz: 100        # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                    # Subcarrier spacing in kHz used for data.
  plmn: 99970                     # PLMN broadcasted by the gNB.
  tac: 7                            # Tracking area code (needs to match the core configuration).
  pci: 1                            # Physical cell ID.

log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: debug                # Logging level applied to all layers.

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.
```

**/etc/open5gs/amf.yaml:**

```
logger:
  file:
    path: /var/log/open5gs/amf.log
#  level: info   # fatal|error|warn|info(default)|debug|trace

global:
  max:
    ue: 1024  # The number of UE can be increased depending on memory size.
#    peer: 64

amf:
  sbi:
    server:
      - address: 127.0.0.5
        port: 7777
    client:
#      nrf:
#        - uri: http://127.0.0.10:7777
      scp:
        - uri: http://127.0.0.200:7777
  ngap:
    server:
      - address: 127.0.0.5
  metrics:
    server:
      - address: 127.0.0.5
        port: 9090
  guami:
    - plmn_id:
        mcc: 999
        mnc: 70
      amf_id:
        region: 2
        set: 1
  tai:
    - plmn_id:
        mcc: 999
        mnc: 70
      tac: 1
  plmn_support:
    - plmn_id:
        mcc: 999
        mnc: 70
      s_nssai:
        - sst: 1
  security:
    integrity_order : [ NIA2, NIA1, NIA0 ]
    ciphering_order : [ NEA0, NEA1, NEA2 ]
  network_name:
    full: Open5GS
    short: Next
  amf_name: open5gs-amf0
  time:
#    t3502:
#      value: 720   # 12 minutes * 60 = 720 seconds
    t3512:
      value: 540    # 9 minutes * 60 = 540 seconds
```

Also, in the logs of open5gs amf, I can see the following:

```
Mar 27 13:11:28 gnb open5gs-amfd[5136]: 03/27 13:11:28.560: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:1273)
Mar 27 13:11:28 gnb open5gs-amfd[5136]: 03/27 13:11:28.560: [amf] INFO: gNB-N2[127.0.0.1] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:860)
Mar 27 13:11:28 gnb open5gs-amfd[5136]: 03/27 13:11:28.560: [amf] WARNING: NG-Setup failure: (../src/amf/ngap-handler.c:378)
Mar 27 13:11:28 gnb open5gs-amfd[5136]: 03/27 13:11:28.560: [amf] WARNING:     Cannot find Served TAI. Check 'amf.tai' configuration (../src/amf/ngap-handler.c:379)
Mar 27 13:11:28 gnb open5gs-amfd[5136]: 03/27 13:11:28.660: [amf] INFO: gNB-N2[127.0.0.1] connection refused!!! (../src/amf/amf-sm.c:873)
Mar 27 13:11:28 gnb open5gs-amfd[5136]: 03/27 13:11:28.660: [amf] INFO: [Removed] Number of gNBs is now 0 (../src/amf/context.c:1301)
```

Any idea what could be the issue? Something with the TAI?

dominikheinz, 2025-03-27:
Seems like the `tac` wasn't the same in the configs. After adjusting the `tac` value to `7`, it works.

closed 2025-03-27 by dominikheinz