srsran/srsRAN_Project #1138: UE can't ping the CN/internet and viceversa

Lab1R531, 2025-03-27:
Hi srsRAN Support!

I know that many issues of this topic have been opened, I tried to read them all to try to understand, but I could not solve the problem.
Can you help me understand where I am going wrong? Thanks in advance!

SET-UP:
- srsRAN_4G --> UE, on PC1
- srsRAN_Project --> GNB, on PC2
- Open5GS --> CN, on PC2
- Ettus B210 --> SDR

CN running in Background --> 10.45.0.1
```
GNB+CN$ ifconfig ogstun
ogstun: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1400
        inet 10.45.0.1  netmask 255.255.0.0  destination 10.45.0.1
        inet6 2001:db8:cafe::1  prefixlen 48  scopeid 0x0<global>
        inet6 fe80::12c2:8605:1f:c8f  prefixlen 64  scopeid 0x20<link>
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 2517  bytes 156446 (156.4 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

Starts UE and connects to the network. Gets IP --> 10.45.0.14
```
UE$ ifconfig tun_srsue
tun_srsue: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 1500
        inet 10.45.0.14  netmask 255.255.255.0  destination 10.45.0.14
        unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 500  (UNSPEC)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

The Route Table on UE is:
```
UE$ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.10.10.1      0.0.0.0         UG    101    0        0 eno2
10.10.10.0      0.0.0.0         255.255.255.0   U     101    0        0 eno2
10.45.0.0       0.0.0.0         255.255.255.0   U     0      0        0 tun_srsue
10.153.0.0      10.153.160.1    255.255.0.0     UG    100    0        0 eno1
10.153.160.0    0.0.0.0         255.255.255.128 U     100    0        0 eno1
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 eno1
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
172.19.1.0      0.0.0.0         255.255.255.0   U     0      0        0 br-17426b678d94
```

I add the default gw to UE and update the route table:
```
UE$ sudo route add default gw 10.45.0.1
UE$ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.45.0.1       0.0.0.0         UG    0      0        0 tun_srsue
0.0.0.0         10.10.10.1      0.0.0.0         UG    101    0        0 eno2
10.10.10.0      0.0.0.0         255.255.255.0   U     101    0        0 eno2
10.45.0.0       0.0.0.0         255.255.255.0   U     0      0        0 tun_srsue
10.153.0.0      10.153.160.1    255.255.0.0     UG    100    0        0 eno1
10.153.160.0    0.0.0.0         255.255.255.128 U     100    0        0 eno1
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 eno1
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
172.19.1.0      0.0.0.0         255.255.255.0   U     0      0        0 br-17426b678d94
```

I check the route table on GNB+CN PC:
```
GNB+CN$ route -n
Kernel IP routing table
Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.10.10.1      0.0.0.0         UG    20100  0        0 eno2
10.10.10.0      0.0.0.0         255.255.255.0   U     100    0        0 eno2
10.45.0.0       0.0.0.0         255.255.0.0     U     0      0        0 ogstun
10.45.0.0       0.0.0.0         255.255.0.0     U     0      0        0 ogstun
10.45.0.8       0.0.0.0         255.255.255.255 UH    0      0        0 ogstun
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 eno2
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
```

**RESULTS**
UE fails to ping the CN
CN fails to ping UE

asheeshtripathi, 2025-03-31:
Please have a look at the section: Adding a route for the UE to have WAN connectivity of Open5GS documentation: https://open5gs.org/open5gs/docs/guide/02-building-open5gs-from-sources/

Since you are using IPV4, you will need to have these commands run for Internet connectivity:

### Enable IPv4/IPv6 Forwarding
$ sudo sysctl -w net.ipv4.ip_forward=1
$ sudo sysctl -w net.ipv6.conf.all.forwarding=1

### Add NAT Rule
$ sudo iptables -t nat -A POSTROUTING -s 10.45.0.0/16 ! -o ogstun -j MASQUERADE
$ sudo ip6tables -t nat -A POSTROUTING -s 2001:db8:cafe::/48 ! -o ogstun -j MASQUERADE
Configure the firewall correctly. Some operating systems (Ubuntu) by default enable firewall rules to block traffic.

$ sudo ufw status
Status: active
$ sudo ufw disable
Firewall stopped and disabled on system startup
$ sudo ufw status
Status: inactive
Optionally, you may consider the settings below for security purposes.

### Ensure that the packets in the `INPUT` chain to the `ogstun` interface are accepted
$ sudo iptables -I INPUT -i ogstun -j ACCEPT

jjose001, 2025-04-07:
Hello @Lab1R531, 

Reading your post, I believe you're facing a routing issue that can be resolved by properly adjusting the routing table.

First of all, the current routing table looks correct, but you have a duplicated route for 10.45.0.0/16 via ogstun. While this is not necessarily a problem, it is redundant.

On the other hand, the core (Open5GS on PC2) is not performing NAT or IP forwarding between the ogstun and eno2 interfaces, so the traffic coming from PC1 (srsUE) gets lost along the way. To fix this, try the following steps:

On pc2, enable NAT as: 
- sudo sysctl -w net.ipv4.ip_forward=1
- sudo iptables -t nat -A POSTROUTING -s 10.45.0.0/16 -o eno2 -j MASQUERADE
- sudo iptables -A FORWARD -i ogstun -o eno2 -j ACCEPT
- sudo iptables -A FORWARD -i eno2 -o ogstun -j ACCEPT

On pc1, the routing table should be: 
`Destination     Gateway         Genmask         Flags Metric Ref    Use Iface
0.0.0.0         10.10.10.1      0.0.0.0         UG    101    0        0 eno2
10.10.10.0      0.0.0.0         255.255.255.0   U     101    0        0 eno2
10.45.0.0       0.0.0.0         255.255.255.0   U     0      0        0 tun_srsue
10.153.0.0      10.153.160.1    255.255.0.0     UG    100    0        0 eno1
10.153.160.0    0.0.0.0         255.255.255.128 U     100    0        0 eno1
169.254.0.0     0.0.0.0         255.255.0.0     U     1000   0        0 eno1
172.17.0.0      0.0.0.0         255.255.0.0     U     0      0        0 docker0
172.19.1.0      0.0.0.0         255.255.255.0   U     0      0        0 br-17426b678d94