srsran/srsRAN_Project #1139: The simple_rc_xapp.py is not working properly

Gmclaro, 2025-03-27:
Hi community!
I have a CU-DU split setup in two separate machines and the DU is connected to B210 USRP and the CU and DU are both connected to a NearRT-RIC(OS-RIC) running in a separate machine. I have tried running the simple_rc_xapp.py to manipulate the PRBs of the UE(Google Pixel 6a) I have connected to the DU. When I run the xApp in the output it shows me these messages of RIC_CONTROL_FAILURE and the upload bitrate doesn't change at all.

simple_rc_xapp.py output

```bash
nap@ric-tests:~/oran-sc-ric/xApps/python$ sudo docker compose exec python_xapp_runner ./simple_rc_xapp.py --ue_id 1
WARN[0000] /home/nap/oran-sc-ric/docker-compose.yml: the attribute `version` is obsolete, it will be ignored, please remove it to avoid potential confusion 
1743089711104 20/RMR [INFO] ric message routing library on SI95 p=4560 mv=3 flg=00 id=a (f447e29 4.9.4 built: Dec 13 2023)
15:35:12 Send RIC Control Request to E2 node ID: gnbd_001_001_00019b_2 for UE ID: 1, PRB_min_ratio: 1, PRB_max_ratio: 5
Received RIC_CONTROL_ACK
15:35:17 Send RIC Control Request to E2 node ID: gnbd_001_001_00019b_2 for UE ID: 1, PRB_min_ratio: 1, PRB_max_ratio: 40
Received RIC_CONTROL_FAILURE
15:35:22 Send RIC Control Request to E2 node ID: gnbd_001_001_00019b_2 for UE ID: 1, PRB_min_ratio: 1, PRB_max_ratio: 100
Received RIC_CONTROL_FAILURE
15:35:27 Send RIC Control Request to E2 node ID: gnbd_001_001_00019b_2 for UE ID: 1, PRB_min_ratio: 1, PRB_max_ratio: 5
Received RIC_CONTROL_FAILURE
15:35:32 Send RIC Control Request to E2 node ID: gnbd_001_001_00019b_2 for UE ID: 1, PRB_min_ratio: 1, PRB_max_ratio: 40
Received RIC_CONTROL_FAILURE
15:35:37 Send RIC Control Request to E2 node ID: gnbd_001_001_00019b_2 for UE ID: 1, PRB_min_ratio: 1, PRB_max_ratio: 100
Received RIC_CONTROL_FAILURE
```

DU configuration

```yaml

gnb_id: 411 # Optional UINT (411). Sets the numerical ID associated with the gNB.
gnb_id_bit_length: 22 # Optional UNIT. Sets the bit length of the gnb_id above. Format: integer between [22 - 32]

gnb_du_id: 2

cells:
  - pci: 2
    dl_arfcn: 368640 # Optional TEXT. Set the ARFCN for the above PCI. Note, this must match the value used in the cell_cfg section.
f1ap:
  cu_cp_addr: 10.0.23.73
  bind_addr: 10.0.22.247

f1u:
  socket:
    - bind_addr: 10.0.22.247
      # ext_addr: 10.0.22.176
      udp:
        max_rx_msgs: 256 # maximum packets read from the socket in a single syscall.
        pool_threshold: 0.9 # Pool occupancy threshold, after which packets are dropped.

ru_sdr:
  srate: 30.72 # Required FLOAT (61.44). Sets the sampling rate of the RF-frontend in MHz.
  device_driver: uhd # Required TEXT (uhd). RF device driver name. Supported: [uhd, zmq].
  device_args: type=b200,num_recv_frames=64,num_send_frames=64 # Optional TEXT. An argument that gets passed to the selected RF driver.
  otw_format: sc12
  tx_gain: 70
  rx_gain: 40

cell_cfg:
  dl_arfcn: 368640
  sector_id: 1
  band: 3
  channel_bandwidth_MHz: 20   # Required UINT (20). Sets the channel Bandwidth in MHz, the number of PRBs will be derived from this. Supported: [5, 10, 15, 20, 25, 30, 40, 50, 60, 70, 80, 90, 100].
  common_scs: 15
  plmn: "00101"
  tac: 1
  pci: 2
  slicing:
    - 
      sst: 1 # Sets the Slice Service Type. Supported: [0 - 255].
      sd: 0  # Sets the Service Differentiator. Supported: [0-16777215].
      sched_cfg:                              # Scheduler configuration for the slice
        min_prb_policy_ratio: 0                 # Optional UINT (0). Sets the minimum percentage of PRBs to be allocated to the slice. Supported [0 - 100].
[... 86 lines cut ...]
# remote_control:
#   enabled: true                        # Optional BOOLEAN (false). Enables the remote control feature. Supported: [false, true].
#   bind_addr: 127.0.0.1                  # Optional TEXT (127.0.0.1). Sets the local IP address to bind for remote control. Supported: IPv4 address.
#   port: 30000                            # Optional UINT (8001). Sets the port where the remote control server listens for incoming connections. Supported: [0 - 65535].
cu_up:
  f1u:
    socket: # Define UDP/IP socket(s) for F1-U interface.
      - # Socket 1
        bind_addr: 10.0.23.73 # Required TEXT. Sets the address that the F1-U socket will bind to.
        udp:
          max_rx_msgs: 256 # maximum packets read from the socket in a single syscall.
          pool_threshold: 0.9 # Pool occupancy threshold, after which packets are dropped.

  ngu:
    socket:
      - bind_addr: 10.0.23.244

metrics:
  cu_cp_statistics_report_period: 1 # Optional UINT (1). Sets the CU-CP statistics report period in seconds. Set this value to 0 to disable this feature. Supported: [0 - inf].
  cu_up_statistics_report_period: 1 # Optional UINT (1). Sets the CU-UP statistics report period in seconds. Set this value to 0 to disable this feature. Supported: [0 - inf].
  pdcp_report_period: 20000 # Optional UINT (0). Sets the PDCP metrics report period.

```

I would like to know if its already  possible allocate prbs to a physicall UE or am i missing something?

h57jafari, 2025-03-30:
Hi, Gmclaro. Can you share your config amf, upf, smf, nrf,... yaml , gNB and UE files.

h57jafari, 2025-03-30:
We all have the same problem with running control xApp, where no change happens or UE disconnect and bit rate drop to zero. check your DL on the UE and DU. I assume your bit rate dropped to zero, that is why you receive only one first ACK message. Here is output of mine, I do not receive any error, but bit rate drop to zero and UE disconnect. 

[image]

Gmclaro, 2025-04-09:
> hi [@Gmclaro](https://github.com/Gmclaro), did the problem fixed?

no, 
not yet :(

h57jafari, 2025-04-09:
As several of us mentioned, the current version of srsRAN have problem to run control xApps.

h57jafari, 2025-04-09:
Hi, Gmclaro. Can you share your open5GS config files: amf, upf, smf, nrf,... yaml and UE files for your setup.

hnavidan, 2025-04-23:
I checked with the new update today, and it's still the same problem. The xapp changes PRBs but it doesn't actually do anything in the RAN.

pgawlowicz, 2025-06-24:
The E2SM-RC PRB quota control was fixed in the newest srsRAN 5G Project release. Could you check?
Please also update the ORAN-SC-RIC repo.

Best,
Piotr

Gmclaro, 2025-06-25:
> The E2SM-RC PRB quota control was fixed in the newest srsRAN 5G Project release. Could you check? Please also update the ORAN-SC-RIC repo.
> 
> Best, Piotr

Hi Piotr thanks for the update
The problem still remains when i generate some traffic, when i do an iperf  the xapps shows messages of RIC_CONTROL_FAILURE

pgawlowicz, 2025-06-25:
did you set the UE-ID correctly?

Gmclaro, 2025-06-25:
RIC logs
```bash
ric_appmgr          | 2025/06/25 13:20:51 Serving app manager at http://[::]:8080
ric_rtmgr_sim       | 2025/06/25 13:21:17 POST /ric/v1/handles/associate-ran-to-e2t  body: [{"E2TAddress":"10.0.2.10:38000","ranNamelist":["gnb_001_001_00019b"]}] elapsed: 33.231µs
ric_rtmgr_sim       | 2025/06/25 13:21:17 POST /ric/v1/handles/associate-ran-to-e2t  body: [{"E2TAddress":"10.0.2.10:38000","ranNamelist":["gnbc_001_001_00019b_0"]}] elapsed: 18.689µs
ric_rtmgr_sim       | 2025/06/25 13:21:31 POST /ric/v1/handles/associate-ran-to-e2t  body: [{"E2TAddress":"10.0.2.10:38000","ranNamelist":["gnbd_001_001_00019b_1"]}] elapsed: 13.285µs
ric_rtmgr_sim       | 2025/06/25 13:21:34 POST /ric/v1/handles/associate-ran-to-e2t  body: [{"E2TAddress":"10.0.2.10:38000","ranNamelist":["gnbd_001_001_00019b_2"]}] elapsed: 74.817µs
ric_rtmgr_sim       | 2025/06/25 13:25:24 POST /ric/v1/handles/xapp-subscription-handle  body: {"address":"10.0.2.20","port":4562,"subscription_id":1} elapsed: 12.935µs
ric_rtmgr_sim       | 2025/06/25 13:25:28 DELETE /ric/v1/handles/xapp-subscription-handle  body: {"address":"10.0.2.20","port":4562,"subscription_id":1} elapsed: 15.746µs
ric_dbaas           | 1:M 25 Jun 2025 13:25:52.090 * 100 changes in 300 seconds. Saving...
ric_dbaas           | 1:M 25 Jun 2025 13:25:52.091 * Background saving started by pid 15
ric_dbaas           | 15:C 25 Jun 2025 13:25:52.096 * DB saved on disk
ric_dbaas           | 15:C 25 Jun 2025 13:25:52.096 * RDB: 0 MB of memory used by copy-on-write
ric_dbaas           | 1:M 25 Jun 2025 13:25:52.191 * Background saving terminated with success
```

xapp
```bash
nap@ric-tests:~/oran-sc-ric/xApps/python$ sudo docker compose exec python_xapp_runner ./simple_rc_xapp.py --gnb_cu_ue_f1ap_id 0
WARN[0000] /home/nap/oran-sc-ric/docker-compose.yml: the attribute `version` is obsolete, it will be ignored, please remove it to avoid potential confusion 
1750858011165 41/RMR [INFO] ric message routing library on SI95 p=4560 mv=3 flg=00 id=a (f447e29 4.9.4 built: Dec 13 2023)
13:26:52 Send RIC Control Request to E2 node ID: gnbd_001_001_00019b_2 for UE ID: 0, PRB_min_ratio: 10, PRB_max_ratio: 30
Received RIC_CONTROL_ACK
13:26:57 Send RIC Control Request to E2 node ID: gnbd_001_001_00019b_2 for UE ID: 0, PRB_min_ratio: 10, PRB_max_ratio: 50
Received RIC_CONTROL_ACK
13:27:02 Send RIC Control Request to E2 node ID: gnbd_001_001_00019b_2 for UE ID: 0, PRB_min_ratio: 10, PRB_max_ratio: 70
Received RIC_CONTROL_ACK
13:27:07 Send RIC Control Request to E2 node ID: gnbd_001_001_00019b_2 for UE ID: 0, PRB_min_ratio: 10, PRB_max_ratio: 100
Received RIC_CONTROL_ACK
13:27:12 Send RIC Control Request to E2 node ID: gnbd_001_001_00019b_2 for UE ID: 0, PRB_min_ratio: 10, PRB_max_ratio: 30
Received RIC_CONTROL_ACK
13:27:17 Send RIC Control Request to E2 node ID: gnbd_001_001_00019b_2 for UE ID: 0, PRB_min_ratio: 10, PRB_max_ratio: 50
Received RIC_CONTROL_ACK
13:27:22 Send RIC Control Request to E2 node ID: gnbd_001_001_00019b_2 for UE ID: 0, PRB_min_ratio: 10, PRB_max_ratio: 70
Received RIC_CONTROL_ACK
13:27:27 Send RIC Control Request to E2 node ID: gnbd_001_001_00019b_2 for UE ID: 0, PRB_min_ratio: 10, PRB_max_ratio: 100
```

console trace in the DU console

```bash

          |--------------------DL---------------------|-------------------------UL----------------------------------
 pci rnti | cqi  ri  mcs  brate   ok  nok  (%)  dl_bs | pusch  rsrp  ri  mcs  brate   ok  nok  (%)    bsr     ta  phr
   2 4601 |   5 1.0   15   993k  489    0   0%      0 |   9.5   ovl   1    8   2.6M 1377  437  24%  20.5k   627n   11
   2 4601 |   5 1.0   15   1.3M  654    0   0%      0 |  10.5   ovl   1    8   2.9M 2095  697  24%  28.6k   627n    2
   2 4601 |   5 1.0   15   1.4M  708    0   0%     58 |   9.3   ovl   1    4   2.9M 2409  431  15%  39.8k   619n    1
   2 4601 |   5 1.0   15   857k  421    0   0%    232 |  15.4   ovl   1   11   2.4M  930  337  26%  2.38k   619n    2
   2 4601 |   5 1.0   15   692k  339    0   0%      0 |  22.8   ovl   1   21   2.3M  424  493  53%  28.6k   615n    9
   2 4601 |   5 1.0   15   719k  352    1   0%      0 |  27.2   ovl   1   26   2.9M  405  713  63%    745   609n   10
   2 4601 |   5 1.0   15   580k  283    0   0%      0 |  26.2  -0.0   1   25   1.6M  273  444  61%  28.6k   619n   10
   2 4601 |   5 1.0   15   678k  332    0   0%      0 |  22.1   ovl   1   21   2.2M  390  434  52%  28.6k   619n    9
   2 4601 |   5 1.0   15   648k  317    0   0%      0 |  21.1   ovl   1   20   2.5M  431  572  57%  28.6k   615n    6
   2 4601 |   5 1.0   15   884k  433    0   0%      0 |  20.0  -0.3   1   17   3.0M  936  721  43%  28.6k   614n    1
   2 4601 |   5 1.0   14   1.6M  802   16   1%      0 |   8.6  -0.1   1    2   2.7M 2777  182   6%      0   616n   
```

now it seems to work fine, but i dont the prb change is being reflected in the console trace

pgawlowicz, 2025-07-01:
hi, what is BW of your cell? do you push enough traffic?