srsran/srsRAN_Project #1141: Real-time failure in RF

dominikheinz, 2025-04-01:
I am running an X310 with srsRAN and Open5GS on Ubuntu22.04.
I noticed in the gnb's tracelog that many underflows/late packets occur.
This happens even if there is no traffic, but the problem is amplified when there is traffic.

```
==== gNB started ===
Type <h> to view help
Late: 203; Underflow: 69; Overflow: 0;
Late: 0; Underflow: 2; Overflow: 0;
Late: 81; Underflow: 37; Overflow: 0;
Late: 41; Underflow: 4; Overflow: 0;
Late: 0; Underflow: 48; Overflow: 0;
Late: 41; Underflow: 3; Overflow: 0;
Late: 78; Underflow: 4; Overflow: 0;
Late: 0; Underflow: 2; Overflow: 0;
Late: 164; Underflow: 35; Overflow: 0;
[DEBUG] [rx_flow_ctrl_state] Flow control state mismatch: bytes reported: 66478684040 bytes counted locally: 66478452736 delta: 231304 Packets reported: 8337499 Packets counted locally: 8337470 delta: 29 src_epid=2 dst_epid=4

Late: 1223; Underflow: 63; Overflow: 2;
Late: 619; Underflow: 36; Overflow: 0;
Late: 1450; Underflow: 81; Overflow: 0;
[DEBUG] [rx_flow_ctrl_state] Flow control state mismatch: bytes reported: 68795969504 bytes counted locally: 68795674392 delta: 295112 Packets reported: 8628124 Packets counted locally: 8628087 delta: 37 src_epid=2 dst_epid=4

Late: 678; Underflow: 51; Overflow: 1;
Late: 0; Underflow: 12; Overflow: 0;
```

It shouldn't be a CPU limitation, the hardware is a Intel 13900H (MiniPC Minisforum MS-01).
Also, none of the cores is at 100%, they sit around 40-50%.

The issue also occurs when setting lower BW, e.g. 80MHz instead of 100MHz.

In the gnb logs, I notcied the following:

```
2025-04-01T13:12:12.436073 [RF      ] [W] Real-time failure in RF: underflow
2025-04-01T13:12:12.436077 [RF      ] [D] Real-time failure in RF: Type=underflow Source=tx Timestamp=na
2025-04-01T13:12:12.436710 [FAPI    ] [D] [   744.9] Sector#0: Slot.indication time_point=1743513132439206169
2025-04-01T13:12:12.436772 [SCHED   ] [D] [   744.9] Slot decisions pci=1 t=22us (0 PDSCHs, 0 PUSCHs):
- PUCCH: c-rnti=0x4602 format=2 prb=[7..8) symb=[4..6) uci: harq_bits=0 sr=0 csi-1_bits=4
2025-04-01T13:12:12.436798 [FAPI    ] [D] [   744.9] Sector#0: UL_TTI.request slot=744.9
	- PUCCH rnti=0x4602 bwp=0:273 format=2 prb=7:1 prb2=0 symb=4:2 harq_bit_len=0 sr_bit_len=0 csi1_bit_len=4
2025-04-01T13:12:12.436937 [RF      ] [D] Real-time failure in RF: Type=end-of-burst Source=tx Timestamp=3259058880
2025-04-01T13:12:12.437295 [RF      ] [W] Real-time failure in RF: underflow
2025-04-01T13:12:12.437298 [RF      ] [D] Real-time failure in RF: Type=underflow Source=tx Timestamp=na
2025-04-01T13:12:12.437305 [RF      ] [W] Real-time failure in RF: underflow
2025-04-01T13:12:12.437305 [RF      ] [D] Real-time failure in RF: Type=underflow Source=tx Timestamp=na
2025-04-01T13:12:12.437308 [RF      ] [W] Real-time failure in RF: underflow
2025-04-01T13:12:12.437309 [RF      ] [D] Real-time failure in RF: Type=underflow Source=tx Timestamp=na
2025-04-01T13:12:12.437468 [RF      ] [W] Real-time failure in RF: underflow
2025-04-01T13:12:12.437469 [RF      ] [D] Real-time failure in RF: Type=underflow Source=tx Timestamp=na
2025-04-01T13:12:12.437471 [RF      ] [W] Real-time failure in RF: underflow
```

Likewise thre are a lot of late packets. But little to no overflows.

```
2025-04-01T13:12:12.457185 [RF      ] [W] Real-time failure in RF: late
2025-04-01T13:12:12.457186 [RF      ] [D] Real-time failure in RF: Type=late Source=tx Timestamp=na
2025-04-01T13:12:12.457188 [FAPI    ] [D] [  746.13] Sector#0: DL_TTI.request slot=746.13, is_last_message_in_slot=true
2025-04-01T13:12:12.457188 [RF      ] [W] Real-time failure in RF: late
2025-04-01T13:12:12.457188 [RF      ] [D] Real-time failure in RF: Type=late Source=tx Timestamp=na
2025-04-01T13:12:12.457346 [RF      ] [W] Real-time failure in RF: late
2025-04-01T13:12:12.457346 [RF      ] [D] Real-time failure in RF: Type=late Source=tx Timestamp=na
2025-04-01T13:12:12.457347 [RF      ] [W] Real-time failure in RF: late
2025-04-01T13:12:12.457348 [RF      ] [D] Real-time failure in RF: Type=late Source=tx Timestamp=na
2025-04-01T13:12:12.457407 [RF      ] [D] Real-time failure in RF: Type=end-of-burst Source=tx Timestamp=3262929600
2025-04-01T13:12:12.457410 [FAPI    ] [D] [  746.14] Sector#0: Slot.indication time_point=1743513132459910025
2025-04-01T13:12:12.457441 [FAPI    ] [D] [  746.14] Sector#0: DL_TTI.request slot=746.14, is_last_message_in_slot=true
2025-04-01T13:12:12.457505 [RF      ] [W] Real-time failure in RF: late
[... 65 lines cut ...]
  all_level: debug                # Logging level applied to all layers.

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.
```

**Note:**

I already enabled tuned for performance tuning, but the problem remains, with or without the profile active.

**Note2:**

Starting the gnb also shows some issues about the MTU of my SFP+ interfaces:

```
[WARNING] [X300] You requested a receive frame size of (8000) but your NIC's max frame size is (7972).Please verify your NIC's MTU setting using 'ip link' or set the recv_frame_size argument appropriately.UHD will use the auto-detected max frame size for this connection.
[WARNING] [X300] You requested a send frame size of (8000) but your NIC's max frame size is (7972).Please verify your NIC's MTU setting using 'ip link' or set the send_frame_size argument appropriately.UHD will use the auto-detected max frame size for this connection.
```

However, I did set the MTU to 8000, and the issue is the same, - also when I lower it. I have an Intel X710, which has a max MTU of 9072, so it shouldnt be a hardware issue.

Any ideas what causes this, and how to resolve this?

asheeshtripathi, 2025-04-01:
The recommended MTU size is 9000 for the 10G interface (gNB machine to USRP x310 ETH port 1). With this MTU size, try lower bandwidth as well, 20 or 40 MHz, to check the performance/errors.

h57jafari, 2025-04-02:
Yes, 184.32 is to much, it is max sample rate on the paper. 
- You may need to operate with DPDK to achieve that sample rate. 
- Check following link to performance tuning: 
https://kb.ettus.com/USRP_Host_Performance_Tuning_Tips_and_Tricks
-also try following benchmark command to check max sample rate that your USRP can achieve without extra computation:
./benchmark_rate --rx_rate 10e6 --tx_rate 10e6      (replace and increase 10e6 to check)

dominikheinz, 2025-04-02:
@asheeshtripathi Thanks for the reply. When setting the MTU to 9000, I still get the following warning when starting up the gnb:

```
[WARNING] [X300] You requested a receive frame size of (9000) but your NIC's max frame size is (8000).Please verify your NIC's MTU setting using 'ip link' or set the recv_frame_size argument appropriately.UHD will use the auto-detected max frame size for this connection.
[WARNING] [X300] You requested a send frame size of (9000) but your NIC's max frame size is (8000).Please verify your NIC's MTU setting using 'ip link' or set the send_frame_size argument appropriately.UHD will use the auto-detected max frame size for this connection.
```
This does not make sense to me. The Intel X710 SFP+ module is capable of frame sizes of up to 9072. Also, when setting the MTU to 9000, I can indeed see it when running `ip a`. So I do not understand the error.

Oh, and as I mentioned the underflow errors happen also when I use e.g. 80MHz. Same issue even with 20MHz.

@h57jafari Thanks for your reply.

> You may need to operate with DPDK to achieve that sample rate

I don't know how to configure this. I never used DPDK.

> Check following link to performance tuning

I did take a look. I tried adjusting the CPU governer, as well as the thread priority scheduling to `@usrp - rtprio  99`. Both had no effect. And I made sure the MTU is set to 9000, the network buffer sizes are set as specified in the guide and I also increased the ring buffers as described in the guide to `sudo ethtool -G <interface> tx 4096 rx 4096`.
All of these settings did not fix the problem.

> 184.32 is to much, it is max sample rate on the paper.

I tried different combinations. With `bw=100mhz/srate=184.32` I get the problem I described above. With `bw=80mhz/srate=184.32`, the same issue, but it seems(?) like it appears later/there are less underflows.
However, for 100mhz, I can't set any other valid srate than 184.32. 92.16 is to low, so the gnb will refuse it, and higher multiples of 1.92 aren't accepted, unless I set 184.32. Any suggestion?

> Try following benchmark command to check max sample rate 

I will try that.

dominikheinz, 2025-04-02:
@h57jafari I am unsure how to proceed. I tried various things, different bandwidths, disabling hyperthreading/virtualization in BIOS (recommended by Etus), CPU govnerer adjustments as well as power profiles etc.
I still get underflow/late packets that degrade the connection/kill it after some time.

How can I even identify the bottleneck? It shouldnt be the system, since none of the cores of the CPU is running at 100%. Maybe at 30%. What can I do?

h57jafari, 2025-04-02:
For the MCR of 184.32 MHz, the achievable sampling rates using an even decimation factor are 184.32, 92.16, 46.08, 30.72, 23.04, 18.432, 15.36, 13.166 Msps, ... 180.0 Ksps.

try 46.08, 30.72, 23.04 with associated 40, 30, 20 MHz bandwidth. 
As I said 184.32, 92.16 are high, you will have U/O error with them, otherwise you need to do perfect tuning.

h57jafari, 2025-04-02:
Although powerful CPU is good, but it is not just CPU speed, OS kernel need to schedule those packets between USRP and your API (srsRAN programs that you are running), that has its priorities that takes time as well. that's why need DPDK to bypass all those for higher sample rate. also ignore those warning.

asheeshtripathi, 2025-04-02:
ping -M do -s 8972 <destination_IP> to see if the interface supports jumbo packets. 

Also, just to make sure that uhd_usrp_probe does not throw a warning for buffer sizes. We have tested x310, N310, and B210, all with similar configurations (BW up to 80 MHz). A little underflow/late/overflow won't degrade the performance. 

Other suggestion on lo_offset:
With X310, in our experience, setting lo_offset more than half of the configured BW significantly improves the 5G signal quality, specially when using higher Tx and Rx gains. This has been verified with the UBX-160 daughterboard. Below is the screenshot for 20 MHz BW configuration. We have seen a significant improvement in the over-the-air throughput both in UL and DL directions with LO leakage moved out of 5G signal.

[image]

dominikheinz, 2025-04-07:
@h57jafari That makes sense. I am currently trying to configure DPDK to see if it makes a difference. I followed the installation guide step by step, but I can't seem to activate the X710 interfaces (?).

After binding the interfaces, the output looks like this:

```
Network devices using DPDK-compatible driver
============================================
0000:02:00.0 'Ethernet Controller X710 for 10GbE SFP+ 1572' drv=vfio-pci unused=i40e
0000:02:00.1 'Ethernet Controller X710 for 10GbE SFP+ 1572' drv=vfio-pci unused=i40e

Network devices using kernel driver
===================================
0000:57:00.0 'Ethernet Controller I226-V 125c' if=enp87s0 drv=igc unused=vfio-pci
0000:58:00.0 'Ethernet Controller I226-LM 125b' if=enp88s0 drv=igc unused=vfio-pci
0000:59:00.0 'MT7922 802.11ax PCI Express Wireless Network Adapter 0616' if=wlp89s0 drv=mt7921e unused=vfio-pci

No 'Baseband' devices detected
==============================

No 'Crypto' devices detected
============================

No 'DMA' devices detected
=========================

No 'Eventdev' devices detected
==============================

No 'Mempool' devices detected
=============================

No 'Compress' devices detected
==============================

No 'Misc (rawdev)' devices detected
===================================

No 'Regex' devices detected
===========================

No 'ML' devices detected
========================
```

This shows that the X710 interfaces are listed under devices that use the DPDK-compatible driver.

Now, the struggle is to adjust the gnb configuration accordingly. This is what I currently have:

```yaml
cu_cp:
  amf:
    addr: 127.0.0.5
  bind_addr: 127.0.0.1
  supported_tracking_areas:
    - tac: 7
  plmn_list:
    - plmn: "99970"
      tai_slice_support_list:
        - sst: 1

ru_sdr:
  device_driver: uhd
  dev_args: type=x300,recv_frame_size=9000,send_frame_size=9000,mtu=9000
  clock: internal
  sync: internal
  srate: 46.08
  tx_gain: 30
  rx_gain: 26

cell_cfg:
  dl_arfcn: 650000
  band: 78
  channel_bandwidth_MHz: 40
  common_scs: 30
  plmn: "99970"
  tac: 7
  pci: 1

expert_execution:
  upper_phy:
    nof_pusch_decoder_threads: 4
    nof_ul_threads: 4
    nof_dl_threads: 4

log:
  filename: /tmp/gnb.log
  all_level: debug

pcap:
  mac_enable: false
  mac_filename: /tmp/gnb_mac.pcap
  ngap_enable: true
  ngap_filename: /tmp/gnb_ngap.pcap
```

However, starting the gnb fails, because it can't find the SDR, - likely because it tries to connect to it via the interfaces that are now managed by DPDK. How do I fix this? What do I have to adjust in the config file?

```
=== srsRAN gNB (commit 2be02d8ea) ===

Lower PHY in quad executor mode.
Available radio types: uhd
[INFO] [UHD] linux; GNU C++ version 13.2.0; Boost_108300; UHD_4.6.0.0-ds1.5-1.ubuntu0.24.04.1
Making USRP object with args 'type=x300,recv_frame_size=9000,send_frame_size=9000,mtu=9000,master_clock_rate=184.32e6'
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Failed to open device with address 'type=x300,recv_frame_size=9000,send_frame_size=9000,mtu=9000': LookupError: KeyError: No devices found for ----->

Device Address:
  type: x300
  recv_frame_size: 9000
  send_frame_size: 9000
  mtu: 9000
  master_clock_rate: 184.32e6

srsRAN ERROR: Unable to create radio session.
```

dominikheinz, 2025-04-07:
@h57jafari I already compiled srsRAN_Project with DPDK flags, as described in the documentation. Didn't solve the issue. I assume I need to change something specifically in the configuration, to make it interact with the SDR. I don't know how tho.

h57jafari, 2025-04-07:
Run the following command to see if DPDK enabled in GNU Radio, otherwise you need to re-install GNU radio, after installing DPDK:

gnuradio-config-info --enabled-components

Also check here:
https://docs.srsran.com/projects/project/en/latest/tutorials/source/dpdk/source/index.html

VuTuLan, 2025-04-07:
hello @h57jafari I built a model including cu/du split built on 2 different VMs connected to USRP N321 and encountered the problem of late underflow as above and could not attach UE. srsDU uses 10gbs interface and srate=7.68 but still the above error occurs. Do you have any solution? Thanks
`f1ap:                               
  cu_cp_addr:  192.168.180.37                
  bind_addr: 192.168.180.250

ru_sdr:
  device_driver: uhd 
  device_args: addr=192.168.210.31,mgmt_addr=192.168.180.93,clock_source=internal,send_frame_size=4000,recv_frame_size=4000,master_clock_rate=245.76e6 
  sync: internal 
  srate:  7.68 
  tx_gain: 20 
  rx_gain: 20

cell_cfg:
  dl_arfcn: 368500 
  band: 3 
  channel_bandwidth_MHz: 5 
  common_scs: 15 
  plmn: "00101" 
  tac: 1 
  pci: 1 
  nof_antennas_dl: 1 
  nof_antennas_ul: 1 
  pdcch:
    dedicated:
      ss2_type: common
      dci_format_0_1_and_1_1: false
    common:      
       ss0_index: 0
      coreset0_index: 0
  prach:
    prach_config_index: 2
  pdsch:
    mcs_table: qam64
  pusch:
    mcs_table: qam64
  pucch:
    f0_or_f1_nof_cell_res_sr: 2
    f2_or_f3_or_f4_nof_cell_res_csi: 2
log:
  filename: /tmp/du.log
  all_level: info`

h57jafari, 2025-04-07:
Hi Dominik,
After you installed DPDK, you need to re-install both UHD and GNU Radio, check the following link:
https://kb.ettus.com/Getting_Started_with_DPDK_and_UHD

h57jafari, 2025-04-07:
Hi VuTuLan, Yes the order is important: 1-DPDK, 2-UHD, 3-GNU Radio, and 4-srsRAN.

h57jafari, 2025-04-07:
GNU Radio is the open source SW tool for signal processing and SDR program/control. If you are using USRP, or other SDR HW or using multiple UE's scenario through srsRAN that requires broker, you need to run GNU Radio. https://www.gnuradio.org/

TULAN2901, 2025-04-07:
@h57jafari can you read my DU conf file above to solve late underflow ? Thanks

h57jafari, 2025-04-07:
can you share all the associated files, cu, du, gNB, UE, and open5GS yaml files (amf, upf, smf, nrf,...)

h57jafari, 2025-04-07:
Use following command to see if your UHD installed/enabled with DPDK:

uhd_config_info --enabled-components

dominikheinz, 2025-04-09:
@h57jafari Trying to follow your suggestion, reinstalling all the required components, to ensure everything has dpdk enabled.
Here is the output of `gnuradio-config-info --enabled-components` and `uhd_config_info --enabled-components`:
```
lab@gnb:~/srsRAN_Project/build/apps/gnb$ gnuradio-config-info --enabled-components
testing-support;python-support;post-install;doxygen;* HTML Docs;* XML Docs;* PDF Docs;man-pages;gnuradio-runtime;common-precompiled-headers;gr-ctrlport;* thrift;gnuradio-companion;JSON/YAML config blocks;gr-blocks;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-dtv;gr-audio;* alsa;* oss;* jack;* portaudio;gr-channels;gr-pdu;gr-iio;* libad9361;gr-qtgui;gr-trellis;gr-uhd;gr-uhd UHD 4.0 RFNoC;gr-utils;gr_modtool;gr_blocktool;gr-video-sdl;gr-vocoder;* codec2;* freedv;* gsm;gr-wavelet;gr-zeromq;gr-network;gr-soapy
```
```
lab@gnb:~/srsRAN_Project/build/apps/gnb$ uhd_config_info --enabled-components
Enabled components: LibUHD, LibUHD - C API, LibUHD - Python API, Examples, Utils, Tests, Python Module (Utils only), USB, B100, B200, USRP1, USRP2, X300, MPMD, SIM, N300, N320, E320, E300, X400, OctoClock, DPDK
```
Seems like uhd has dpdk enabled, but gnuradio doesn't?

I am currently trying to checkout the latest version (v3.9.8.0), but i fail to compile it:

```
/usr/include/volk/volk_common.h:176:35: error: unable to find numeric literal operator ‘operator""f’
  176 |     const float a1 = +0x1.ffffeap-1f;
      |                                   ^~
/usr/include/volk/volk_common.h:177:34: error: unable to find numeric literal operator ‘operator""f’
  177 |     const float a3 = -0x1.55437p-2f;
      |                                  ^~
/usr/include/volk/volk_common.h:178:35: error: unable to find numeric literal operator ‘operator""f’
  178 |     const float a5 = +0x1.972be6p-3f;
      |                                   ^~
/usr/include/volk/volk_common.h:179:34: error: unable to find numeric literal operator ‘operator""f’
  179 |     const float a7 = -0x1.1436ap-3f;
      |                                  ^~
/usr/include/volk/volk_common.h:180:35: error: unable to find numeric literal operator ‘operator""f’
  180 |     const float a9 = +0x1.5785aap-4f;
      |                                   ^~
/usr/include/volk/volk_common.h:181:36: error: unable to find numeric literal operator ‘operator""f’
  181 |     const float a11 = -0x1.2f3004p-5f;
      |                                    ^~
/usr/include/volk/volk_common.h:182:36: error: unable to find numeric literal operator ‘operator""f’
  182 |     const float a13 = +0x1.01a37cp-7f;
```
I am unsure if I am even on the right track here?
- Am I even using the right version of gnuradio?
- I don't see any mention about DPDK in the cmake output..?
- This error indicates either a missing flag or an outdated toolchain. I have gcc13.3, so that should be fine. And manually setting the flag `-ftext-numeric-literals` doesnt solve the issue either

Any pointers?

dominikheinz, 2025-04-09:
Okay, turns out that release of GNU Radio was just borked. I checked out v3.10.8.0, and was able to compile it. But I still don't think it has DPDK support enabled - at least I see no mention about it when running `gnuradio-config-info --enabled-components`.
I am clearly doing something wrong..

dominikheinz, 2025-04-10:
@h57jafari Thanks for the reply. I think I have now everything installed correctly - followed your guide step by step.
And of course `uhd_config_info --enabled-components` does indicate that DPDK is infact enabled.

Also, I am seeing the EAL output when trying to start the gnb.
See below:
```
lab@gnb:~/srsRAN_Project-dpdk/build/apps/gnb$ sudo ./gnb -c siso100.yml 

--== srsRAN gNB (commit 2be82d8ea) ==--

Lower PHY in quad executor mode.
Available radio types: uhd.
[INFO] [UHD] linux; GNU C++ version 13.3.0; Boost_108300; DPDK_23.11; UHD_4.8.0.HEAD-0-g308126a4
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'use_dpdk=1,pci=0000:02:00.0,pci_master=0000:02:00.0,pci_secondary=0000:02:00.1,type=x300,recv_frame_size=9000,send_frame_size=9000,mtu=9000,master_clock_rate=184.32e6'
[DEBUG] [DPDK] DPDK version: DPDK 23.11.0
EAL: Detected CPU lcores: 14
EAL: Detected NUMA nodes: 1
EAL: Detected shared linkage of DPDK
EAL: Multi-process socket /var/run/dpdk/rte/mp_socket
EAL: Selected IOVA mode 'VA'
EAL: VFIO support initialized
EAL: Using IOMMU type 1 (Type 1)
EAL: Ignore mapping IO port bar(1)
EAL: Ignore mapping IO port bar(4)
EAL: Probe PCI driver: net_i40e (8086:1572) device: 0000:02:00.0 (socket -1)
i40e_GLQF_reg_init(): i40e device 0000:02:00.0 changed global register [0x002689a0]. original: 0x00000000, new: 0x00000029 
i40e_GLQF_reg_init(): i40e device 0000:02:00.0 changed global register [0x00268ca4]. original: 0x00001840, new: 0x00009420 
i40e_aq_debug_write_global_register(): i40e device 0000:02:00.0 changed global register [0x0026c7a0]. original: 0xa8, after: 0x28
EAL: Ignore mapping IO port bar(1)
EAL: Ignore mapping IO port bar(4)
EAL: Probe PCI driver: net_i40e (8086:1572) device: 0000:02:00.1 (socket -1)
TELEMETRY: No legacy callbacks, legacy socket not created
[ERROR] [DPDK] Could not find route to destination address 10.45.255.255
[ERROR] [X300] X300 Network discovery error RuntimeError: DPDK: Could not find route to destination address 10.45.255.255
[ERROR] [DPDK] Could not find route to destination address 192.168.1.255
[ERROR] [X300] X300 Network discovery error RuntimeError: DPDK: Could not find route to destination address 192.168.1.255
Failed to open device with address 'use_dpdk=1,pci=0000:02:00.0,pci_master=0000:02:00.0,pci_secondary=0000:02:00.1,type=x300,recv_frame_size=9000,send_frame_size=9000,mtu=9000': LookupError: KeyError: No devices found for ----->
Device Address:
    use_dpdk: 1
    pci: 0000:02:00.0
    pci_master: 0000:02:00.0
    pci_secondary: 0000:02:00.1
    type: x300
    recv_frame_size: 9000
    send_frame_size: 9000
    mtu: 9000
    master_clock_rate: 184.32e6

srsRAN ERROR: Unable to create radio session.
```

However, as you can see, it still fails to interact with the device, and likely, my configuration file is incorrect.
Do you mind helping me to adjust my config to interact with the device accordingly.
A small example config that uses DPDK and interacts with my X310 would be nice, then I can expand from there.

Currently I have this:
```
lab@gnb:~/srsRAN_Project-dpdk/build/apps/gnb$ cat siso100.yml 
cu_cp:
  amf:
    addr: 127.0.0.5                # The address or hostname of the AMF.
    bind_addr: 127.0.0.1              # A local IP that the gNB binds to for traffic from the AMF.
    supported_tracking_areas:
      - tac: 7
        plmn_list:
          - plmn: "99970"
            tai_slice_support_list:
              - sst: 1
ru_sdr:
  device_driver: uhd                # The RF driver name.
  # device_driver: dpdk # The RF driver name.
  # device_args: type=x300,recv_frame_size=9000,send_frame_size=9000,mtu=9000            # Optionally pass arguments to the selected RF driver.
  device_args: use_dpdk=1,pci=0000:02:00.0,pci_master=0000:02:00.0,pci_secondary=0000:02:00.1,type=x300,recv_frame_size=9000,send_frame_size=9000,mtu=9000            # Optionally pass arguments to the selected RF driver.
  clock: internal # gpsdo # external                           # Specify the clock source used by the RF.
  sync: internal # gpsdo # external                            # Specify the sync source used by the RF.
  #srate: 184.32                    # RF sample rate might need to be adjusted according to selected bandwidth.
  #srate: 92.16
  srate: 46.08
  tx_gain: 30                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 26                      # Receive gain of the RF might need to adjusted to the given situation.
  #time_alignment_calibration: -54

cell_cfg:
  dl_arfcn: 650000                  # ARFCN of the downlink carrier (center frequency).
  band: 78                          # The NR band.
  channel_bandwidth_MHz: 40        # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                    # Subcarrier spacing in kHz used for data.
  plmn: 99970                     # PLMN broadcasted by the gNB.
  tac: 7                            # Tracking area code (needs to match the core configuration).
  pci: 1                            # Physical cell ID.

  #pdsch:
  #  mcs_table: qam256

# hal:
#   eal_args: "--lcores '(0-1)@(0-13)'"

# ru_ofh:
#   cells:
#     - network_interface: 0000:02:00.0
      

expert_execution:
  threads:
    upper_phy:
      nof_pusch_decoder_threads: 4
      nof_ul_threads: 4
      nof_dl_threads: 4

log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: debug                # Logging level applied to all layers.

pcap:
  mac_enable: false                 # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: false                # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.
```

But even after reading the Configuration Reference, I am unsure what to do.

dominikheinz, 2025-04-10:
The same issue happens if I just try to run `uhd_find_devices` with the `use_dpdk=1,type=x300` args.
No idea how to make it find my SDR.

h57jafari, 2025-04-10:
Does "uhd_find_devices" and "uhd_usrp_probe" work? can you share their output, with and without DPDK.

h57jafari, 2025-04-10:
what is that broadcast IP of "10.45.255.255". you may need add routing, I am not sure!

dominikheinz, 2025-04-10:
@h57jafari I don't know what this broadcast IP is either. Furthermore, I don't know how to add a routing for this IP address since the kernel is no longer managing the SFP Plus interfaces and thus I can't specify the interface when adding a route.
Does this routing need to be specified via DPDK?

How do I know which address is addr and which one is the mgmt_addr?

dominikheinz, 2025-04-16:
@h57jafari Any ideas?

h57jafari, 2025-04-16:
Hi Domimik, it seems you have 4 SFP ports: 
|   |  ip-addr0: 192.168.10.4
|   |   subnet0: 255.255.255.0
|   |   ip-addr1: 192.168.20.2
|   |   subnet1: 255.255.255.0
|   |   ip-addr2: 192.168.30.2
|   |   subnet2: 255.255.255.0
|   |   ip-addr3: 192.168.40.2

but I believe you only have two of them active. in the following link check the "Running UHD Applications with DPDK" part: https://kb.ettus.com/Getting_Started_with_DPDK_and_UHD

I need to mention that successfully operation of DPDK is one of most difficult work in SDR world. Please be patient and continue to debug. I hope you will solve it soon.

dominikheinz, 2025-04-17:
@h57jafari Hmm not sure why there are 4 listed. The SDR definitely only has 2 SFP+ Port (though, it can also be operated via PCIe).

The link you provided mainly refers to running the application as root (not just sudo), though, that didn't solve the discovery issue.

I am aware its difficult to set up, but I assume it should be doable. Everything is mostly in place, it is just the discovery of the SDR with DPDK that is not working.
I was experimenting with creating an `uhd.conf` file and define the mac addresses of the interfaces in there, but that did not work out either.

Do yo have any other ideas how I could proceed, or should I try to reach out to the guys at Ettus/UHD?

Alan1321, 2025-06-30:
Facing the same issue. It solves itself when I restart my computer, but start getting it after some time again (~2/3 days).

luyuanfan, 2026-01-31:
> Does anyone solved dominikheinz, issue ([Real-time failure in RF])? I am facing the same problem right now.

I remember trying the realtime kernel and the problem went away (low latency kernel worked as well and is easier to install)

luyuanfan, 2026-02-04:
> Hi [@luyuanfan](https://github.com/luyuanfan), Thanks for your input. I have tried the low-latency kernel, but the issue is still

Just to make sure, you ran the gnb with something like `sudo chrt --rr 99 ./srsRAN_Project/build/apps/gnb/gnb -c gnb.yml`?