srsran/srsRAN_Project #1143: srsUE won't connect to/see gNB

Alan1321, 2025-04-02:
Hello,

**I have the following setup:**
gNB -> B210 with an octoclock
srsUE -> B200mini with an octoclock
Intel core i7 cpu with 32 gigs of RAM

I am using the latest version of srs4G for the srsUE.
I am using somewhat recent(built it about a month ago) version of srsRAN_Project. 

I used the exact same configuration file as mentioned in:
https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html
But the srsUE doesn't connect to the gNB. I checked the logs and pcap but I don't see anything there. I will attach all the files below.
[alan_ue_gnb_pcap_logs.zip](https://github.com/user-attachments/files/19574886/alan_ue_gnb_pcap_logs.zip)

When I try to connect this same srsUE to the 4G LTE network (with a different config) it works. 
The gNB won't connect to COTS UE either. I have tested 3 till now:
1. Motorola one 5G ace
2. Sony Xperia 10 III
3. Google Pixel 6a

gNB(docker) and Open5GS(docker) core are able to connect, but any UE's I've tried until now whether it be srsUE or COTS UE won't connect.

Would really appreciate any help.

Alan1321, 2025-04-03:
These are my console logs:

**Open5GS core:**
`5gc-1  | 04/03 18:07:46.172: [app] INFO: Configuration: 'open5gs-5gc.yml' (../lib/app/ogs-init.c:130)
5gc-1  | 04/03 18:07:46.174: [dbi] INFO: MongoDB URI: 'mongodb://127.0.0.1/open5gs' (../lib/dbi/ogs-mongoc.c:130)
5gc-1  | 04/03 18:07:46.174: [sbi] INFO: NF Service [nudr-dr] (../lib/sbi/context.c:1812)
5gc-1  | 04/03 18:07:46.174: [sbi] INFO: nghttp2_server() [http://127.0.0.20]:7777 (../lib/sbi/nghttp2-server.c:414)
5gc-1  | 04/03 18:07:46.174: [app] INFO: UDR initialize...done (../src/udr/app.c:31)
5gc-1  | 04/03 18:07:46.174: [nrf] INFO: [8ba9c4ca-10b6-41f0-b1c0-81892cdabd37] NF registered [Heartbeat:10s] (../src/nrf/nf-sm.c:190)
5gc-1  | 04/03 18:07:46.175: [sbi] INFO: (NRF-notify) NF registered [8ba9c4ca-10b6-41f0-b1c0-81892cdabd37:1] (../lib/sbi/nnrf-handler.c:924)
5gc-1  | 04/03 18:07:46.175: [sbi] INFO: [UDR] (NRF-notify) NF Profile updated [8ba9c4ca-10b6-41f0-b1c0-81892cdabd37:1] (../lib/sbi/nnrf-handler.c:938)
5gc-1  | 04/03 18:07:46.175: [sbi] INFO: (NRF-notify) NF registered [8ba9c4ca-10b6-41f0-b1c0-81892cdabd37:1] (../lib/sbi/nnrf-handler.c:924)
5gc-1  | 04/03 18:07:46.175: [sbi] INFO: [UDR] (NRF-notify) NF Profile updated [8ba9c4ca-10b6-41f0-b1c0-81892cdabd37:1] (../lib/sbi/nnrf-handler.c:938)
5gc-1  | 04/03 18:07:46.175: [sbi] INFO: (NRF-notify) NF registered [8ba9c4ca-10b6-41f0-b1c0-81892cdabd37:1] (../lib/sbi/nnrf-handler.c:924)
5gc-1  | 04/03 18:07:46.175: [sbi] INFO: [UDR] (NRF-notify) NF Profile updated [8ba9c4ca-10b6-41f0-b1c0-81892cdabd37:1] (../lib/sbi/nnrf-handler.c:938)
5gc-1  | 04/03 18:07:46.175: [sbi] INFO: [8ba9c4ca-10b6-41f0-b1c0-81892cdabd37] NF registered [Heartbeat:10s] (../lib/sbi/nf-sm.c:221)
5gc-1  | 04/03 18:07:46.176: [nrf] INFO: [8baa2244-10b6-41f0-b1f1-0dd959715589] Subscription created until 2025-04-04T18:07:46.176067+00:00 [validity_duration:86400] (../src/nrf/nnrf-handler.c:445)
5gc-1  | 04/03 18:07:46.176: [sbi] INFO: [8baa2244-10b6-41f0-b1f1-0dd959715589] Subscription created until 2025-04-04T18:07:46.176067+00:00 [duration:86400,validity:86399.999802,patch:43199.999901] (../lib/sbi/nnrf-handler.c:708)
5gc-1  | 04/03 18:07:57.058: [amf] INFO: gNB-N2 accepted[10.53.1.3]:54943 in ng-path module (../src/amf/ngap-sctp.c:113)
5gc-1  | 04/03 18:07:57.058: [amf] INFO: gNB-N2 accepted[10.53.1.3] in master_sm module (../src/amf/amf-sm.c:741)
5gc-1  | 04/03 18:07:57.061: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:1231)
5gc-1  | 04/03 18:07:57.061: [amf] INFO: gNB-N2[10.53.1.3] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:780)`

**gNB**
`root@82f178953d9f:/build/srsRAN_Project/build/apps/gnb# gnb -c /home/ubuntu/gnb_rf_b200_tdd_n78_20mhz.yml

--== srsRAN gNB (commit 8120ffbce) ==--

Lower PHY in quad executor mode.
Available radio types: uhd and zmq.
[INFO] [UHD] linux; GNU C++ version 14.2.0; Boost_108300; UHD_4.8.0.0-64-g0dede88c
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Making USRP object with args 'type=b200,num_recv_frames=512,num_send_frames=512'
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Setting master clock rate selection to 'automatic'.
[INFO] [B200] Asking for clock rate 16.000000 MHz... 
[INFO] [B200] Actually got clock rate 16.000000 MHz.
[INFO] [MULTI_USRP] Setting master clock rate selection to 'manual'.
[INFO] [B200] Asking for clock rate 30.720000 MHz... 
[INFO] [B200] Actually got clock rate 30.720000 MHz.
Cell pci=1, bw=10 MHz, 1T1R, dl_arfcn=123400 (n71), dl_freq=617 MHz, dl_ssb_arfcn=122890, ul_freq=663 MHz

N2: Connection to AMF on 10.53.1.2:38412 completed
==== gNB started ===
Type <h> to view help
t: 
Available commands:
        cfo <sector_id> <cfo>:           set CFO
        tx_gain <port> <gain>:                set Tx gain
        log <channel> <level>:                set log level for the given channel (experimental)
                                              Allowed channels are: PHY
                                              Note: Certain log messages might not be available if they depend on
                                              logging decorators that are not instanced with the initial log level
        t:                                    start/stop console trace
        q:                                    quit application
        rx_gain <port> <gain>:                set Rx gain
        ru_metrics:                           prints RU metrics once
        ho <serving pci> <rnti> <target pci>: force UE handover
        sleep <seconds>:                      stops the execution of console sequential commands for the specified time

ru_metrics
| ------------------- TX ------------------ | ------------------- RX ------------------ |
| Avg. power | Peak power | PAPR | Clipping | Avg. power | Peak power | PAPR | Clipping |
|      -24.5 |       -4.5 | 20.0 |  0.0e+00 |        nan |     -379.3 |  nan |      nan |
Late: 0; Underflow: 2; Overflow: 0;
Late: 12; Underflow: 1; Overflow: 0;`

apoprassas94, 2025-04-07:
Hello,

I have partially the same issue.

More specifically, I followed this guide https://docs.srsran.com/projects/project/en/latest/tutorials/source/srsUE/source/index.html, too. The use case with the 2 machines and 2 usrps.

The setup I used is 2 nuc machines with:
Intel(R) Core(TM) i7-5557U CPU @ 3.10GHz & 16GB of RAM
and 2 Ettus Research USRP B210s (connected over USB3).

The config files that I used are the ones provided in this guide.

Following the guide, I deploy core network via the docker compose command.

Then, I deployed the gNB to one of the two nuc, and attach to it one of the two usrps.
Core network and gNB runs on the same nuc and they are connected successfully.

On the other nuc, I run srsUE and attach the second usrp.

According to the log output of the open5g core, seems the UE to be connected and instantly to be disconnected.

I attach the files with the relative log outputs.
The configuration files are exactly the same as in the relative guide.

[core_ouput.txt](https://github.com/user-attachments/files/19630548/core_ouput.txt)
[gNB_output.txt](https://github.com/user-attachments/files/19630546/gNB_output.txt)
[ue_output.txt](https://github.com/user-attachments/files/19630547/ue_output.txt)

pgawlowicz, 2025-04-09:
Please execute this script on both PCs: https://github.com/srsran/srsRAN_Project/blob/main/scripts/srsran_performance

Could you try to apply the patch from this comment to srsUE, recompile and try again?
https://github.com/srsran/srsRAN_Project/issues/826#issuecomment-2370588502

How are the USRPs connected over-the-air or with the RF cables?

apoprassas94, 2025-04-09:
After executing the performance script in both nodes and applying the patch, the srsUE is not connected at all.

I attach the full logs of each component (srsUE, gNB and core).

The usrps are connected over the air.

[core_log.txt](https://github.com/user-attachments/files/19664125/core_log.txt)
[core_ouput.txt](https://github.com/user-attachments/files/19664124/core_ouput.txt)
[gnb_log.txt](https://github.com/user-attachments/files/19664126/gnb_log.txt)

pgawlowicz, 2025-04-09:
could you share ue log

apoprassas94, 2025-04-09:
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file /home/nuc1/ue_rf.conf...

Built in Release mode using commit ec29b0c1f on branch master.

Opening 1 channels in RF device=uhd with args=clock=internal
Supported RF device list: UHD zmq file
[INFO] [UHD] linux; GNU C++ version 11.2.0; Boost_107400; UHD_4.1.0.5-3
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [B200] Detected Device: B210
[INFO] [B200] Operating over USB 3.
[INFO] [B200] Initialize CODEC control...
[INFO] [B200] Initialize Radio control...
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Performing register loopback test... 
[INFO] [B200] Register loopback test passed
[INFO] [B200] Asking for clock rate 23.040000 MHz... 
[INFO] [B200] Actually got clock rate 23.040000 MHz.
Setting manual TX/RX offset to 300 samples
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=6254
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=6414
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=6574
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=6734
RF status: O=3, U=0, L=0
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=6894
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=7054
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0x39, tti=7214

pgawlowicz, 2025-04-09:
ue.log file

apoprassas94, 2025-04-09:
I edited the previous comment

[ue_log.txt](https://github.com/user-attachments/files/19664883/ue_log.txt)

pgawlowicz, 2025-04-09:
After applying the UE patch, could you try setting `timing_adv_nsamples=auto` ?

Could you also tune down a bit the tx_gain of srsgnb and also ue? sth like 60

apoprassas94, 2025-04-09:
You mean time_adv_nsamples, correct?

The result is as before the applying of the pacth, the UE is connected and instantly disconnected.

open5gs_5gc  | 04/09 13:50:53.951: [amf] INFO: gNB-N2 accepted[10.53.1.1]:59521 in ng-path module (../src/amf/ngap-sctp.c:113)
open5gs_5gc  | 04/09 13:50:53.951: [amf] INFO: gNB-N2 accepted[10.53.1.1] in master_sm module (../src/amf/amf-sm.c:741)
open5gs_5gc  | 04/09 13:50:53.958: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:1231)
open5gs_5gc  | 04/09 13:50:53.958: [amf] INFO: gNB-N2[10.53.1.1] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:780)
open5gs_5gc  | 04/09 13:51:00.257: [amf] INFO: InitialUEMessage (../src/amf/ngap-handler.c:401)
open5gs_5gc  | 04/09 13:51:00.257: [amf] INFO: [Added] Number of gNB-UEs is now 1 (../src/amf/context.c:2550)
open5gs_5gc  | 04/09 13:51:00.257: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[7] TAC[7] CellID[0x66c000] (../src/amf/ngap-handler.c:562)
open5gs_5gc  | 04/09 13:51:00.257: [amf] INFO: [suci-0-001-01-0000-0-0-0123456780] known UE by SUCI (../src/amf/context.c:1833)
open5gs_5gc  | 04/09 13:51:00.257: [gmm] WARNING: Registration request (../src/amf/gmm-sm.c:1595)
open5gs_5gc  | 04/09 13:51:00.257: [gmm] INFO: [suci-0-001-01-0000-0-0-0123456780]    SUCI (../src/amf/gmm-handler.c:166)
open5gs_5gc  | 04/09 13:51:00.257: [amf] WARNING: GUTI has already been allocated (../src/amf/context.c:1426)
open5gs_5gc  | 04/09 13:51:01.833: [amf] INFO: UE Context Release [Action:2] (../src/amf/ngap-handler.c:1698)
open5gs_5gc  | 04/09 13:51:01.833: [amf] INFO:     RAN_UE_NGAP_ID[0] AMF_UE_NGAP_ID[7] (../src/amf/ngap-handler.c:1699)
open5gs_5gc  | 04/09 13:51:01.833: [amf] INFO:     SUCI[suci-0-001-01-0000-0-0-0123456780] (../src/amf/ngap-handler.c:1702)
open5gs_5gc  | 04/09 13:51:01.833: [amf] INFO: [Removed] Number of gNB-UEs is now 0 (../src/amf/context.c:2557)

For clarification, I attach onde again the logs and the configuration files:

[ue_config.txt](https://github.com/user-attachments/files/19666342/ue_config.txt)
[gnb_log.txt](https://github.com/user-attachments/files/19666344/gnb_log.txt)
[ue_log.txt](https://github.com/user-attachments/files/19666343/ue_log.txt)
[gnb_config.txt](https://github.com/user-attachments/files/19666341/gnb_config.txt)

pgawlowicz, 2025-04-09:
not sure if the UE config is correct, you have all parameters twice, could you check?

apoprassas94, 2025-04-09:
I have already try many different combinations for tx and rx gains in both nodes.

It seems to be random when the connection happens (instantly) and when there is not at all.

The two usrps are on the same desk. Does it matter?

Also, I have noticed in the UE's output the message: "RF status: O=22, U=0, L=0"

Is it related to the overrun (the receiver is producing samples faster than the software can consume them) ?

If so, what else parameters could tune?

I remind you that I have already execute the srsran_performance script in both nodes.

pgawlowicz, 2025-04-09:
Could you try with small BW cell, e.g. 5MHz? And reduce sampling rates accordingly. 
Do you have a possibility to use external clock for the B210s?