srsran/srsRAN_Project #1152: CU unable to attach to ORAN SC RIC

Keilwerth11270, 2025-04-08:
# Issue with CU E2 Configuration not being applied

I'm encountering an issue where my srsRAN CU fails to connect to the O-RAN SC RIC through the E2 interface. When I run the CU, I get a connection error followed by a segmentation fault:

```
cciadmin@CCI-NUC2:~/srsRAN_Project/build/apps/cu$ sudo srscu -c srscu_mobile_ric.yml
--== srsRAN CU (commit ef4b0749a) ==--
E2AP: Failed to connect to RIC on 127.0.0.1:36421. error="Connection refused" timeout=0ms
N2: Connection to AMF on 127.0.0.5:38412 completed
Stack trace (most recent call last) in thread 188560:
#10   Object "", at 0xffffffffffffffff, in #
...
#0    Object "/usr/local/bin/srscu", at 0x55b70fe4c19e, in srsran::e2_setup_procedure::send_e2_setup_request()
Segmentation fault (Address not mapped to object [(nil)])
Segmentation fault
```

## My Configuration

I am using Open5GS Core with the CU, DU, and Near RT RIC all colocated on the same physical machine (Intel NUC).

srsRAN version information:
```
cciadmin@CCI-NUC2:~/srsRAN_Project$ git tag --contains HEAD
release_24_10_1
cciadmin@CCI-NUC2:~/srsRAN_Project$ git log -1
commit ef4b0749a12a3b1a8347ae01c937a621603b4069 (HEAD, tag: release_24_10_1)
Author: Andre Puschmann <andre@srs.io>
Date:   Thu Jan 9 08:46:21 2025 +0100
```

ORAN SC RIC version information:
```
cciadmin@CCI-NUC2:~/oran-sc-ric$ git log -1
commit 1356ee99cec9489acf8906768a7a8c927b07597d (HEAD)
Author: Piotr Gawlowicz <piotr@srs.io>
Date:   Tue Aug 27 10:47:32 2024 +0200
```

In my [CU configuration](https://github.com/user-attachments/files/19653617/cu_config.txt), I've explicitly set the E2 interface to use the RIC's Docker network IP:

```yaml
e2:
  enable_cu_e2: true
  addr: 10.0.2.10                   # RIC IP address
  bind_addr: 10.0.2.1
  port: 36421                       # RIC port
  sctp_rto_initial: 120
  sctp_rto_min: 120
  sctp_rto_max: 500
  sctp_init_max_attempts: 3
  sctp_max_init_timeo: 500
  e2sm_kpm_enabled: true
  e2sm_rc_enabled: true
```

However, looking at the logs (`tail -f /tmp/srscu_mobile_ric.log`), I can see the CU is ignoring these settings and trying to connect using localhost instead:

```
2025-04-08T16:53:54.157543 [SCTP-GW ] [I] E2AP: Bind to 127.0.0.1:0 was successful
2025-04-08T16:53:54.159801 [E2      ] [D] Establishing E2 connection to Near-RT RIC (127.0.0.1:36421)...
2025-04-08T16:53:54.159861 [SCTP-GW ] [E] E2AP: Failed to connect to RIC on 127.0.0.1:36421. error="Connection refused" timeout=0ms
```

## gNB also shows some problem

Here is the [gNB configuration file](https://github.com/user-attachments/files/19655263/gnb.txt):

```yaml
e2:
[... 107 lines cut ...]
      - .env
    environment:
      - CONTAINER_NAME=ric_${E2TERM_NAME}
      - HOST_NAME=ric_${E2TERM_NAME}_host
      - POD_NAME=${E2TERM_NAME}_pod
      - SERVICE_NAME=ric_${E2TERM_NAME}_service
      - print=1
      - RMR_SEED_RT=/opt/e2/dockerRouter.txt
      - RMR_SRC_ID=${E2TERM_IP}
    command: ./e2 -p config -f config.conf
    volumes:
      - type: bind
        source: ./ric/configs/e2term.conf
        target: /opt/e2/config/config.conf
      - type: bind
        source: ./ric/configs/routes.rtg
        target: /opt/e2/dockerRouter.txt
    networks:
      ric_network:
        ipv4_address: ${E2TERM_IP:-10.0.2.10}
```

Has anyone encountered this issue before? Any tips on how to get the CU to properly connect to the RIC through E2? My primary concern is getting the disaggregated CU to attach to the RIC. The information about the gNB behavior may be useful in troubleshooting.

I'm happy to share more logs or configuration files if needed. Many thanks!

pgawlowicz, 2025-04-09:
Hi @Keilwerth11270,
Thanks for reporting. Could you try with the newest version of the srsRAN_Project? I just checked and it works. 
Best,
Piotr

newt2105, 2025-04-09:
Hi @pgawlowicz ,
I already pull the newest version but the problem still there:

```
# minhkeke @ minhkeke in ~/Code/5G-RAN/srsRAN_Project/build/apps/cu on git:main x [23:04:48] C:139
$ sudo ./srscu -c cu.yml e2                                              

--== srsRAN CU (commit 2be82d8ea) ==--

N2: Connection to AMF on 10.53.1.2:38412 completed
E2AP: Failed to connect to RIC on 127.0.0.1:36421. error="Connection refused" timeout=0ms
F1-C: Listening for new connections on 127.0.10.1:38472...
Stack trace (most recent call last) in thread 66438:
#10   Object "", at 0xffffffffffffffff, in 
#9    Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 78, in clone3 [0x79b6f0d29c3b]
#8    Source "./nptl/pthread_create.c", line 447, in start_thread [0x79b6f0c9caa3]
#7    Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.33", at 0x79b6f10ecdb3, in 
#6    Object "/home/minhkeke/Code/5G-RAN/srsRAN_Project/build/apps/cu/srscu", at 0x5ed600d814b5, in std::thread::_State_impl<std::thread::_Invoker<std::tuple<srsran::unique_thread::make_thread(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, srsran::unique_function<void (), 32ul, false>, srsran::os_thread_realtime_priority, srsran::os_sched_affinity_bitmask const&)::{lambda()#1}> > >::_M_run()
#5    Object "/home/minhkeke/Code/5G-RAN/srsRAN_Project/build/apps/cu/srscu", at 0x5ed601a65db1, in std::_Function_handler<void (), (anonymous namespace)::worker_task_factory<srsran::detail::priority_task_queue>(srsran::detail::priority_task_queue&, unsigned int, unsigned int)::{lambda()#1}>::_M_invoke(std::_Any_data const&)
#4    Object "/home/minhkeke/Code/5G-RAN/srsRAN_Project/build/apps/cu/srscu", at 0x5ed601a46b43, in srsran::detail::task_strand_impl<srsran::priority_task_worker_pool_executor, srsran::detail::strand_queue<(srsran::concurrent_queue_policy)1> >::run_enqueued_tasks()
#3    Object "/home/minhkeke/Code/5G-RAN/srsRAN_Project/build/apps/cu/srscu", at 0x5ed601783eff, in srsran::task_details::smallbuffer_table_t<srsran::e2_entity::start()::{lambda()#1}, void>::call(void*) const
#2    Object "/home/minhkeke/Code/5G-RAN/srsRAN_Project/build/apps/cu/srscu", at 0x5ed601783442, in srsran::e2_entity::start()::{lambda()#1}::operator()() const::{lambda(srsran::detail::base_coro_frame<srsran::async_task<void>::promise_type>&)#1}::operator()(srsran::detail::base_coro_frame<srsran::async_task<void>::promise_type>&) const
#1    Object "/home/minhkeke/Code/5G-RAN/srsRAN_Project/build/apps/cu/srscu", at 0x5ed6017e201d, in srsran::e2_setup_procedure::operator()(srsran::detail::base_coro_frame<srsran::async_task<srsran::e2_setup_response_message>::promise_type>&)
#0    Object "/home/minhkeke/Code/5G-RAN/srsRAN_Project/build/apps/cu/srscu", at 0x5ed6017e030f, in srsran::e2_setup_procedure::send_e2_setup_request()
Segmentation fault (Address not mapped to object [(nil)])
[1]    66413 segmentation fault  sudo ./srscu -c cu.yml e2

``` 

Thank you!

Keilwerth11270, 2025-04-09:
We have updated the srsRAN to the most recent version as per these logs:

```
cciadmin@CCI-NUC2:~/srsRAN_Project$ git log -1
commit 2be82d8ea38e3a729850b702254952c04118cc38 (HEAD -> main, origin/test, origin/main, origin/HEAD)
Merge: f0eb3b90a b9adc663e
Author: codebot <group_11283275_bot_29fb07365fb84a160b31ccf01e2e2aa2@noreply.gitlab.com>
Date:   Fri Mar 14 09:21:42 2025 +0000
    Update main
    
    # Conflicts:
    #       include/srsran/e2/e2_cu_factory.h
    #       include/srsran/phy/upper/uplink_slot_pdu_repository.h
    #       include/srsran/ru/ru_configuration.h
    #       lib/cu_cp/task_schedulers/du_task_scheduler.cpp
    #       lib/du/du_low/metrics/CMakeLists.txt
    #       lib/ofh/transmitter/ofh_uplink_request_handler_task_dispatcher.h
    #       lib/phy/upper/uplink_processor_task_dispatcher.h
```

CU is finally able to attach to the RIC over E2:
```
cciadmin@CCI-NUC2:~/srsRAN_Project/build/apps/cu$ sudo srscu -c srscu_mobile_ric.yml
--== srsRAN CU (commit 2be82d8ea) ==--
N2: Connection to AMF on 127.0.0.5:38412 completed
E2AP: Connection to Near-RT-RIC on 10.0.2.10:36421 completed
F1-C: Listening for new connections on 10.147.20.114:38472...
E2AP: Connection to Near-RT-RIC on 10.0.2.10:36421 completed
==== CU started ===
```

We see two E2AP Connection completion messages because in the configuration, we have CU_CP and CU_UP set as true:
```yaml
e2:
  enable_cu_cp_e2: true 
  enable_cu_up_e2: true 
```

The RIC Prints this when the CU attaches to it:
```
ric_rtmgr_sim       | 2025/04/09 18:41:08 POST /ric/v1/handles/associate-ran-to-e2t  body: [{"E2TAddress":"10.0.2.10:38000","ranNamelist":["gnb_001_001_0000019b"]}] elapsed: 23.527µs
ric_rtmgr_sim       | 2025/04/09 18:41:08 POST /ric/v1/handles/associate-ran-to-e2t  body: [{"E2TAddress":"10.0.2.10:38000","ranNamelist":["gnbc_001_001_0000019b_0"]}] elapsed: 10.262µs
```

Also, when we enable E2 on the DU, we see this as we did before:
```
E2AP: Connection to Near-RT-RIC on 10.0.2.10:36421 completed
```

And the RIC Prints the corresponding message:
```
ric_rtmgr_sim       | 2025/04/09 18:41:35 POST /ric/v1/handles/associate-ran-to-e2t  body: [{"E2TAddress":"10.0.2.10:38000","ranNamelist":["gnbd_001_001_0000019b_0"]}] elapsed: 27.64µs
```

We attached a Quectel modem to the network and successfully ran iPerf traffic in UL and DL. Next, we tested two of the sample xApps from the ORAN SC RIC repository. 

To begin, we ran the simple_mon_xapp.py xApp with this command:
```
sudo docker compose exec python_xapp_runner ./simple_mon_xapp.py --metrics=DRB.UEThpDl,DRB.UEThpUl
```

We got a successful output:
```
744223707665 48/RMR [INFO] ric message routing library on SI95 p=4561 mv=3 flg=00 id=a (f447e29 4.9.4 built: Dec 13 2023)
Subscribe to E2 node ID: gnbd_001_001_0000019b_0, RAN func: e2sm_kpm for metrics ['DRB.UEThpDl', 'DRB.UEThpUl']
Successfully subscribed with Subscription ID:  2vVIFC7XjMfiuhblMwC2IteHI1z
Received Subscription ID to E2EventInstanceId mapping: 2vVIFC7XjMfiuhblMwC2IteHI1z -> 2
{'response': 'OK', 'status': 200, 'payload': '{}', 'ctype': 'application/json', 'attachment': None, 'mode': 'plain'}
10.0.2.13 - - [09/Apr/2025 18:35:08] "POST /ric/v1/subscriptions/response HTTP/1.1" 200 -
RIC Indication Received from gnbd_001_001_0000019b_0 for Subscription ID: 2
[... 32 lines cut ...]
cciadmin@CCI-NUC2:~/oran-sc-ric$ sudo docker compose exec python_xapp_runner ./simple_rc_xapp.py
1744224645447 33/RMR [INFO] ric message routing library on SI95 p=4560 mv=3 flg=00 id=a (f447e29 4.9.4 built: Dec 13 2023)
18:50:46 Send RIC Control Request to E2 node ID: gnbd_001_001_0000019b_0 for UE ID: 0, PRB_min: 1, PRB_max: 20
Received RIC_CONTROL_ACK
18:50:51 Send RIC Control Request to E2 node ID: gnbd_001_001_0000019b_0 for UE ID: 0, PRB_min: 1, PRB_max: 50
```

When we ran the KPM xApp, the RIC printed a subscription message:
```
ric_rtmgr_sim       | 2025/04/09 18:34:10 POST /ric/v1/handles/xapp-subscription-handle  body: {"address":"10.0.2.20","port":4561,"subscription_id":1} elapsed: 10.081µs
```

This message never appeared when we ran the RC xApp.

For reference, I am on this commit of the ORAN SC RIC:
```
cciadmin@CCI-NUC2:~/oran-sc-ric$ git log -1
commit 1356ee99cec9489acf8906768a7a8c927b07597d (HEAD)
Author: Piotr Gawlowicz <piotr@srs.io>
Date:   Tue Aug 27 10:47:32 2024 +0200
    add option to run RIC on a separate host PC
cciadmin@CCI-NUC2:~/oran-sc-ric$ 
```

Other users have also reported getting a [similar issue](https://github.com/srsran/oran-sc-ric/issues/63) when sending RC messages to the latest version of srsRAN.

pgawlowicz, 2025-04-09:
Hi, indeed the per-slice PRB-quota control using the e2sm-rc module has some regression bug. I will look into it.

closed 2025-04-10 by Keilwerth11270

newt2105, 2025-04-11:
Hi @Keilwerth11270, can you share your gnb.log or du.log for me?
Thanks!

Keilwerth11270, 2025-04-14:
@newt2105 We have been running many trials since resolving this issue, so unfortunately I do not have those log files. However, I can confirm that I was able to resolve this by removing my srsRAN directory entirely, then pulling it again from scratch and rebuilding. Make sure your gcc/g++ versions are 11.4.0 or greater, and that your CMake command uses those versions.

Arbuzikk, 2025-04-17:
> Hi, indeed the per-slice PRB-quota control using the e2sm-rc module has some regression bug. I will look into it.

Hello! Do you have any news?

pgawlowicz, 2025-06-24:
The E2SM-RC PRB quota control was fixed in the newest srsRAN 5G Project release. Could you check?
Please also update the ORAN-SC-RIC repo.

Best,
Piotr

Keilwerth11270, 2025-06-27:
We made some modifications to the srsRAN codebase and were able to trigger handover via xApp with E2SM-RC and E2SM-KPM for measurement based trigger using the O-RAN SC Near RT RIC. We used two USRP B210s with both DUs running on the same machine. CU, Near RT RIC, and Open5GS Core were on separate machines. This work has been submitted to an IEEE/ACM publication and we plan to share the code changes publicly at a later stage if needed. May not be necessary given srsRAN has officially supported this now.