srsran/srsRAN_Project #1154: Possible wrong uint_to_five_qi() value

improperaffo, 2025-04-09:
In `include/srsran/cu_cp/cu_cp_configuration_helpers.h`, when making the `make_default_cu_cp_qos_config_list()`, there might be some typos for the 5QI indeces.

Namely, 5QI 6 and 8 have index 5 and 9, respectively.
5QI 6:
```
  {
    // 5QI = 6 e.g Video (Buffered Streaming)
    // PDB = 300ms PER = 10^-6
    srs_cu_cp::cu_cp_qos_config cfg{};
    pdcp_config                 pdcp_cfg{};

    pdcp_cfg.rb_type                       = pdcp_rb_type::drb;
    pdcp_cfg.rlc_mode                      = pdcp_rlc_mode::am;
    pdcp_cfg.ciphering_required            = true;
    pdcp_cfg.integrity_protection_required = false;

    // > Tx
    pdcp_cfg.tx.sn_size                = pdcp_sn_size::size18bits;
    pdcp_cfg.tx.discard_timer          = pdcp_discard_timer::ms300;
    pdcp_cfg.tx.status_report_required = false;

    // > Rx
    pdcp_cfg.rx.sn_size               = pdcp_sn_size::size18bits;
    pdcp_cfg.rx.out_of_order_delivery = false;
    pdcp_cfg.rx.t_reordering          = pdcp_t_reordering::ms100;

    cfg.pdcp                     = pdcp_cfg;
    qos_list[uint_to_five_qi(5)] = cfg;
  }
```

And 5QI 8:
```
  {
    // 5QI = 8 e.g Video (Buffered Streaming)
    // PDB = 300ms PER = 10^-6
    srs_cu_cp::cu_cp_qos_config cfg{};
    pdcp_config                 pdcp_cfg{};

    pdcp_cfg.rb_type                       = pdcp_rb_type::drb;
    pdcp_cfg.rlc_mode                      = pdcp_rlc_mode::am;
    pdcp_cfg.ciphering_required            = true;
    pdcp_cfg.integrity_protection_required = false;

    // > Tx
    pdcp_cfg.tx.sn_size                = pdcp_sn_size::size18bits;
    pdcp_cfg.tx.discard_timer          = pdcp_discard_timer::ms300;
    pdcp_cfg.tx.status_report_required = false;

    // > Rx
    pdcp_cfg.rx.sn_size               = pdcp_sn_size::size18bits;
    pdcp_cfg.rx.out_of_order_delivery = false;
    pdcp_cfg.rx.t_reordering          = pdcp_t_reordering::ms100;

    cfg.pdcp                     = pdcp_cfg;
    qos_list[uint_to_five_qi(9)] = cfg;
  }

pgawlowicz, 2025-04-09:
@alvasMan could you check?

robertfalkenberg, 2025-04-25:
Thanks for reporting this; its fixed and will be available upon next release.

closed 2025-04-25 by robertfalkenberg