srsran/srsRAN_Project #1157: Not create slice

newt2105, 2025-04-10:
Hi everyone,
I'm using srsRAN_Project at commit: c4996c8
When I search for slicing by using --help:
```
$ sudo ./gnb --help                                                                      

--== srsRAN gNB (commit c4996c8ee) ==--

srsGNB application
Usage: ./gnb [OPTIONS] [SUBCOMMAND]

Options:
  -h,--help                   Print this help message and exit
  -v,--version                Display program version information and exit
  -c                          Read config from file
  --ran_node_name [cu_cp_01]  RAN node name
  --slicing TEXT ...          Network slicing configuration
  --gnb_id [411]              gNodeB identifier
  --gnb_id_bit_length :INT in [22 - 32] [22] 
                              gNodeB identifier length in bits
  --gnb_du_id ENUM:UINT in [0 - 68719476735] [0] 
                              gNB-DU Id
  --cells TEXT ...            Sets the cell configuration on a per cell basis, overwriting the default configuration defined by cell_cfg
  --qos TEXT ...              Configures RLC and PDCP radio bearers on a per 5QI basis.
  --srbs TEXT ...             Configures signaling radio bearers.

Subcommands:
  log                         Logging configuration
  buffer_pool                 Buffer pool configuration
  metrics                     Metrics configuration
  e2                          E2 parameters
  expert_execution            Expert execution configuration
  hal                         HAL configuration
  amf                         AMF parameters
  cu_cp                       CU-CP parameters
  pcap                        PCAP configuration
  cu_up                       CU-CP parameters
  test_mode                   CU-UP test mode parameters
  cell_cfg                    Default cell configuration
  du                          DU parameters
  ntn                         NTN parameters
  expert_phy                  Expert physical layer configuration
  fapi                        FAPI configuration
  ru_ofh                      Open Fronthaul Radio Unit configuration
  ru_sdr                      SDR Radio Unit configuration
  ru_dummy                    Dummy Radio Unit configuration

(base) 

``` 
I see 
```
--slicing TEXT ...
``` 
but when I use that:
```
sudo ./gnb --slicing slice_id=1,sst=1,sd=0x010203,prb=40  -c gnb_zmq.yaml e2 --addr="10.0.2.10" --bind_addr="10.0.2.1"

``` 
and connect with UE, I see no slice created in gnb.log
Does this version support slicing? And how I use that?
Thanks!

jaiglesias-optare, 2025-04-30:
According to [reference](https://docs.srsran.com/projects/project/en/latest/user_manuals/source/config_ref.html), slicing options "can only be set via the configuration file."

closed 2025-05-13 by newt2105

AlejandraOliver, 2025-05-15:
Hi @newt2105 , did you manage to create slices? Did you run any xApps in FlexRIC like cust_slice_moni_ctrl?

Thanks!

newt2105, 2025-05-17:
> Hi [@newt2105](https://github.com/newt2105) , did you manage to create slices? Did you run any xApps in FlexRIC like cust_slice_moni_ctrl?
> 
> Thanks!

yeah, I create slice like this instruction : https://github.com/srsran/srsRAN_Project/discussions/756#discussioncomment-10240374
And I use oran-sc-ric for xapp. I used Flexric before, but it seems that it only run xapp_moni.

AlejandraOliver, 2025-05-17:
What xapp fform oran-sc-ric did you use? @newt2105

newt2105, 2025-05-18:
> What xapp fform oran-sc-ric did you use? [@newt2105](https://github.com/newt2105)

xapp moni and xapp rc control, but xapp rc control have some problem in recent version so I use some version from last year. I'm using this commit: `https://github.com/srsran/srsRAN_Project/commit/9d5dd742a70e82c0813c34f57982f9507f1b6d5d`
Here is the problem: https://github.com/srsran/oran-sc-ric/issues/63
What have you done in this project, we can have a discussion?

newt2105, 2025-05-19:
> But indeed the simple_rc_xapp.py from oran-sc-ric only work with one slice, does it contemplate asign specific PRBs to different slices?

really? Hmm, I'm not try that, I used to ask a question about that but haven't get the answer yet.
Did you find any other way?

AlejandraOliver, 2025-05-19:
No, nowadays I can´t get any xapp working with slicing

newt2105, 2025-05-20:
> No, nowadays I can´t get any xapp working with slicing

do you try the commit that I mentioned?