srsran/srsRAN_Project #1160: Issue related to time sync between benetel RAN550 (RU) and DU

Timez-zx, 2025-04-11:
# PTP Synchronization Issue with Benetel RAN550

## Issue Description

I'm encountering an issue while trying to achieve PTP synchronization with a Benetel RAN550 following the instructions in the [[srsRAN E2E O-RAN testbed documentation](https://docs.srsran.com/projects/project/en/latest/tutorials/source/hw_kit/source/index.html#ptp-grandmaster)](https://docs.srsran.com/projects/project/en/latest/tutorials/source/hw_kit/source/index.html#ptp-grandmaster).

When running ptp4l with the configuration files provided in the documentation, the slave device consistently reports a master offset of 0 and path delay of 0, which suggests that proper synchronization is not occurring despite the port state changing from UNCALIBRATED to SLAVE.

## System Information

- Hardware: Benetel RAN550
- Software: LinuxPTP (ptp4l)
- Network Interface: eno8303np0
- PTP Hardware Clock: /dev/ptp1

## Observed Behavior

When running the following command at RU side:
```
ptp4l -2 -E -s -i eth0 -f ./ptp/PTP_RU.cfg -m -p /dev/ptp1
```

I observe the following output:
```
ptp4l[3208988.418]: master offset          0 s0 freq    +151 path delay         0
ptp4l[3208990.418]: master offset          0 s0 freq    +166 path delay         0
ptp4l[3208992.418]: master offset          0 s0 freq    +157 path delay         0
ptp4l[3208994.418]: master offset          0 s0 freq    +161 path delay         0
ptp4l[3208996.418]: master offset          0 s0 freq    +163 path delay         0
```

The device successfully connects to the master clock and transitions to SLAVE state:
```
ptp4l[3212541.790]: port 1 (eth0): UNCALIBRATED to SLAVE on MASTER_CLOCK_SELECTED
```

However, the master offset and path delay consistently show 0 values, which is highly unusual for a properly functioning PTP synchronization.

## Configuration Files

I'm attaching my PTP configuration files to this issue. The key settings in my slave (PTP_RU.cfg) configuration include:

[srs-ptp-gm.txt](https://github.com/user-attachments/files/19698904/srs-ptp-gm.txt)
[PTP_RU.txt](https://github.com/user-attachments/files/19698903/PTP_RU.txt)

## Expected Behavior

In a properly functioning PTP setup, I would expect to see non-zero offset values that gradually converge toward zero as synchronization improves, and non-zero path delay measurements reflecting the actual network delay between master and slave.

## Questions

1. What might cause the constant zero values for both master offset and path delay?
2. Are there specific configuration parameters that need to be adjusted for the Benetel RAN550?
3. Could this be related to hardware timestamping issues or configuration conflicts?

Any guidance or troubleshooting suggestions would be greatly appreciated.

mmailand, 2025-04-11:
I think there is a mismatch in the logAnnounceInterval between the GM and RU:

- GM logAnnounceInterval -3 -> 8 Messages per second
- RU logAnnounceInterval 1 ->  2 Messages per second

The intervals need to be identical in a time domain.

You can see the identity of the master on the RU by issuing the following command:

pmc -u -d 24 -s /var/run/ptp4l
get PARENT_DATA_SET
	0XXXXa.fffe.c79134-0 seq 1 RESPONSE MANAGEMENT PARENT_DATA_SET
		parentPortIdentity                    0XXXXXa.fffe.c79134-0
		parentStats                           0
		observedParentOffsetScaledLogVariance 0xffff
		observedParentClockPhaseChangeRate    0x7fffffff
		grandmasterPriority1                  128
		gm.ClockClass                         248
		gm.ClockAccuracy                      0xfe
		gm.OffsetScaledLogVariance            0xffff
		grandmasterPriority2                  128
		grandmasterIdentity                   0XXXXa.fffe.c79134

In my example the parentPortIdentity and grandmasterIdentity are identical, that means I am "synced" to my local clock, if they are different you are synced to a remote clock.

Timez-zx, 2025-04-11:
## Update on PTP Synchronization Issue with Benetel RAN550

Thank you for your response. I've updated the configuration based on your suggestion regarding the mismatch in `logAnnounceInterval`. I modified the RU configuration to match the RU

[srs-ptp-gm.txt](https://github.com/user-attachments/files/19709679/srs-ptp-gm.txt)
[PTP_RU.txt](https://github.com/user-attachments/files/19709680/PTP_RU.txt)

's setting:

```
# Updated in DU config
logAnnounceInterval 1  # Changed from -3 to 1 to match RU
```

However, after making this change, I'm still experiencing the same issue. The PTP synchronization still shows constant zero values for both master offset and path delay:

```
ptp4l[70379.083]: master offset          0 s0 freq     -58 path delay         0
```

I also ran the suggested PMC command to check the parent data set:

```
root@benetelru:~# ./pmc -u -d 24 -s /var/run/ptp4l
get PARENT_DATA_SET
sending: GET PARENT_DATA_SET
        8c1f64.fffe.d113d3-0 seq 0 RESPONSE MANAGEMENT PARENT_DATA_SET 
                parentPortIdentity                    c8xxxx.fffe.9d5b1f-1
                parentStats                           0
                observedParentOffsetScaledLogVariance 0xffff
                observedParentClockPhaseChangeRate    0x7fffffff
                grandmasterPriority1                  128
                gm.ClockClass                         6
                gm.ClockAccuracy                      0x20
                gm.OffsetScaledLogVariance            0xffff
                grandmasterPriority2                  128
                grandmasterIdentity                   c8xxxx.fffe.9d5b1f
```

In this output, I notice that `parentPortIdentity` (c8xxxx.fffe.9d5b1f-1) and `grandmasterIdentity` (c8xxxx.fffe.9d5b1f) appear to match (aside from the -1 port number suffix). This MAC address (c8xxxx.9d5b1f) corresponds to my DU (Distributed Unit) network interface, not my local RU clock.

This suggests that my RU is detecting the DU as the grandmaster clock, but for some reason, the synchronization is not properly measuring offsets or path delays.

I've attached my updated configuration files. Any additional guidance would be greatly appreciated.

Timez-zx, 2025-04-11:
I just do some adjustment on the configuration file, it seems working now:

```
root@benetelru:~# cat /tmp/logs/radio_sync_status
Configuring RU for PTP Sync mode.
PTP Settings configured
Directory /usr/lib/benetel OK
Directory /usr/lib/benetel/splane OK
Directory /usr/lib/benetel/splane/flags OK
todsync started
PTP4L started
PCM4L started
Syncmon started
Waiting for initial PTP sync...
PTP locked, synchronizing system time.
[INFO] Radio synchronized
 16:30:20 up 18 min,  load average: 0.01, 0.17, 0.28
```
I also check this part:

```
root@benetelru:~# ./pmc -u -d 24 -s /var/run/ptp4l
get CURRENT_DATA_SET
sending: GET CURRENT_DATA_SET
        8c1f64.fffe.d113d3-0 seq 0 RESPONSE MANAGEMENT CURRENT_DATA_SET 
                stepsRemoved     1
                offsetFromMaster -1.0
                meanPathDelay    98.0
        c8xxxx.fffe.9d5b1f-1 seq 0 RESPONSE MANAGEMENT CURRENT_DATA_SET 
                stepsRemoved     0
                offsetFromMaster 0.0
                meanPathDelay    0.0
get PARENT_DATA_SET
sending: GET PARENT_DATA_SET
        8c1f64.fffe.d113d3-0 seq 1 RESPONSE MANAGEMENT PARENT_DATA_SET 
                parentPortIdentity                    c8xxxx.fffe.9d5b1f-1
                parentStats                           0
                observedParentOffsetScaledLogVariance 0xffff
                observedParentClockPhaseChangeRate    0x7fffffff
                grandmasterPriority1                  128
                gm.ClockClass                         6
                gm.ClockAccuracy                      0x20
                gm.OffsetScaledLogVariance            0xffff
                grandmasterPriority2                  128
                grandmasterIdentity                   c8xxxx.fffe.9d5b1f
        c8xxxx.fffe.9d5b1f-1 seq 1 RESPONSE MANAGEMENT PARENT_DATA_SET 
                parentPortIdentity                    c8xxxx.fffe.9d5b1f-0
                parentStats                           0
                observedParentOffsetScaledLogVariance 0xffff
                observedParentClockPhaseChangeRate    0x7fffffff
                grandmasterPriority1                  128
                gm.ClockClass                         6
                gm.ClockAccuracy                      0x20
                gm.OffsetScaledLogVariance            0xffff
                grandmasterPriority2                  128
                grandmasterIdentity                   c8xxxx.fffe.9d5b1f
```

mmailand, 2025-04-11:
Looks good now, happy that you could solve the issue.

Timez-zx, 2025-04-11:
Thanks for your help! I will close this with working configuration for ptp.

[PTP_RU.txt](https://github.com/user-attachments/files/19712122/PTP_RU.txt)
[srs-ptp-gm.txt](https://github.com/user-attachments/files/19712121/srs-ptp-gm.txt)

closed 2025-04-11 by Timez-zx