srsran/srsRAN_Project #1178: RX_LATE packets in Benetel RAN 550 with ORAN 7.2

antonioo420, 2025-04-24:
## Issue Description ##
I have been following this [tutorial](https://docs.srsran.com/projects/project/en/latest/tutorials/source/oranRU/source/index.html). I managed to see the cell in the UE, so i can assure the RU is transmitting, but the UE does not connect. 

The reason must be that I see some late packets on the kpi output in the benetel. However, the PTP output seems to be fine, and I have tried a couple of GNSS antennas and all of them have the same result.

Also the the Benetel seems to be receiving PTP synchronization:

`root@benetelru:~# cat /var/syncmon/sync-state `
`0`
`root@benetelru:~# cat /etc/ru-sync-mode`
`PTP`

The cat /var/syncmon/sync-state command responds with one of three numbers which align the following Sync Status:
0: Locked
1: Holdover
2: Freerun

kpi.sh
[image]

ptp4l
[image]

phc2sys
[image]

Configs

[gnb_ru_ran550_tdd_n78_100mhz_4x2.yml.txt](https://github.com/user-attachments/files/19888750/gnb_ru_ran550_tdd_n78_100mhz_4x2.yml.txt)

[ru_config.cfg.txt](https://github.com/user-attachments/files/19888752/ru_config.cfg.txt)

## Setup Details ##
OS: Debian 12
Server: DELL EMC PowerEdge R6525
CPU: AMD EPYC 7513 32-Core
Mem: 503 Gi
NIC: Intel E-810 QSFP
Switch: Falcon-RX
UE: Oppo Reno8 Lite 5G
CN: Dockerized Open5GS

## Expected Behavior ##
I expect the UE to connect to the network, and the RU to be perfectly synchronized

## Actual Behaviour ##
The UE does not connect, and I receive late packets on the Benetel RU.

jshacke, 2025-05-01:
Are you able to attach your FibroLAN RX configuration files along with the synchronization config file (.cfg)

antonioo420, 2025-05-02:
Sure, here they are

[ptp.cfg.txt](https://github.com/user-attachments/files/20009440/default.cfg.txt)

[Falcon_1970-01-01T00-09-00+00-00_startup-config.conf.txt](https://github.com/user-attachments/files/20009645/Falcon_1970-01-01T00-09-00%2B00-00_startup-config.conf.txt)

antonioo420, 2025-06-06:
No, I did not. What was exactly the mismatch? I have not changed any prach parameter from the config files of srsRAN tutorial.

antonioo420, 2025-08-25:
Followed this PTP configuration https://gitlab.eurecom.fr/oai/openairinterface5g/-/blob/develop/doc/ORAN_FHI7.2_Tutorial.md#ptp-configuration and now it works fine. All packets arrive on time and UE connects successfully.

closed 2025-08-25 by antonioo420