srsran/srsRAN_Project #1192: How to get lower latency?

meong3000, 2025-05-02:
SDR: USRP b205mini-i, UHD 4.7.0
Open5GS + srsRAN 5G SA (each on different PC, connected via ethernet), OTA setup

Is there any way to get lower latency? Successfully tried using Oppo Reno 8 5G, the lowest latency is 19 ms. The throughput is 114 Mbps (DL), 59 Mbps (UL). Already implemented the low_latency.yml config, thanks.

Here is my gnb config file
```bash
cu_cp:
  inactivity_timer: 7200
  amf:
    addr: 192.168.1.2 #127.0.1.100
    port: 38412
    bind_addr: 192.168.1.10
    supported_tracking_areas:
      - tac: 7
        plmn_list:
          - plmn: "00101" 
            tai_slice_support_list:
              -
                sst: 1

ru_sdr:
  device_driver: uhd
  device_args: type=b200, num_recv_frames=64, num_send_frames=64 
  clock: internal
  srate: 46.08  
  otw_format: sc12
  tx_gain: 80  #50
  rx_gain: 30  #60 #40
#  time_alignment_calibration: -165

cell_cfg:
  dl_arfcn: 627340 
  band: 78  
  channel_bandwidth_MHz: 40
  common_scs: 30 
  plmn: "00101" 
  tac: 7
  pci: 1

#low latency configs
  pusch:
    min_k2: 2
  tdd_ul_dl_cfg:
    dl_ul_tx_period: 4
    nof_dl_slots: 2
    nof_dl_symbols: 10
    nof_ul_slots: 1
    nof_ul_symbols: 0
  pucch:
    sr_period_ms: 2
    min_k1: 2
  mac_cell_group:
    bsr_cfg:
      periodic_bsr_timer: 1

  pdsch:
    mcs_table: qam256
  ssb:
    ssb_period: 20

log:
  filename: /tmp/gnb.log
  all_level: info

pcap:
  mac_enable: true
  mac_filename: /tmp/gnb_mac.pcap
  ngap_enable: true
  ngap_filename: /tmp/gnb_ngap.pcap

metrics: #untuk grafana metrics
    enable_json_metrics: true       
    addr: 172.19.1.4                
    port: 55555                  
```

andergalvaan, 2025-06-10:
Hello @meong3000 

Have you achieved lower latency? Any updates? I assume you deploy these performance tests against the UPF, right?

Best regards,
Ander

meong3000, 2025-06-14:
> Hello [@meong3000](https://github.com/meong3000)
> 
> Have you achieved lower latency? Any updates? I assume you deploy these performance tests against the UPF, right?
> 
> Best regards, Ander

Hi, i haven't reach a lower latency than 19 ms, and i use a speed test on my UE, so yes!