srsran/srsRAN_Project #1200: grafana metrics service func not available now???

PeterTheSparrow, 2025-05-10:
[image]
https://docs.srsran.com/projects/project/en/latest/user_manuals/source/grafana_gui.html

PeterTheSparrow, 2025-05-10:
i think maybe 1~2 months ago it's ok?

fllay, 2025-05-10:
The configuration options may be changed and the document may not be updated. The best way when we got the error like this is just try to figure out by ourself what is the new option. Just follow the instruction here 

https://github.com/srsran/srsRAN_Project/tree/main/configs

For example

```
❯ ./gnb -h

--== srsRAN gNB (commit 122a1377e3) ==--

srsGNB application
Usage: ./gnb [OPTIONS] [SUBCOMMAND]

Options:
  -h,--help                   Print this help message and exit
  -v,--version                Display program version information and exit
  -c                          Read config from file
  --dryrun                    Enable application dry run mode
  --ran_node_name [cu_cp_01]  RAN node name
  --gnb_cu_up_id ENUM:UINT in [0 - 68719476735] [0] 
                              gNB-CU-UP Id
  --gnb_id [411]              gNodeB identifier
  --gnb_id_bit_length :INT in [22 - 32] [22] 
                              gNodeB identifier length in bits
  --gnb_du_id ENUM:UINT in [0 - 68719476735] [0] 
                              gNB-DU Id
  --cells TEXT ...            Sets the cell configuration on a per cell basis, overwriting the default configuration defined by cell_cfg
  --qos TEXT ...              Configures RLC and PDCP radio bearers on a per 5QI basis.
  --srbs TEXT ...             Configures signaling radio bearers.

Subcommands:
  log                         Logging configuration
  buffer_pool                 Buffer pool configuration
  expert_execution            Expert execution configuration
  metrics                     Metrics configuration
  hal                         HAL configuration
  remote_control              Remote control configuration
  cu_cp                       CU-CP parameters
  pcap                        PCAP configuration
  e2                          E2 parameters
  cu_up                       CU-UP parameters
  cell_cfg                    Default cell configuration
  du                          DU parameters
  ntn                         NTN parameters
  test_mode                   Test mode configuration
  fapi                        FAPI configuration
  expert_phy                  Expert physical layer configuration
  ru_ofh                      Open Fronthaul Radio Unit configuration
  ru_sdr                      SDR Radio Unit configuration
  ru_dummy                    Dummy Radio Unit configuration
```
You can see `metrics`. Then you can just list the options and subcommands for `metrics`
```
❯ ./gnb -h metrics

--== srsRAN gNB (commit 122a1377e3) ==--

Metrics configuration
Usage: ./gnb metrics [OPTIONS] [SUBCOMMAND]

Options:
  -h,--help                   Print this help message and exit
  --autostart_stdout_metrics BOOLEAN [false] 
                              Autostart stdout metrics reporting
  --addr [127.0.0.1]          IP address JSON metrics will be reported to
  --port :INT in [0 - 65535] [55555] 
                              UDP port JSON metrics will be reported to
  --enable_json               Enable JSON metrics reporting
  --enable_log                Enable log metrics reporting

Subcommands:
  layers                      Layer basis metrics configuration
  periodicity                 Metrics periodicity configuration

 ~/src/srsRAN_Project/build/apps/gnb  main ?1 
```

now you can see that the one you interested in changed to `matrices.enable_json`.

closed 2025-05-11 by PeterTheSparrow