srsran/srsRAN_Project #1220: KPMs all 0 for simple_mon_xapp.py

Johnny0115, 2025-05-22:
Hi, I found the returned KPM values are all 0 when running simple_mon_xapp.py with osc-ric. Do you know the possible reasons?
I think there is no problem with the subscription.

[image]

Best regards.

newt2105, 2025-05-24:
Did you create traffic?

Johnny0115, 2025-05-28:
> Did you create traffic?

Thanks for your reply. I created traffic by surfing webpage with a cellphone.

newt2105, 2025-05-28:
> > Did you create traffic?
> 
> Thanks for your reply. I created traffic by surfing webpage with a cellphone.

Do you check the gnb trace? press t to trace

goyalvarun02, 2025-05-29:
I am also facing the same issue .
Please let me know if any progress.
I have created traffic via iperf and can se DL/UL brate around 11MBps.