srsran/srsRAN_Project #1239: slicing file not able to parse in gNB

27-Parth, 2025-06-05:
I am trying to implement network slicing using the open5Gs core network, and after doing the required changes in the network function files, I tried to change the configuration files srsRAN (gnb_zmq.yaml), I am using an external USRP to implement the gNodeB. This is the error which I encountered after doing the changes in the gnb file, I tried reverting back the changes but as I have changed the network functions, so that also won't work.

This is the error

**iisst@iisst:~/srsRAN_Project/build/apps/gnb$ sudo ./gnb -c gnb_zmq.yaml

-== srsRAN gNB (commit 4cf7513e9) ==--

INI was not able to parse cell_cfg.s_nssai
Run with --help for more information.**

This is the config file of the gnB

# This configuration file example shows how to configure the srsRAN Project gNB to allow srsUE to connect to it. 
# This specific example uses ZMQ in place of a USRP for the RF-frontend, and creates an FDD cell with 10 MHz bandwidth. 
# To run the srsRAN Project gNB with this config, use the following command: 
#   sudo ./gnb -c gnb_zmq.yaml

amf:
  addr: 127.0.0.5                   # The address or hostname of the AMF.
  port: 38412
  bind_addr: 127.0.0.5              # A local IP that the gNB binds to for traffic from the AMF.
ru_sdr:
  device_driver: uhd                # The RF driver name.
  device_args: type=x300,addr=192.168.10.2,master_clock_rate=184.32e6    #tx_port=tcp://127.0.0.1:2000,rx_port=tcp://127.0.0.1:2001,base_srate=23.04e6 # Optionally pass arguments to the selected RF driver.
  #sync:  gpsdo                     # Specify the sync source used by the RF.
#device_args: tx_port=tcp://192.168.0.105:2000,rx_port=tcp://192.168.0.114:2001,base_srate=23.04e6 # Optionally pass arguments to the selected RF driver.
  clock: internal 
  srate: 23.04                      # RF sample rate might need to be adjusted according to selected bandwidth.
  tx_gain: 30                       # Transmit gain of the RF might need to adjusted to the given situation.
  rx_gain: 30                       # Receive gain of the RF might need to adjusted to the given situation.

cell_cfg:
  dl_arfcn: 627340                  # ARFCN of the downlink carrier (center frequency).
  band: 78                          # The NR band.
  channel_bandwidth_MHz: 20         # Bandwith in MHz. Number of PRBs will be automatically derived.
  common_scs: 30                    # Subcarrier spacing in kHz used for data.
  plmn: "00101"                     # PLMN broadcasted by the gNB.
  tac: 1                            # Tracking area code (needs to match the core configuration).
  
  
  
#THE FOLLOWING LINES ARE ADDED IN order to test it for network slicing, remove if any issue is observed, regards---
  s_nssai: # <--- THIS LINE IS NOW CORRECTLY INDENTED BY 2 SPACES
    - sst: 1
      sd: 000001
    - sst: 8
      sd: 000002
      
#-------------------------------
  pdcch:
    common:
      ss0_index: 0                  # Set search space zero index to match srsUE capabilities
      coreset0_index: 12            # Set search CORESET Zero index to match srsUE capabilities
    dedicated:
      ss2_type: common              # Search Space type, has to be set to common
      dci_format_0_1_and_1_1: false # Set correct DCI format (fallback)
  prach:
    prach_config_index: 1           # Sets PRACH config to match what is expected by srsUE
  
cu_cp:
  inactivity_timer: 7200            # Sets the UE/PDU Session/DRB inactivity timer to 7200 seconds. Supported: [1 - 7200].

log:
  filename: /tmp/gnb.log            # Path of the log file.
  all_level: info                   # Logging level applied to all layers.
  hex_max_size: 0

pcap:
  mac_enable: true                  # Set to true to enable MAC-layer PCAPs.
  mac_filename: /tmp/gnb_mac.pcap   # Path where the MAC PCAP is stored.
  ngap_enable: true                 # Set to true to enable NGAP PCAPs.
  ngap_filename: /tmp/gnb_ngap.pcap # Path where the NGAP PCAP is stored.

pgawlowicz, 2025-06-05:
Please check here: https://docs.srsran.com/projects/project/en/latest/user_manuals/source/config_ref.html
You need to add proper sections in `amf` and `cell_cfg`.

27-Parth, 2025-06-05:
> Please check here: https://docs.srsran.com/projects/project/en/latest/user_manuals/source/config_ref.html You need to add proper sections in `amf` and `cell_cfg`.

I am trying to implement slicing and for that I made changes in the config files of Open5Gs, where I updated the NSSF, AMF files and also changed the subscriber database.

pgawlowicz, 2025-06-05:
Please open the link that I shared and look for `tai_slice_support_list` under the `amf` section, and `slicing` under `cell_cfg`.

27-Parth, 2025-06-06:
I went through the document and I also did the changes, but I am still getting the same error, GPT models are giving random reasons and none of them are working.