srsran/srsRAN_Project #1288: where to find the open5gcore for ntn

ggsnpdsn, 2025-07-09:
According to the section on supporting NTN NR applications in the SRSRSRAN manual, the open5gcore for NTN is located in the directory SRSGNB/Docker, but it was not found to exist in SRSRSRAN-Product. Where can I obtain it

pgawlowicz, 2025-07-09:
go to this directory https://github.com/srsran/srsRAN_Project/tree/main/docker on your host and start the core network with:
```
docker compose up 5gc
```

ggsnpdsn, 2025-07-09:
This is the v2.7.5 version of open5gs, haven't you seen any patches for ntn? Does the release of v2.7.5 support ntn nr by default? Where can I configure satellite information?

pgawlowicz, 2025-07-10:
The core network is aware that a connected gNB serves a NTN cell.

ggsnpdsn, 2025-07-12:
There are no specific configuration options for NTN in open5gs, right?

pgawlowicz, 2025-07-14:
No, but note that RTT in case of a GEO scenario might be up to 250ms. Therefore, some core network timeouts might require to be extended correctly, but from my experience Open5gs works out of the box as anyway the default timers are in order of seconds.

pgawlowicz, 2025-07-18:
srsRAN supports 5G NTN, and does not support NB IoT.

closed 2025-08-15 by pgawlowicz