srsran/srsRAN_Project #1309: 5G NTN srsRAN gNB logs requirement

Arun-Cdot, 2025-07-28:
We have successfully installed and run the srsRAN NTN gNB with a USRP. It is connected to the TM500 UE simulator, where SIB1 is successfully decoded. Additionally, we observe that SIB19 is also being scheduled.

[image]

From the image below, we can see that two PDSCHs are being transmitted. The transport block size (TBS) of 92 corresponds to SIB1, and we believe the TBS of 40 corresponds to SIB19.
[image]

The TM500 team has reported that since the si-WindowLength parameter is not present in SIB1, the TM500 is rejecting SIB19. According to them, si-WindowLength is only included in SIB1 when other SIBs such as SIB2 are also scheduled. However, when we attempt to schedule both SIB2 and SIB19, the srsRAN gNB throws the following error:

[image]

That said, we understand that the srsRAN 5G NTN gNB has already been tested with an Amarisoft UE. Was any workaround applied during that testing to address this issue?

pgawlowicz, 2025-07-28:
Hi, it should be possible but never checked.

pgawlowicz, 2025-09-26:
@Arun-Cdot any update on this issue?